EdgeTX/edgetx #5898: MT12 Companion

chofchop, 2025-02-12:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

The long-awaited companion for MT12 has finally been released. This is very gratifying.
Without further ado, I would like to point out a few points that caught my attention.

1. When Axis X and Axis Y are specified for Pots, they are not reflected in the simulator.
2. The simulator's button layout and the actual machine's switch panel are slightly different.
3. Unable to input "return" key using mouse wheel button.

### Describe the solution you'd like

1. Axis X and Axis Y inputs can be operated on the simulator.
2. Make the button layout of the simulator the same as the switch panel of the actual machine.
3. Since the rotary encoder is operated with the mouse wheel, the return key should be a mouse button.

### Describe alternatives you've considered

Regarding 3. above, the same applies not only to MT12 but also to all models equipped with a rotary encoder.

chofchop, 2025-02-12:
In the first post, I confused the "enter" and "return" keys. It has now been fixed.

pfeerick, 2025-02-12:
> Even if Axis X or Axis Y is assigned to S1 or S2,

Ok, so this _is_ related to the gyro/IMU simulation (or lack there of) then... meaning this is not MT12 specific. 

>  Is it really that difficult to move the button a little as shown in the image?

Probably not. Just need to bear in mind the shown layout only applies when you have "all" simulators (not just the MT12 one) configured to show the buttons on the right side, rather than the *Default* setting of buttons on both sides of the display.

> Registering multiple shortcut keys is not a difficult task, right? 

Nope, this is why multiple shortkeys already exist for some controls. The main question will be if QT5 allows for that event to be captured in the first place. 

> It would be best if the user could customize the shortcut keys.

This has been discussed already as something we already want, and are simply just not there yet. Also well beyond this scope of this issue ;)

chofchop, 2025-02-12:
Thank you for your detailed explanation.

elecpower, 2025-02-12:
The 1st iteration has tried to balance 3 OSes and multiple keyboard sizes/layouts.
Once this iteration has stabilised then customisation is an area for attention subject to priorities.