EdgeTX/edgetx #5899: Whilst powered on and inserting USB lead connected to pc radio powers off every time 2.11.0 RC

Kevltan, 2025-02-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Radio powers off when inserting usb lead connected to PC every time in 2.11.0 RC - TX16S mk1.
Does not do in 2.10.6.

### Expected Behavior

Expect to enter into usb mode and not power off.

### Steps To Reproduce

As above

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

Kevltan, 2025-02-12:
@philmoz 
I have loaded a basic SD card for 2.10.6 no extra files added and as soon as the usb lead inserted the radio crashes to off. So it doesnt seem to be related to SD card content.

3djc, 2025-02-12:
Cannot reproduce, the the USB choice menu pops like it should here. Tested on both MK1 and MK2

Kevltan, 2025-02-12:
What can possibly be the issue here then works on 2.10.6 does not work on 2.11.0 rc. New SD card basic 2.10.6 loaded on no additions whatsoever on it, start radio, put usb cable in to top of radio and instantly powers down every single time. Could there be something residual left in rom when loading firmware? Is worth doing a chip erase in STM32CUBE. I am lost for ideas here.

Kevltan, 2025-02-12:
Usually I have default usb mode set. But this is not the case with the last try as there was nothing set. I just plugged usb into the top and the radio just powered of into to sort of boot mode with the left hand led of six on.

koaxheli, 2025-02-12:
Hello, I have the same error now that I have the new v2.11.0-RC1 on the transmitter.
When I connect the transmitter with USB cable to the PC and
with USB memory (SD), the transmitter switches off and the connection is interrupted.

philmoz, 2025-02-12:
Are you able to build the firmware yourself?
If so can you try using the source from PR #5902 and commenting out line 488 in radio/src/main.cpp (call to handleUsbConnection).

Kevltan, 2025-02-12:
Yes will have ago.

Kevltan, 2025-02-12:
Have built this and this removes the usb capabilities completely - there is
now no option for usb and a little usb logo appears in the top right of the
screen and a drive is no longer created in windows when plugged in!
VERS: 3.0.0-selfbuild (ad2c6dec)

Kevltan, 2025-02-12:
OK will do

Kevltan, 2025-02-12:
Unedited 3.0.0 firmware the same..... crashes!

philmoz, 2025-02-12:
> Unedited 3.0.0 firmware the same..... crashes!

Crashes when USB mode is set to Joystick?

philmoz, 2025-02-12:
What about 'Storage' and 'Ask' modes for USB?

Kevltan, 2025-02-12:
Doesn't crash in ask mode until either storage or Joystick is selected from this. Crashes immediately if any one of these 2 modes is already selected/active.

philmoz, 2025-02-13:
You should not be able to power on the radio if the USB lead is plugged in. It should go into DFU mode if you plug in the USB cable when turned off.

Did you flash both bootloader and firmware?

It would be helpful if you could narrow down at what commit the problem started happening; but this can be a very tedious process.

Kevltan, 2025-02-13:
Yes meant inserting the lead after it is has started to power on.
Yes I have tried both 2.10.6 bootloader and 2.11.0 bootloader and directly from STM32CUBE from the PC same results. Can any data be left on the internal flash/rom in the chip that could be affecting this? Or is this some sort of firmware glitch or conflict?

philmoz, 2025-02-13:
It's not easily reproducible which makes things harder.

You said the problem is not there in 2.10 - have you reverted and retested to be sure?

If 2.10 works then some change in the code is not behaving on your camera - if we can isolate which commit it started happening in that might help identify the cause.

Kevltan, 2025-02-13:
Absolutely, it does not happen in 2.10.6 it is instant on 2.11.0 and reproducable every time. I have put minimum software on the SD card as well and still does it when 2.11.O firmware loaded. It's definately firmware related I think.

Kevltan, 2025-02-13:
If radio works in dfu mode surely it should work in usb mode unless there is problem with the coding for the TX16S MK1 as it would appear I am not the only one with this problem. Question is what have my radio and his radio got in common🤔
Roll on the Radiomaster Flagship!!!

philmoz, 2025-02-13:
What about in the bootloader - can you connect the USB and see the SD card mounted on your PC?

Kevltan, 2025-02-13:
Yes it connects in bootloader mode in device manager comes up as a usb STM32 bootloader on the pc.

I am able to upload firmware to the radio using STMcubeprogramer succesfully.

But as expected in this mode there is no Disk drive is created on the pc.

Kevltan, 2025-02-13:
Yes connects successfully in bootloader mode and creates a drive on the pc.

philmoz, 2025-02-13:
At this point all we can do is try and identify where in all the 2.11 changes the problem starts.
The only way to do that is to work through the 2.11 commits in Git; building and testing.
The most efficient way is pick a point halfway between latest commit to 2.11 and the first one (28th February 2024).
If it works then move forward to a later commit, otherwise move backwards.

Kevltan, 2025-02-13:
@philmoz 
I have noticed this problem has been creeping in for ages even in 2.10 originally but it went away. I created an issue for it but it didn't seem to get addressed I got "I can't reproduce on my radio".
In that instance it would indescrimately power off when in USB mode could be as soon as you plugged in or in about 10 minutes whilst you were downloading files to the card. But it seems completely stable in the latest 2.10.6 firmware. So something has changed in 2.11.0 which has exasperated the issue again.

Kevltan, 2025-02-13:
Could be a long and laborious task. But if you can send me some commits I could try these but as I said when I was testing 2.11 before it was so inconsistent sometimes it would power off sometimes it wouldn't. It will be so hard to find because it is so intermittent in the earlier versions - you thought it was fixed and then bang I had to resort to taking the card out and putting it in a reader to edit. 
But if I had known about the the bootloader option I probably would have used that instead!

J-Sorenson, 2025-02-13:
Are you familiar with the nightlies?  It sounds like you might need to try some of the older 2.11 nightlies to see if you can fine one that still works.  That will let the devs know which of the new commits broke the feature for your hardware.

Kevltan, 2025-02-13:
Tried this https://github.com/EdgeTX/edgetx/commit/a35246b5fa7a3658beb8dc4c3713aef792df2b6f

Kevltan, 2025-02-13:
@philmoz 
This is agood one from my selfbuild archive (5281b8b3) and have it written down as no conflicts.

koaxheli, 2025-02-13:
Hello, I have tested the last 3 versions with the result that the version from 17.12.24 was OK.
The version 22.12.24 is OK once and also crashed on a new attempt.
The last version from 12.2.25 always crashes after vPC connection.
Here are the compiled versions:
BOOT10edgetx-tx16s-2.11.0-selfbuild (12e84b57) still OK from 17.12.2024
BOOT10edgetx-tx16s-2.11.0-selfbuild (2b920a01) crash from 22.12.2024
BOOT10edgetx-tx16s-2.11.0-selfbuild (46089fe7) crashed on 12.02.2025 RC1

Kevltan, 2025-02-13:
Have tried  (12e84b57)  still crashes imediately when usb lead inserted on mine.

philmoz, 2025-02-13:
> Hello, I have tested the last 3 versions with the result that the version from 17.12.24 was OK. The version 22.12.24 is OK once and also crashed on a new attempt. The last version from 12.2.25 always crashes after vPC connection. Here are the compiled versions: BOOT10edgetx-tx16s-2.11.0-selfbuild ([12e84b5](https://github.com/EdgeTX/edgetx/commit/12e84b5710b225a5a271d189b2e574cfa9e431c9)) still OK from 17.12.2024 BOOT10edgetx-tx16s-2.11.0-selfbuild ([2b920a0](https://github.com/EdgeTX/edgetx/commit/2b920a014b0d69684eecfc20a74855d8b553e005)) crash from 22.12.2024 BOOT10edgetx-tx16s-2.11.0-selfbuild ([46089fe](https://github.com/EdgeTX/edgetx/commit/46089fe7d090ddd201ba10ce25ee7a60a3f01b04)) crashed on 12.02.2025 RC1

There is nothing changed between 17.12.2024 and 22.12.2024 that would affect USB.
This is sounding more like a hardware issue with the radio.

Kevltan, 2025-02-13:
How can this be a hardware problem. When 2.10.6 works perfectly fine and so does

philmoz, 2025-02-13:
> How can this be a hardware problem. When 2.10.6 works perfectly fine and so does

Bad RAM or bad FLASH memory.
The difference between the builds will be the memory map of where functions lie in the FLASH memory or which areas of RAM are being used.

3djc, 2025-02-13:
That could explain the variations in testing results

Kevltan, 2025-02-13:
Is it possible to erase the chip completely on STM32CUBE  or is this not a good idea as this will erase bootloader too. Why would this only effect the usb and nothing else????

Kevltan, 2025-02-13:
This could be true as I have noticed that the early version of the 2.11.0  that the usb works on has a smaller memory footprint than the later
2.11.0 RC1.

Kevltan, 2025-02-13:
I have looked at file sizes, and 2.10.6 is bigger than 2.11.0 RC1 as regards memory footprint but not sure how that equates as regards resources and memory usage.

pfeerick, 2025-02-13:
> Is it possible to erase the chip completely on STM32CUBE  or is this not a good idea as this will erase bootloader too. Why would this only effect the usb and nothing else????

Depends on what you mean by "bootloader". If you mean bootloader as it is described by EdgeTX,  then yes it will be erased, and will be written back when you write the firmware back (via STM32Cube /DFU). 

The DFU "bootloader" is in ROM, and CANNOT be erased. This is why these radios are generally referred to as unbeickable, as you can always use DFU to reflash firmware (except for hardware failure, naturally). 

> Doesn't the STM32CUBE verify when writing to the flash and RAM.

There is a specific verify option you can enable / tick where you configure firmware writing.

Kevltan, 2025-02-13:
@philmoz
I have checked on STM32CUBE  and it has verified that firmware on the chip is the same that is in the firmware file. It says there are no differences in the files.

Kevltan, 2025-02-13:
@pfeerick 
@pfeerick 
Have done full chip erase in STM32CUBE and reloaded 2.11.0 RC1 - no change.

Kevltan, 2025-02-20:
Any update (its gone quiete) or a possible resolution for this issue devs ?🤔 

I believe that once a few more people take the plunge and start updating to 2.11.0 that there are going to be few more instances of this problem arrising!

tonims1, 2025-02-22:
I have a mk1 and a mk2 there must be some difference between the two because in the mk1 as soon as I connect the USB and select storage it turns off, but in the mk2 it doesn't happen. both with the same firmware 2.11 RC1

Kevltan, 2025-02-22:
> As far as we know there isn't as far as Etx is concerned

As I said mine is not going to be an isolated case. I think there are going
to be a lot of TX16S's with this problem. There is something in 2.11 that
some of these radios don't like and it's bad enough to crash the radio to
bootloader mode.

3djc, 2025-02-22:
I don't disagree with that, but it clearly isn't as simple as mk1 have and mk2 not, or vice versa

philmoz, 2025-02-22:
Whatever is happening, I don't think it is some change in 2.11.
This issue show very similar things happening in 2.8 - #3039

Kevltan, 2025-02-22:
@philmoz 
Yes I think your right Phil, I seem to remember having faults with the usb shutting down before 2.10 but it would go away and I have some of my nightlies that I have written down as "No problem with usb". I did mention it I think on Discord but I thought it was something to do with IMU hardware I had added and this was causing some sort of conflict. I did ask someone to have a look at this issue but no one came back to me. It because it was so sporadic I just thought it would go away with later iterations of EdgeTX.

3djc, 2025-02-23:
All radio (no matter what the connector format is) use 'old usb'

Salto42, 2025-02-23:
Following my latest tests, the USB fault disappears if you switch the internal module “OFF”, whether it's “MULTI” or “ISRM”.

3djc, 2025-02-23:
Sounds like some kind of race condition. Just in case, has one of you tested a very recent nightly ? We found that in some conditions we might run out of RTOS timers so we increased their number, wonder if it might help

Salto42, 2025-02-23:
My last test was the “2.11.0-selfbuilt(2e1becd1)” version.

3djc, 2025-02-23:
Ok, so you had those changes

koaxheli, 2025-02-23:
I cannot confirm this, I have installed the same version BOOT10edgetx-tx16s-2.11.0-selfbuild (2e1becd1).
The error that the TX switches off is still present when plugging in the USB cable.
With “MULTI” switched on or off, the transmitter switches off.

Kevltan, 2025-02-23:
@3djc 
Race condition possible as when it was "sporadic" it seemed to be relative to when you plugged the lead in and how long you left it after start up. There seemed to be an ideal time to plug it in to "catch" it without it crashing, but there was no obvious time afterwards it seemed to be random , just luck I think - if that helps!?

Kevltan, 2025-03-02:
RC2 still hasn't sorted this issue out!!

Kevltan, 2025-03-11:
@pfeerick @philmoz @3djc 
RC2 still hasn't sorted this issue out!!
Any update or resolution to this problem as it does not seem to be in RC3 to-do list.

raphaelcoeffic, 2025-03-12:
@Kevltan it doesn't seem trivial to reproduce, and at least I could not. So there must be something special about your radio.

Possible differences:
- some specific configuration (some option, maybe widgets, etc...)
- something about the hardware itself (is your radio modified in any way? Did you add/remove something?)

Other than this, did anyone else reproduce?

Salto42, 2025-03-12:
From the previous messages and the few reports, I suspected a problem on my main board, so I replaced it with an MK2 and everything works fine in RC2.
My TX16S MK1 had been pre-ordered and I received it in May 2020.
The only modification was, in 2022, the replacement of a diode for the operation of an internal ISRM.

Kevltan, 2025-03-12:
Is the TX16S Mk2 main board a direct replacement for the Mk1 ?

Salto42, 2025-03-12:
I replaced the MK1 motherboard by the Mk2 with the same SD card and I no longer have any problems.

Kevltan, 2025-03-14:
@raphaelcoeffic 
@philmoz 
@3djc 
@pfeerick 
Can confirm that exchanging the Mk1 main board for the Mk2 cures the fault with the radio switching off on insertion of an active usb cable.🙂👍👍

TTGG4400, 2025-03-14:
My TX16S started faulting on USB storage mode after I upgraded to RC2. Plugging in the USB cord between Tx and laptop by itself does nothing - but when I select USB Storage mode on the Tx - the Tx puts up the USB Storage mode icon in white (no color) and then dies - takes a second or so.

[66 more replies]