EdgeTX/edgetx #5907: S1 and S2 are inverted

pierrotm777, 2025-02-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

S1 et S2 buttons are inverter from several version ago.
When I turn on the right, value pass from 1500 to 1000 and 1500 to 2000 when I turn on left

### Expected Behavior

Logically, for more value we turn on the right, is there a reason why the buttons are reversed ?

### Steps To Reproduce

Always

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

philmoz, 2025-02-13:
Check the hardware settings:

[image]

pierrotm777, 2025-02-13:
Thanks for your answer.
It's a good solution.
But these pot would be already as I hope, on the left at 1000 and on the right at 2000.
But I can do with.

pfeerick, 2025-02-14:
@pierrotm777 What were they set to? As generally they should both be unticked for stock setup of TX16S, to have increasing value as you turn clockwise.

pierrotm777, 2025-02-14:
Really sorry, it's solved now :-(

closed 2025-02-14 by 3djc