EdgeTX/edgetx #5909: Widget UI glitch with v2.11.0-RC1

agnauck, 2025-02-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When enabling/disabling the Sliders or Top bar settings the widgets are sometimes not adjusting to the change in available screen size for the widgets.

See this screenshot here. In this example sliders and Top bar were disabled and then got enabled. The Widgets are drawn outside of the desired screen area.
[image]

### Expected Behavior

It should look like this:

[image]

### Steps To Reproduce

Try to enable and disable Top Bar and/or Sliders. It does not always happen, but I can reproduce quite often.

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

agnauck, 2025-02-14:
When removing a view I just had similar issues. I assume this is addressed as well in #5910

pfeerick, 2025-02-15:
I believe so yes, as I didn't get any artefacts when deleting the current view, which was configured differently to the one that was switched to.

PR #5910 (2025-02-15): fix(color): view screen layout may not update correctly after changing screen setup.
Fixes #5909 

Ensure all view screens are correctly updated.

To reproduce, create a model with multiple view screens. Enter screen setup and change to a view other than the current visible one. Change settings (topbar, trims, sliders, etc). Exit screen setup and scroll to the changed screen.
closed 2025-02-15 by pfeerick