EdgeTX/edgetx #5915: (Lua 5.3) Large Hexadecimal Values - Conversion to integer change results in script not woking

MRC3742, 2025-02-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

Jumper T16 with ISRM internal module update hack and running v2.11.0-RC1.
After updating TX firmware the [ISRM Mode Changer](https://github.com/PD5DJ/ISRM-Mode-Changer) script no longer works.

It seems with the update to Lua 5.3 that larger Hex values are being converted as a two complement decimal, turning them into a negative integer. Debug screenshots below.

Is this just a consequence of updating to 5.3, requiring a script change, or is there a way to retain the previous behavior of 5.2 ?

**Code snippet causing the issue**
`local function telemetryPop()`
`  physicalId, primId, dataId, value = sportTelemetryPop()`
`    if primId == 0x32 and dataId >= 0X0C40 and dataId <= 0X0C4F then`
`      mode = math.floor(value / 0x1000000) - 0xA0`
`    end`
`end`

**EdgeTX v2.10.6 debug screen**
[image]

**EdgeTX v2.11.0-RC1 debug screen**
[image]

**Rapid Tables - Hex to Decimal Converter screenshot**
[image]

Thanks - Rich

### Expected Behavior

.......see above

### Steps To Reproduce

....

### Version

2.11.0-rc

### Transmitter

Jumper T16

### Operating System (OS)

Windows

### OS Version

10

3djc, 2025-02-15:
Some background: https://stackoverflow.com/questions/78213071/problems-with-large-numbers-in-lua

MRC3742, 2025-02-15:
Unfortunately "local value = 0.0" doesn't seem to work in this situation.
"value" is taken from the telemetry pop read and is converted to the two's complement decimal.

_It seems an option for a compatibility solution exists in the Lua 5.3 source code "luaconf.h" file._

`/*`
`@@ LUA_COMPAT_FLOATSTRING makes Lua format integral floats without a`
`@@ a float mark ('.0').`
`** This macro is not on by default even in compatibility mode,`
`** because this is not really an incompatibility.`
`*/`
`/* #define LUA_COMPAT_FLOATSTRING */`

Would this be able to be added to the current EdgeTX lua code ?  -  Rich

3djc, 2025-02-15:
I don't think so, we clearly want 5.3 to use integers where possible, and certainly not treat all variables as float. Did you try initialising to 0.0 and then assigning value from pop read ?

pfeerick, 2025-02-15:
Perhaps following the official notes on conversion, such a explicit formatting, might help here?

While I know it is annoying/frustrating, I do think this is something that really needs to be fixed in the Lua script, i.e. by _actually_ describing how a string should be formatted for display, if that is all that is happening. Flipping the compatibility toggles is only a quick fix anyway, and puts the problem off, as they will be removed sometime in the future. 
 https://www.lua.org/manual/5.3/manual.html#8

MRC3742, 2025-02-17:
I haven't been successful to initialize the variable "value" with 0.0 and have the script continue to work with Lua 5.3.
However I have successfully updated it with an 'if, else, end' statement to be compatible with both versions.

I thought the goal was to keep current compatibility with Lua 5.2 on version 2.11 until ETX  v3.0 was released?
:warning: Closing this issue because the current output is expected going forward with Lua 5.3. :warning:

Thanks for the advice and links to references - Rich

closed 2025-02-17 by MRC3742