EdgeTX/edgetx #5919: Radio settings changed in simulator persist temporarily when they should be reset.

philmoz, 2025-02-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

After setting a custom name on an Axis, Pot or Switch in the simulator the custom name persists after closing and re-opening the simulator.
The custom name is not shown in the companion radio settings, and does not survive closing and re-opening companion.

### Expected Behavior

All fields should be reset on closing the simulator window.

### Steps To Reproduce

- Start companion, create a new blank ETX file (no models)
- Start simulator, a default model is created
- Set a custom name for an axis, pot or switch in the radio hardware page
- Close the simulator
- Run the simulator again - check the axis/pot/switch settings. The names persist.

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2025-02-16:
Imma gonna place a $2 bet on temporary files being reused 😅

philmoz, 2025-02-16:
On MacOs the simulator temporary folder is deleted when the simulator is closed.

elecpower, 2025-02-16:
Ditto on Ubuntu

elecpower, 2025-02-16:
My $2 bet is on a deleteLater and later hasn't come around yet. I haven't checked the code yet so a real guess

pfeerick, 2025-02-16:
Spoilsports!

elecpower, 2025-02-18:
Tested as per steps above in Ubuntu X9D+, V16, T12MAX, MT12, T20V2 and TX16S. So far only TX16S remembers previous run settings. Even set names before launching simulator and makes no difference.

For all but TX16S the libsim is unloaded on dialog close despite the debug log stating the TX16S libsim was unloaded successfully. The V16 which is also a colour radio unloads so appears not to be colour vs B&W issue.

Checked via cmd 'cat /proc/NNNN/maps | awk '{print $6}' | grep 'libedge' | sort | uniq' where NNNN is the process id from cmd 'top'.

Tested other radios by themselves, before and after TX16S and TX16S is the only one not to unload. TX16S only unloads when the parent Companion executable unloads.

For the TX16S on sim launch the correct radio settings are written to the temp directory but as soon as the libsim radio starts running it overwrites the radio.yml file with the last run data from the libsim.

Open to ideas and wild guesses why TX16S libsim lingers.

elecpower, 2025-02-18:
Update: in the 1st test the V16 did not have an sd path set and unloaded from memory however a 2nd test with a valid sd path set results in the lib not unloading from memory.

The X9D+ did have a valid sd path set and unloaded.

elecpower, 2025-02-18:
I'm going to revert #5778 to see if it makes any difference

philmoz, 2025-02-19:
Looks like it started in commit f28c1150e (PR #4407) - the big LVGL refactor.

No idea why.

elecpower, 2025-02-19:
Couldn't start in a small PR could it aaaargh.

elecpower, 2025-02-19:
When Companion and TX16S simulator running the cmd 'lsof /src/build-debug/native/libedgetx-tx16s-simulator.so' reports in use by Companion and in my case the debugger 'gdb'.
When unloaded the cmd displays the same results.

For X9D+ loaded, same as for TX16S
For X9D+ unloaded, no results displayed as expected.

A diff of Companion process with X9D+ vs TX16S reports just the libsims as differences.

QLibrary::isLoaded() returns false before the TX16S libsim is loaded and after unloading even thought the OS reports it is loaded.

elecpower, 2025-02-19:
The issue goes away (well in testing so far) by reverting 5778. Why since all libsims remain loaded whereas only the used ones remain loaded with 5778 applied?

philmoz, 2025-02-19:
I have to go all the way back to just before PR 4407.

Here is a workaround patch that resets the radio and model memory on starting the simulator,

[sim_patch.diff.zip](https://github.com/user-attachments/files/18865627/sim_patch.diff.zip)

elecpower, 2025-02-19:
Its probably me but the diff appears to be doing lots more than stated objective eg yaml

philmoz, 2025-02-19:
I needed to move the switch and analog name storage into g_eeGeneral to make it easier to clear.

elecpower, 2025-02-20:
I tested your diff with the deferred libsim loading applied (ie main). The TX16S and V16 libsims behave as expected. Great work.

There is still the issue of colour libsims not always unloading. If I start a v16 libsim make changes and close it, it will remain loaded. If I then switch profiles to TX16S and start it, make changes and close, it gets unloaded and the v16 remains loaded.

So far b&w libsims alway get unloaded no matter the order.

philmoz, 2025-02-20:
Yeah, I don't know why this is happening - maybe something in LVGL needs shutting down.
I tried deleting all the styles and calling lv_deinit() on shutdown; but it still did not unload, and crashed when restarting the simulator.

elecpower, 2025-02-20:
In my trolling the web for a clue, there are cases of threads pointing to the parent which increments the in use counter stopping the unload if the tread does not clean up before termination. And many more ponderings...

The inconsistency in our case is baffling.

elecpower, 2025-02-20:
Since the workaround is your work I leave it for you to create the PR as you deserve the credit. Thank you.

I'll keep the issue on my todo list as could even be a Qt bug. I've started work on upgrading to Qt 6.8 so I'll retest once I have a working environment (it is not pretty at the moment)

raphaelcoeffic, 2025-02-26:
> I needed to move the switch and analog name storage into g_eeGeneral to make it easier to clear.

How is it normally cleared? Isn't it enough to just cancel the temporary files? The current fix moving things back to `g_eeGeneral` really looks like a way to hide another bug rather than a proper fix, IMHO.

raphaelcoeffic, 2025-03-16:
> As noted in the workaround PR the underlying cause in unknown. If you have a better fix fell free to update the PR.

Ok, so this is really an issue with properly unloading the dynamic libraries, so that when the simu is started again, the simulator's static memory is not re-initialised.

raphaelcoeffic, 2025-03-16:
A bit further down the rabbit hole: it seems really non-trivial to have C++ dynamic libraries unload properly for various reasons (see here: https://developer.apple.com/forums/thread/122591).

So the broader issue here is that we rely on static initialisation in various cases (the issue here is only one visible consequence, but not the only one), which won't happen again if the library is not properly unloaded.

So the only real fix would be to start the simulation in a different process, which would guarantee that the plugin library is really loaded on each execution.

raphaelcoeffic, 2025-03-16:
If that is really needed, I can however come up with a tiny fix for this particular issue at hand.

elecpower, 2025-03-16:
Linux has the same issue with not always unloading. It would appear that the B&W libsims always unload and a random (though maybe a pattern based on the load sequence) couple of colour ones stay loaded. If you load and unload all colour libsims a couple always stay loaded.

elecpower, 2025-03-16:
Shut down Companion and they all unload so they are tagged just to the app

raphaelcoeffic, 2025-03-16:
> Linux has the same issue with not always unloading. It would appear that the B&W libsims always unload and a random (though maybe a pattern based on the load sequence) couple of colour ones stay loaded. If you load and unload all colour libsims a couple always stay loaded.

This is most probably linked to some C++ stuff, as we use quite a bit more in color targets than we do with monochrome.

So, IMHO, the only option we have is to run the simulator as a separate process, which 100% unloads everything once the process is terminated. @elecpower is that something we could do? How much effort would that be?

elecpower, 2025-03-16:
> So, IMHO, the only option we have is to run the simulator as a separate process, which 100% unloads everything once the process is terminated.

Convert each so/dll to exe correct? If yes, would lose two way "communication" with Companion and Simulator executables. Also they are run inside modal dialogs so that pre-launch (eg passing models and settings) and post close activities can be managed. 

> [@elecpower](https://github.com/elecpower) is that something we could do? How much effort would that be?

Technically yes but I envisage lots of work. If executables then how to deal with a user launching multiple of same or different radios  (from Companion, Simulator or even from OS) and clashes with read/write shared resources.

I expect compilation times to blow out further and resulting executables to be larger.

raphaelcoeffic, 2025-03-17:
> Convert each so/dll to exe correct? If yes, would lose two way "communication" with Companion and Simulator executables. Also they are run inside modal dialogs so that pre-launch (eg passing models and settings) and post close activities can be managed.

Not exactly. I would use a sub-process which would load the regular simulator plug-in. Companion can monitor that process, and even communicate with it.

What I'm looking at right now would a process that would provide the bare minimum to interface properly with the simulator dialog via some kind of RPC. This would be very similar to what web browser do when running rendering processes for single web-sites inside a separate process.

> Technically yes but I envisage lots of work. If executables then how to deal with a user launching multiple of same or different radios (from Companion, Simulator or even from OS) and clashes with [...]

That would be tackled by basically having each simulator window be paired with a single "simulator process" controlled by the dialog.

elecpower, 2025-03-17:
Using QProcess would likely need to be called by Companion and Simulator apps and be the exe per radio approach but would seem to provide a "wait" on child process ability.

gagarinlg, 2025-03-17:
You could have a simulator exe that gets a parameter which radio to load

elecpower, 2025-03-17:
Issue I just noticed is that a lot of the "nicer" QProcess features have been progressively added in Qt 6 versions.

elecpower, 2025-03-17:
Qt 5 reference https://doc.qt.io/qt-5/qprocess.html

elecpower, 2025-03-17:
> You could have a simulator exe that gets a parameter which radio to load

On the face of it that appears a good idea. The stand alone Simulator accepts a command line now and no reason it cannot be tweaked if necessary. Then Companion runs the Simulator app in its own process bypassing the initial startup dialog going straight to the simulator main window.

Rip out the simulator code from Companion which would not be used.

Food for thought.

PR #6005 (2025-04-01): fix(sim): libsims not unloading
Fixes #5919
Alternative solution to #5931

Summary of changes:
- Companion writes models and settings to a temporary directory and launches the standalone Simulator in a new process
- the Simulator startup dialog is bypassed unless something goes unexpectantly wrong
- Companion waits for the Simulator to finish before becoming responsive (like a modal dialog)
- new Simulator cli option flags to permit Companion to pass flags that it previously passed to its self contained simulator