EdgeTX/edgetx #5920: dropdown list loop

shiron007, 2025-02-16:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Hello there and thankful for all EdgeTX i'm wondering if this help for all:
in the dropdown list of the Source/Switches or 
trigger/Value, it is recommended that when scrolling with the scroll wheel to the end of the list, it should loop back to the beginning of the list if the scrolling continues in the same direction. Currently, with a long list, it is necessary to scroll all the way back manually. While it is possible to exit and re-enter, which resets the list to the beginning, this is not always convenient.

### Describe the solution you'd like

it is recommended that when scrolling with the scroll wheel to the end of the list, it should loop back to the beginning of the list if the scrolling continues in the same direction

pfeerick, 2025-02-16:
2.11 does this already ;)

From the release notes for 2.11rc1

> Colorlcd: menus now "wrap around" when using the rotary encoder, allowing
you to get to back to the top of the list quicky (#4867)

---------- Forwarded message ---------

shiron007, 2025-02-16:
> 2.11 does this already ;) <br> <br> From the release notes for 2.11rc1
> <br> <br> &gt; Colorlcd: menus now &quot;wrap around&quot; when using the
> rotary encoder, allowing <br> you to get to back to the top of the list
> quicky (#4867) <br> <br> <br> <br> ---------- Forwarded message ---------
> <br> From: shiron007 ***@***.***&gt; <br> Date: Mon, 17 Feb 2025, 7:18 am
> <br> Subject: [EdgeTX/edgetx] dropdown list loop (Issue #5920) <br> To:
> EdgeTX/edgetx ***@***.***&gt; <br> Cc: Subscribed ***@***.***&gt; <br> <br>
> <br> Is there an existing issue for this feature request? <br> <br> - I
> have searched the existing issues <br> <br> Is your feature request related
> to a problem? <br> <br> Hello there and thankful for all EdgeTX i&#39;m
> wondering if this help for all: <br> in the dropdown list of the
> Source/Switches or <br> trigger/Value, it is recommended that when
> scrolling with the scroll wheel <br> to the end of the list, it should loop
> back to the beginning of the list if <br> the scrolling continues in the
> same direction. Currently, with a long list, <br> it is necessary to scroll
> all the way back manually. While it is possible <br> to exit and re-enter,
> which resets the list to the beginning, this is not <br> always convenient.
> <br> Describe the solution you&#39;d like <br> <br> it is recommended that
> when scrolling with the scroll wheel to the end of <br> the list, it should
> loop back to the beginning of the list if the scrolling <br> continues in
> the same direction <br> Describe alternatives you&#39;ve considered <br>
> <br> *No response* <br> Additional context <br> <br> *No response* <br>

Thanks 🙏🏽
Liron Dagan - from my gmail

closed 2025-02-16 by shiron007