EdgeTX/edgetx #5924: Allow to limit the mixes added value

Danitegue, 2025-02-17:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

For glider users that do not want to use the throttle stick for flaps/butterfly like me, I have in my TX16S the top left slider for butterfly, and the top right for flaps.
This means we need a mix with two inputs (LS+RS) for each flap. Currently there is no way to limit the sum of these two inputs to be in the range [0-100%], which would facilitate to use the overall output of that mix as input for other mixes (like flaps to elevator compensation).

### Describe the solution you'd like

When we have a mix with multiple inputs, it would be marvelous to be able to define, in each input, a multiplexor condition "add_within_range": i.e.  contribute but only until the sum of the inputs is within a user defined range [Max, Min]. If this is not possible per mix input, then it would be good to be able to limit the overall mix output.

inventor7777, 2025-02-18:
This would be helpful for me as well as a surface user, as I need a mix like that too.

philmoz, 2025-02-20:
Outputs already have min & max limits; but at the moment you can't select an 'Output' as the source for an input or mixer line.

Maybe the simplest solution would be to allow output values to be used as the source for inputs and mixer lines.

inventor7777, 2025-02-20:
Would that allow (for example) a surface throttle mix that acts normally and also have an additional throttle mix (activated by switch) that is limited to 0-100%?

philmoz, 2025-02-20:
I'm not sure I completely understand what you need to do.

Going back to the original scenario.
If you could assign an output as the source for a mix line, then you could have the LS + RS mixer going to 200%; but the corresponding output value would be limited to the output max (defaults to 100%).
So you could then take the output value (limited to 100%) and use it as a source for another mix such as flaps to elevator adjustment).

3djc, 2025-02-20:
> Outputs already have min & max limits; but at the moment you can't select an 'Output' as the source for an input or mixer line.
> 
> Maybe the simplest solution would be to allow output values to be used as the source for inputs and mixer lines.

Makes a lot of sense

inventor7777, 2025-02-20:
@philmoz maybe I should open an issue later where I can explain what I want in more detail, as my request seems to be a little bit different than what’s described here.

Danitegue, 2025-02-20:
True, for the specific case of the flaps to elevator mix we could use the flaps "channel" as input for the elevator mix. (This is possible via selecting the "channel" of one of the flaps, when adding a line to the elevator mix). But this "channel" does not look like the scaled output, it looks more like the overall flap mix output, which is unlimited and unscaled. 

Therefore I have to apply a custom curve over that channel, to avoid values beyond 100%, when both LS and RS are activated and summed. 

Apart from that, using this as input involves other problems that we are not addressing here, such as the acro mode: in acro mode, the flaps helps to the ailerons for faster rolls. But we don't want the flap to elevator mix to be activated when one of the flaps is up and the other is down, just for helping the ailerons.... so this mix has to be deactivated in ACRO flight mode, and then you should not forget to move to normal flight mode when landing !...

Here, my intention was a bit different, but I can imagine it could be applicable to other use cases: 

Example: rc glider with ailerons + flaps
LS for butterfly -> When only LS is at 100%, flaps should move down to 80% and ailerons up to 50%. This means that, when only LS is at 50%, the flaps would be down 40%. 
RS for flaps -> When only RS is at 100%, the flaps should move down to 100%. This means that, when only RS is at 50%, the flaps would be down 50%.

Case 1: when RS is at 50% + LS is at 50%, I want the flaps to go to the max weighted value between LS and RS, max(40%,50%) = 50%. (And not to the summed value of both 40%+50%=90%)

Case 2: when both RS and LS are at 100%, I want the flaps to go to 100%, with its mix to 100% (not 180%). In this way I can use the overall flap mix value (which is now limited) as input into the elevator mix, without needing a custom curve to limit this added value of 180% down to 100%. 

How could we achieve this without an extremely complex setup based in custom curves, logical switches, and conditions everywhere? Sorry for not remembering each one of the cases in which I have thought that it would be marvelous to have a way to limit the overall mix value. 

I initially tried to create a "virtual input" as a new input with two sources (LS and RS), but it is not working for me since it keeps stuck at 100%. But in any case if this worked, then it would be needed to have a way to adjust the overall input value as the max(LS,RS) which is also an option that I cannot see implemented.

Then I tried to use the mixer, and I was a bit surprised to see how complex it would be to achieve this, in this marvelous swiss knife of the rc radio firmwares!. 

So, I see two possible options to make it easier:

a) Allow "virtual inputs" with multiple sources and selectable logic (min, max, etc)
b) Allow to scale/limit the overall value of a mix, so that we can use it for other mixes.

philmoz, 2025-02-20:
> True, for the specific case of the flaps to elevator mix we could use the flaps "channel" as input for the elevator mix. (This is possible via selecting the "channel" of one of the flaps, when [...]

When you select a 'Channel' as a source it is the Mixer value, not the Output value that gets used.

That is why I suggested adding the ability to select the 'Output' value as a source - this way you get the final value with the limits set in the output line.

inventor7777, 2025-02-20:
I think even if you allow outputs as sources, it would still be a good idea to <b) Allow to scale/limit the overall value of a mix, so that we can use it for other mixes.

I feel like it would be more user friendly and useful for other applications too. Just my two cents :)

pfeerick, 2025-02-20:
Possibly, but one step at a time. Using the outputs as a source should allow for scaling/limiting, since they already have min/max limiting. The question will then be is more needed after that, or is it more a matter of learning how to use the full capabilities of the mixer?

philmoz, 2025-02-20:
I had a look at adding Outputs as sources and it's gonna be messy.

When you select a 'Channel' as a source - you see the 'Output' names; but the code uses the Mix value.

The problem is Mixes don't have their own unique names - they just show the Output names.
So we don't currently have a unique name to differentiate a Mix from an Output.

pfeerick, 2025-02-21:
> Mixes don't have their own unique names - they just show the Output names.

Indeed. And the icons aren't used for them when showing the name, so can't even rely on that to visually differentiate between them (at present, anyway 🤔 ). Maybe that is what is needed? As at present you can differentiate between hardware (axis/pots/switches) and inputs via the icon, even when they have the same as the hardware, but the mixer outputs just have the output name.

Danitegue, 2025-02-21:
Using Outputs as Inputs for a mixer has to be done carefully, since we could end into a closed feedback loop.

In other hand, at the documentation:

https://manual.edgetx.org/bw-radios/model-select/inputs-mixes-and-outputs/mixes

"The Mixes screen is where several Inputs can be combined into one "Channel Mix"."

I would not talk about channels at all in the mixer section. This should be "mix output", or similar, totally decoupled from the outputs section. It is at the output section where I would bind a radio channel to a specific mixer output.

But, if for retro-compatibility issues there is currently no way to decouple these concepts, then I would say that the unique solution would be to add more stuff into the mixer, in order to be able to balance/limit/scale the overall channel mix value. (And possibly also more stuff to control how each line of the mix affects the overall value of the mix, ie more options than just add/multiply)