EdgeTX/edgetx #5936: Radiomaster Zorro - SBUS Trainer - Missing slave option

handcraftedcc, 2025-02-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

While I'm able to enable sbus trainer, there is no option to do serial slave, only serial master.

### Expected Behavior

Have both options to do slave and master over sbus.

### Steps To Reproduce

Enable sbus trainer in settings. Then under the model setup page, under trainer mode you have jack master and slave, but only master serial - no slave serial.

### Version

2.11.0-rc

### Transmitter

RadioMaster Zorro

### Operating System (OS)

Windows

pfeerick, 2025-02-21:
This isn't Zorro specific. Rather, for SBUS student trainer output, you need to go to SYS -> HARDWARE and assign "SBUS Trainer" to AUX1 (or AUX2, if using a handset where that is an option). This should be independent of the model level "trainer" option, and output SBUS on the AUX1 port all the time - but probably a good idea to leave the trainer setting "off" so that it isn't also trying to receive/generate trainer signals somewhere else ;) 

But the fun part for the Zorro is that the  AUX port is located inside the handset... as solder pads. 

[image]

3djc, 2025-02-21:
Yes, you need to setup this specific trainer mode in each model requiring it. Trainer channel come as TRx in the mixer

handcraftedcc, 2025-02-21:
That makes sense. But when wired up I only have the option to select the serial master as an option in the model settings. But what I need is serial slave (which is not available), to be receiving values no?

3djc, 2025-02-21:
Have you: SYS -> HARDWARE and assign "SBUS Trainer" to AUX1 ?

handcraftedcc, 2025-02-21:
Yes. If I disable that setting the serial master disappears as well. But when enabled only the master option shows.

3djc, 2025-02-21:
wait, how can you be serial slave ? Either you are master serial (trainer radio) for the radio receiving the signal through that receiver, or the trainee radio that uses regular signal to that receiver

handcraftedcc, 2025-02-21:
So I'm using a headtracker and have been using the wired slave via the jack with the trainer signal coming from the headtracker into the zorros trainer jack input.

Now I've been working to make this wireless. The headtracker is sending the signal to a second receiver board via Bluetooth that's wired up the aux port inside the zorro. The board is outputting the trainer values via sbus, but while the radio has the setting to OUTPUT trainer signal via sbus (auxiliary master) it is missing the option to receive trainer signal via sbus (auxiliary slave)

handcraftedcc, 2025-02-21:
Oooh. I just realized I got the two confused :D
I thought the setting is what to see the port as, not what to see the signal as. You are in fact right that the setting I need is actually the sbus master. So I guess the issue lies with my setup then, that I wasn't getting an input. Sorry for the confusion.
Thanks for the help!

closed 2025-02-21 by handcraftedcc

pfeerick, 2025-02-22:
lol, yeah, if you are "receiving" SBUS, that means the handset role is master. If you are "transmitting" SBUS, it is slave. In the context of a trainer setup. So in your case, you wanted SBus master, and that is going to come in via the module bay. _Probably_ the rightmost pin for signal (if not, you'll have to find it 🕵), and one to the left of it for GND. Next one along is VBat, so be careful you don't short it to GND ;)