EdgeTX/edgetx #5941: Radiomaster Pocket - Special functions audio keeps playing in a loop

wumfi, 2025-02-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have flashed the latest EdgeTX "Jolly Mon" v2.11.0-RC1 to my radio to try it out.

The flash went ok, but now when I power on my radio it constantly loops all the audio that I have set for my special functions.  Example: Disarmed, Low Rate, Flaps up etc.  It just keeps playing those audo clips over and over.

### Expected Behavior

For it to play the audio once on startup.

### Steps To Reproduce

Power on radio with switches in normal positions.

### Version

2.11.0-rc

### Transmitter

RadioMaster Pocket

### Operating System (OS)

Linux

### OS Version

Mint 22.1

wumfi, 2025-02-23:
Issue fixed.  After the update, it had changed the "Repeat" value for all my special functions to "Repeat 1s".

Using companion, I was able to change these back to "No repeat".

Leaving issue open for now in case this really is a bug.

pfeerick, 2025-02-23:
What version did you update *from*? As this was an expected upgrade issue
(i.e. settings would need updating/correction) when upgrading to an earlier
version (I think it was 2.10?). Meaning this also applies to 2.11 (and
later).

wumfi, 2025-02-23:
V2.10.6.

If it was an expected upgrade issue, I didn’t spot it on the release notes. If it was there, then I apologise for raising it.