EdgeTX/edgetx #5944: Cannot select or receive S.Port telemetry with external module using SBUS protocol

nspector8, 2025-02-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

While using v2.11.0-RC1 firmware with TX16S mkII radio with RFD TXMOD in external bay, "Sub Type" drop down box opens in "External Radio Module" settings page for SBUS protocol, however "SPort" cannot be selected or changed from "No Telemetry".

After powering and connecting FC, RC control is correctly established, radio reports "Telemetry Connected", all sensors are detected and show active (flashing dots) on Telemetry page in Model settings, however no values are presented next to the sensors, and no telemetry is passed through to yaapu.  Additionally, an error page presents itself in the yaapu script, which can be seen below.

### Expected Behavior

Expected behavior is that telemetry values are detected and shown in the radio, and then passed through S.Port to the yaapu script in order to be utilized.

### Steps To Reproduce

1.  Power on TX16S mkII running v2.11.0-RC1 and latest yaapu telemetry script (v2.0.0 beta3), with RFD TXMOD 2.0 in external bay.
2.  Power on aircraft/FC with connected RFD modem (RFD900ux in my case). 

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

This issue has also been discussed and worked on in issue #5064 and fix #5689.

Screenshot of yaapu error:

[image]

keithluneau, 2025-05-11:
Just confirming the same. EdgeTX 2.11.0 Jolly Mon on a Radiomaster TX16S.

raphaelcoeffic, 2025-05-11:
It seems the UI prevented the setting to stick. From what I could find, it *should* actually work.

closed 2025-05-11 by pfeerick

nspector8, 2025-05-12:
@raphaelcoeffic, 

Thank you for working on this for us!  

I'm a novice about fixes or when they are available to use. 

Would this fix be implemented in the next available nightly release or when?

Maybe you can educate me on how to tell if a release includes a particular fix.

Thanks again,

Mike

pfeerick, 2025-05-12:
It will be present in the next nightly (which is the development for 3.0), and will be in 2.11.1 also since the PR is milestones for that version.

keithluneau, 2025-05-12:
I just loaded the nightly from a couple hours ago onto my spare radio. I can now select SPort telemetry while Sbus is selected for the external module. Tomorrow I'll test it with a plane feeding telemetry to the radio and see if everything checks out and report back. Thanks for taking the time to address this!

keithluneau, 2025-05-12:
I've just tested, and telemetry seems to work now, but Yaapu crashes as soon as telemetry starts to flow... lol I think this is a different issue with Yaapu itself not running properly in 2.11, if I load Yaapu Debug script I see telemetry flowing in there. This is the error shown when Yaapu crashes...

[image]

3djc, 2025-05-12:
Clearly looks like an lua 5.3 related error

keithluneau, 2025-05-12:
A quick peak at the lua suggests it's using the bit32 library which is deprecated in lua 5.3

I wouldn't know where to start fixing it, but I'll wait patiently and hope someone capable has the time... lol I see a few people have already reported similar/related issues over at Yaapu.

david9981, 2025-05-13:
Thanks for working on this; s.port telemetry under sbus is finally working again. Why we disabled it in v2.10 anyway?

UI bug still exists however, in settings page for SBUS protocol, "SPort" can be seen but cannnot be selected.

raphaelcoeffic, 2025-05-27:
@nspector8 as long as the sensors are marked active (you can test with a full delete + re-discovery), you're probably hitting another issue. Regarding Yaapu, it's one guy, and I'm not sure how active he is these days. Besides, I have no clue how that RF module communicates with EdgeTx, would you mind explaining?

raphaelcoeffic, 2025-05-27:
> do we have evidence that the external module SPORT port is working fine with custom passthrough telemetry on edgetx 2.11?

@yaapu not sure what you mean *exactly*, but we have several users reporting that S.PORT telemetry works now with SBUS external output, which this issue is about.

keithluneau, 2025-05-27:
> Question: do we have evidence that the external module SPORT port is working fine with custom passthrough telemetry on edgetx 2.11?

I did test mine with Dragonlink, running Mav2PT on the ESP32 sending SPort telemetry out into the radio. This was with 2.11 and your newest build of the script for lua 5.3, it worked find on the ground with all telemetry and messages coming through as best I could tell.

PR #6237 (2025-05-11): fix(color): selecting S.PORT as telemetry for SBUS
Fixes #5944