EdgeTX/edgetx #5951: H750 bootloader cannot jump to the firmware.

Xlio5, 2025-02-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

Referring to the stm32h7s78-dk version, it was ported to the h750, with the main modifications being the change from XSPI Flash to QSPI Flash, and the starting address set to 0x90000000.
After testing the bootloader's jump to the firmware, the firmware did not execute anything.
However, when using Cube MX to generate a program with an offset of 0x90000000, it was able to jump and function properly.
Environment: Chip stm32h750xbh6, Flash w25q256, SDRAM IS42S32400F
code： https://github.com/Xlio5/edgetx/tree/h750_test

### Expected Behavior

Using Cube to generate the app, the normal execution at offset 0x90000000 indicates that the bootloader jump is functioning correctly. With the same bootloader, the firmware should execute normally, but it does not.

### Steps To Reproduce

download boot and firmware

### Version

2.11.0-rc

### Transmitter

Other (Please specify below)

### Anything else?

NORFLASH_START = 0x90000000;
NORFLASH_SIZE = 32M;

MEMORY
{
  FLASH (rx)     : ORIGIN = 0x08000000, LENGTH = 128K
  ITCMRAM (xrw)  : ORIGIN = 0x00000000, LENGTH = 64K
  DTCMRAM (xrw)  : ORIGIN = 0x20000000, LENGTH = 128K
  RAM_D1 (xrw)   : ORIGIN = 0x24000000, LENGTH = 512K
  RAM_D2 (xrw)   : ORIGIN = 0x30000000, LENGTH = 288K
  RAM_D3 (xrw)   : ORIGIN = 0x38000000, LENGTH = 64K
  SDRAM (rwx)    : ORIGIN = SDRAM_START, LENGTH = SDRAM_SIZE
  NORFLASH (rx)  : ORIGIN = NORFLASH_START, LENGTH = NORFLASH_SIZE
}

REGION_ALIAS("REGION_BOOTLOADER",   FLASH);
REGION_ALIAS("REGION_TEXT_STORAGE", NORFLASH);
REGION_ALIAS("REGION_TEXT",         SDRAM);
REGION_ALIAS("REGION_ISR_VECT",     DTCMRAM);
REGION_ALIAS("REGION_DATA",         DTCMRAM);
REGION_ALIAS("REGION_BSS",          RAM_D1);
REGION_ALIAS("REGION_RAM",          RAM_D1);

gagarinlg, 2025-02-27:
Since we have a couple of different boards with h750 that can boot successfully, I suspect an issue on your side.
How did you program the external flash?

Xlio5, 2025-02-27:
Okay, during the bootloader phase, I used the following function to enable the NOR flash
```
  stm32_qspi_nor_init();
  stm32_qspi_nor_memory_mapped();
```
After completing this initialization process, I executed

`jumpTo(0x90000000);`

download firmware, when reading the content of the flash, the offset appears to be normal.

[image]