EdgeTX/edgetx #5953: TBS Agent Lite syntax error on latest nightly build ( Radiomaster GX12)

simoccc, 2025-02-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When launching the TBS Agent Lite Lua script, I encounter the error:

Syntax error TBSagentlite.lua:8 attempt to call a nil value

With stock Radiomaster GX12 EDGETX build it was working , now it's not.

I?ve tried to install it again cleaning directories 
I've tried the old 0.96 version

nothing changes , still got the same errors.

### Expected Behavior

launching the lua without errors 

### Steps To Reproduce

install TBS Agent Lite LUA script fiel and directory TBSAGENTLITE in scripts tools directory as from instructions 

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster GX12

### Operating System (OS)

macOS

### Anything else?

EdgeTX version 3.0.0-nightly 
GIT: a20ea36
DATE: 2025-02-28 02:51:40

pfeerick, 2025-03-01:
Since you are running a nightly (and thus also Lua 5.3), you need to
contact TBS support to get a newer build of TBS Agent Lite, which is
compiled for the newer 2.11 of EdgeTX.

simoccc, 2025-03-01:
thanks for the feedback. I'm asking TBS right away.

I'm not a lua expert, but is this just to be recompiled with a new version of lua ? 5.3 like you said , or is it much more complicated than that ?

3djc, 2025-03-01:
What came with it was a version of 2.11 dated before the time of release, which was still early in 2.11 making. LUA 5.3 was added to 2.11 after GX12 factory firmware was built

pfeerick, 2025-03-01:
> I'm not a lua expert, but is this just to be recompiled with a new version of lua ? 5.3 like you said , or is it much more complicated than that ?

Yeah, it just needs to be recompiled with the newer version of EdgeTX that uses Lua 5.3... AFAIK TBS have already done this, but just haven't listed it for download.

wimalopaan, 2025-03-02:
Version 0.99 is for LUA 5.3. But I think this is still not official. Maybe they only forgot to update the download link.

psfales, 2025-05-05:
I knew at some level (but forget until spent a bunch of time debugging this tonight) that the .zip file TBS Agent Lite (0.99) includes two directories:  one with the 5.2 version and one with the 5.3 version.    You just need to install whichever one is appropriate for your radio.   (EdgeTX-2.11 uses Lua 5.3)

simoccc, 2025-05-17:
Hi and thans for the feedback. TBS has now published TBS agent lite 0.99 and it works !

Just select as @psfales said Lua 5.3 folder if you have EdgeTX 2.11 . 

I close the issue

closed 2025-05-17 by simoccc

bellhawkais, 2025-05-22:
Hi @all

Yes, they forgot to update the link, but in luck, they already uploaded the newer version. You have two choices:

1) Copy the "old" link (https://www.team-blacksheep.com/tbs-agent-lite-098.zip) and change the version number to 099 or

2) just click here: [https://www.team-blacksheep.com/tbs-agent-lite-099.zip](url)

I also needed the 99er version and this one works!