EdgeTX/edgetx #5962: The jumper t15 has a persistent trailing beep sound when playing any audio file that is not in maximum screen backlight brightness

CheZhangLEO, 2025-03-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After setting the change knob to control the brightness of the backlight, the JUMTER T15 plays any audio file (if unlocked) when the screen is not at its maximum brightness. After the playback is complete, a continuous beep will sound for about 1-2s. This phenomenon will disappear when the backlight brightness is at its maximum

### Expected Behavior

Fix this because it happens when I want to save power by lowering the screen brightness and I have to turn on the maximum brightness all the time

### Steps To Reproduce

1. Set the backlight to less than the maximum brightness
2. Set the lever to activate the audio file
3. Move the lever
4. There is a trailing beep after the sound is played

### Version

2.10.5

### Transmitter

Jumper T15

### Operating System (OS)

Windows

pfeerick, 2025-03-04:
So the noise starts after some other audio plays, and goes away after 1-2 seconds of no sound? This sounds like you are hearing the RF/power switching noise before the audio mute kicks in. If you turn Radio settings -> Hardware -> Mute if no sound off (uncheck the toggle), do you hear it all the time now? If you turn the internal RF "OFF", does this go away?

3djc, 2025-03-04:
@pfeerick imho, he is picking up backlight PWM sound, not RF. Anyhow, nothing we can do about that software wise. Try to re-route the audio wire inside the radio

closed 2025-03-08 by 3djc

pfeerick, 2025-03-08:
@CheZhangLEO Have you had a chance to check https://github.com/EdgeTX/edgetx/issues/5962#issuecomment-2696309890 ?