EdgeTX/edgetx #5974: Cannot extract RADIO/radio.yml error on Windows 11 and Companion 211 rc2 when Reading Models on SD Card

fabioeh, 2025-03-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Cannot extract RADIO/radio.yml error on Windows 11 and Companion 211 rc2 when Reading Models on SD Card
Works on OSX.
Same error on multiple radios. 
Reading from SD Path works after copying from SD from radio.

### Expected Behavior

Models successfully loaded into Companion  

### Steps To Reproduce

1. Mount Sd card from radio
2. Click on Load Models

### Version

2.11.0-rc

### Transmitter

Jumper T16

### Operating System (OS)

Windows

### OS Version

OS Name:   Microsoft Windows 11 Home OS Version:    10.0.26100 N/A Build 26100

### Anything else?

[image]

3djc, 2025-03-07:
Your companion is set to the wrong radio model since you say you use Jumper T16 and it is set for Horus X10

pfeerick, 2025-03-07:
Except he should be getting this error if that was the issue - most likely the X10 is one of the other radios this happens with as he said it is happening with multiple radios. 

[image]

Can you go to Settings -> Edit Settings -> Application Settings and turn on Debug logging for Companion. Take note of the file path it will save to, as you'll need to go to that folder later to retrieve the logs. Close Companion, open it again, and attempt to read from the Radio again. Close Companion. If you browse to that folder, you'll have the debug log... please zip up the log file and upload it to the issue (see the prompt at the bottom right of the input text area). 

[image]

pfeerick, 2025-03-07:
Do you have a folder in G:\ called RADIO? Rename that to literally anything
else ;)

fabioeh, 2025-03-08:
No you can see the folders in the screen shot.

The log shows no such folder found.

The G: dive is a Google drive not related to EdgeTx

fabioeh, 2025-03-08:
[image]

G: drive expanded No Radio Folder.

pfeerick, 2025-03-08:
Ok, the two lines above your snippet in conjunction with the next two tell tell the story of what is happening... it looks to be a bug... Companion actually _found_ D:/RADIO , moved onto checking G:/RADIO, didn't find anything interesting there, but then proceeded to using G:/RADIO from then onwards. 

```
[D] findMassstoragePath():314 - Searching for "D://RADIO"
[D] findMassstoragePath():317 - "D://RADIO" found
[D] findMassstoragePath():314 - Searching for "G://RADIO"
[D] readSettingsSDCard():198 - Searching for SD card, found "G:/"
```

@elecpower What do you think of something like this?

```diff
--- a/original.cpp
+++ b/fixed.cpp
@@ -1,7 +1,8 @@
 QString findMassstoragePath(const QString & filename, bool onlyPath, ProgressWidget *progress)
 {
-  QString temppath;
-  QString probefile;
+  QString foundPath;
+  QString foundProbefile;
   int found = 0;
 
   QRegularExpression fstypeRe("^(v?fat|msdos|lifs)", QRegularExpression::CaseInsensitiveOption);  // Linux: "vfat"; macOS: "msdos" or "lifs"; Win: "FAT32"
@@ -10,12 +11,15 @@ QString findMassstoragePath(const QString & filename, bool onlyPath, ProgressWid
     //qDebug() << si.rootPath() << si.name() << si.device() << si.displayName() << si.fileSystemType() << si.isReady() << si.bytesTotal() << si.blockSize();
     if (!si.isReady() || si.isReadOnly() || !QString(si.fileSystemType()).contains(fstypeRe))
       continue;
-    temppath = si.rootPath();
-    probefile = temppath % "/" % filename;
+    
+    QString temppath = si.rootPath();
+    QString probefile = temppath % "/" % filename;
     qDebug() << "Searching for" << probefile;
+    
     if (QFile::exists(probefile)) {
       found++;
+      foundPath = temppath; 
+      foundProbefile = probefile;
       qDebug() << probefile << "found";
     }
   }
@@ -23,7 +27,7 @@ QString findMassstoragePath(const QString & filename, bool onlyPath, ProgressWid
   }
 
   if (found == 1)
-    return onlyPath ? temppath : probefile;
+    return onlyPath ? foundPath : foundProbefile;
   else if (found > 1) {
     QMessageBox::critical(progress, CPN_STR_TTL_ERROR, filename % " " % QCoreApplication::translate("RadioInterface", "found in multiple locations"));
   }
```

elecpower, 2025-03-08:
@pfeerick I broke it you fix it ;-)

PR #5978 (2025-03-09): fix(cpn): sometimes reads wrong SD card path
Due to multiple location detection (#5657) whoopsie. 

Fixes #5974

Summary of changes:
- store found sdcard path/file only if found rather than on each (and possibly invalid) check
closed 2025-03-09 by pfeerick