EdgeTX/edgetx #5975: SD Log symbol missing in top bar

Merga189, 2025-03-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Activating SD Log the usual symbol in top bar does not show up,

Works in companion, but not on the radio.

### Expected Behavior

That the SD Log symbol shows up.

### Steps To Reproduce

But it is a bit more complicated here. Last December I exchanged switch SE from a 3 pos switch to a 2 pos switch (hardware and in software).

Works  correct in companion, SD Log files are generated. But no symbol in top bar on radio.

### Version

2.10.6

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

### Operating System (OS)

Linux

### OS Version

Debian bookworm

### Anything else?

The strange behavior here is, that this 2 pos switch SE shows, when put on a channel, normal behavior with -100% - 100%

When configuring 3 pos switch SG with that function (SD Log) and having set in radio as a 2 pos switch and monitoring SG on a channel, I see this here going through the 3 positions.

SG &#x2B06; -100% no symbol

SG **-** +100% no symbol

SG &#x2B07; +100% symbol on

So using a (hardware) 2 pos switch here looks good on companion (symbol is shown), works in activating the logging (radio) and behaves 'normal' as a channel input. Only on the radio no symbol is showing.

Bit strange, that companion accepts that mod, but the radio obviously not.

br

Merga189, 2025-03-07:
Made a short test with openTX 2.3.15 (Still have a SD card with 0penTX here)

New model, configured switch SE to 2 pos. SF switch SE > SD Log.

Back to model screen. Activate SD Log and symbol in top bar shows up.

So it does not in my eyes seem to be a hardware problem ....

pfeerick, 2025-03-07:
So what do you have the trigger for your SD Logging SF/GF set for when the
symbol doesn't show?  SEdown? And is the SF/GF line highlighting to show
it's active?

Merga189, 2025-03-07:
Thank you for having a look. At the moment I don't get, how to answer your questions, sorry. 

But this might clear things a little.

This are the model settings 

[P-47.pdf](https://github.com/user-attachments/files/19133567/P-47.pdf)

and this is the behavior in companion, I tried to describe

SE logging off

[image]

SE logging on (symbol slowly blinking)

[image]

then over to the radio. (I made one modification: SE on CH 8 to visualize the position of SE!)

SE logging off

[image]

SE logging on (no symbol, **NO*** logging) ***Think I made a mistake in my 1st post here, that logging was performed**

[image]

This is not a problem for me here, can build back to a 3 pol switch or put SD Log on another switch, but as this effect does not show up in companion and also not in openTX, I posted it here.

pfeerick, 2025-03-08:
That sounds more like what I suspected, that logging was not actually happening. i.e. If you go to the special functions page while moving SE from up to down, if the line does not highlight, that special function is not active (thus also the flashing dot to indicate logging is happening has no reason to flash). 

Ok, so you have logging configured via special function, it's enabled, and the trigger is SE(down). Now can you double check on the handset that SE is configured as 2POS, just in case the setting is somehow getting lost. 

One other thing to consider. I can replicate a similar scenario using Simulator, which may or may not explain what is going on. If you start the simulator with the radio in stock settings (i.e. SE is 3POS), and change the switch to 2POS in the simulator, the SEmiddle switch position will act like SEdown as far as the channel output is concerned, but **not** the special function. It isn't until you move the switch fully down to SEdown that it will trigger the SF. Is perhaps your switch not wired up correctly for two POS?

https://github.com/user-attachments/assets/25b37753-6cb5-4afd-9b3a-a9cb0fe7f743

Merga189, 2025-03-08:
> That sounds more like what I suspected, that logging was not actually happening. i.e. If you go to the special functions page while moving SE from up to down, if the line does not highlight, that [...]

Like you say!

> Ok, so you have logging configured via special function, it's enabled, and the trigger is SE(down). Now can you double check on the handset that SE is configured as 2POS, just in case the setting is somehow getting lost.

Checked this several times. And please see the the SIM pictures. There SE is visualized as a 2 pos switch and in the radio screenshots I also put SE on CH 8. 2 pos.

Your video seems corrupt?

> Is perhaps your switch not wired up correctly for two POS?

I ask myself that!! 3 wires, RED, BLACK and GREEN. BLACK on mid contact, RED on side contact, other side contact free and GREEN wire isolated. It works when having SE on a channel. -100% and 100%. SE only on BLACK and RED. Connected or not. GREEN permanently open.  

I don't know, what is wrong with that.

br

Merga189, 2025-03-08:
It is a bit crazy. Played a bit and made a changed here

[image]

and now it works (somehow)

[image]

[image]

In SIM SE in active 100% position NO LOGGING

[image]

SE in default inactive -100% position LOGGING

[image]

and on the radio the same situation

[image]

[image]

That means, SE is working totally  (vice versa) to all the other switches .... I don't get, it doesn't make sense.

br

Merga189, 2025-03-08:
INVERT SE makes it!

[image]

[image]

SE ACTIVE on the radio and logging. Inverting did it, but is that supposed to be the way, when replacing a 3 pos switch with a 2 pos switch? That is the question here ....

But nevertheless, thank you for looking into this ;-)

pfeerick, 2025-03-08:
>Your video seems corrupt?

In what way? If your web browser is having trouble playing it, try downloading it...

[image]

> And please see the the SIM pictures. There SE is visualized as a 2 pos switch and in the radio screenshots I also put SE on CH 8. 2 pos.

This is why I said to deliberately leave SE configured as 3POS when you *start* the simulator, and only change it to 2POS _inside_ the simulator... you then get the 3POS input on the simulator interface... *just like the radio hardware* ;) And in doing so, can then perfectly reproduce what you are seeing happen on the handset.

>  3 wires, RED, BLACK and GREEN. BLACK on mid contact, RED on side contact, other side contact free and GREEN wire isolated. It works when having SE on a channel. -100% and 100%. SE only on BLACK and RED. Connected or not. GREEN permanently open.

Ok, this sounds good... BLACK (mid contact) and RED (side contact). I was more concerned it might have been RED (side) to GREEN (side), and *somehow* half working ;) 

I can also see this on another transmitter on 2.10 (T16), if you simply change a 3POS switch to 2POS, but don't make any electrical changes. channel will go from channel low to high when moved from switch up to switch middle position (an indeterminate state, as it is now neither high or low), but the _perceived_ switch state will still be high. And finally registers as a switch high when switch down position.

pfeerick, 2025-03-08:
@3djc @raphaelcoeffic What do you think. It seems like a really nasty bug to me that the **channel** value would go from low (switch up) to high (switch mid/down) when a 3POS is configured as 2POS, yet the **switch** state as processed by the handset does not reflect that - it requires it to be high or low, and we're only gonna get one of these two states with an actual 2POS switch switch wired there.

Merga189, 2025-03-09:
About the video, you posted. I see this here

[image]

Download is working!

---------------------------------------------------------------

After it seems to be working now here, I was thinking of closing this issue. But after seeing your last post here, I'll keep it open.

If I can be of any help or you need further information, don't hesitate .... I thought, I was possibly doing a thing (by changing hardware a 3 pos to a 2 pos switch), the user is not entitled to do.

br.

PR #6067 (2025-04-11): fix(radio): a 2POS switch connected to a 3POS socket may not behave as expected
Fixes #5975 

If a 2POS switch is wired to a 3POS connector, then treat the mid switch position as equivalent to the down position.
closed 2025-04-11 by pfeerick