EdgeTX/edgetx #5976: RadioMaster TX16S Screen and Audio Hang with Many Mixers

BasculethiTellur, 2025-03-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

My radio will occasionally hang, the screen and audio freezing. This is noticeable as I'm running timers for F5J glider competitions and I'll hear the timer stop. The model is still controllable, and all the surfaces move. Additionally, opening the Mixers page on the Model takes 3-4 seconds.
Here is a RCGroups topic with a few of us commenting about this:
https://www.rcgroups.com/forums/showthread.php?3493859-RadioMaster-TX16S-16Ch-2-4ghz-Multi-Protocol-OpenTX-Radio-System/page1091#post53553273

Here are the steps I've taken to try to isolate the issue:
Used a brand new SD card and installed the clean EdgeTX SD card contents. Then I copied over my models, scripts, model images, my custom glider sounds.
Upgraded through several EdgeTx revisions/versions.
I'm running ExpressLRS on Ranger Micro with the latest firmware upgrades to both the Tx and Rx.
I've tried removing the ribbon cables, cleaning with IPA, and replacing them.
I'm flying with a fairly complex template with a lot of curves, logical switches, global variables, etc. This is from the F5J templates on rc-soar.com.
I've not been able to fix it, and the problem usually occurs in a few hours of flying.
I did replace the onboard battery, the CR1220.
Changed to my spare TX16S MKII transmitter with a different Ranger backpack.
Disabled the ShowItAll widget.

None of these tests fixed the problem, and someone else also has hanging problems. I don't get any error on the radio screen, just the audio hangs, and after the audio comes back, the timer tries to count down fast to catch back up.

To me it feels like the radio CPU is overloaded and gets stuck. I can't help wondering if the complex F5J template I'm using is too much for the radio?

### Expected Behavior

For timers etc not to freeze.

### Steps To Reproduce

RadioMaster TX16S MKII running Edge 2.10.6
Install my SD card from link below - select models Eternity B or Eternity C or Eternity Light
Try opening Mixers on the Model page - takes 3-4 seconds to open.
The problem with freezing audio can take a few hours of flight to turn up, and the sound including timers and special function callouts freezes and no audio is produced for a period of time lasting roughly between 5-60 seconds.

### Version

2.10.6

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

A zipped copy of my SD card is here:

https://drive.google.com/file/d/1CDhex4t9lQnqZD3zQr08_xsiWDeDldtv/view?usp=sharing

3djc, 2025-03-07:
What are your elrs settings for that model ?

BasculethiTellur, 2025-03-11:
[image]

[image]

I caught the bug happening on video here: https://drive.google.com/file/d/17djjQx3TVsvWlX07fQ9y7DziQCaT6mX-/view?usp=sharing

RC-SOAR, 2025-04-03:
I've managed to replicate the "switch hang" issue on my TX16S running ETX 2.10.6, running the basic template. Though in my case it happened a few moments after switch on.

BasculethiTellur, 2025-04-03:
I've just been away in Larissa Greece for 10 days of full time practice and competition F5J gliding, and had the issue multiple times on both of my TX16S MKII radios. Sometimes it was a partial loss of sound, so a timer might reach 6m34s, but I'd hear "minutes seconds" and no numbers. Mostly it was the classic audio and screen hang as I've described at the start of this issue.
At least I came 2nd in the first EuroTour competition of the season. I can work around the sound/timer issue as I can ask the person coaching me what time remains, or look at the timer board myself, but it's not ideal as it adds workload when I should keep my mind free for thermalling and tactics.

BasculethiTellur, 2025-05-10:
I've upgraded one of my radios to 2.11.0 and the issue is now far worse, to the extent that when powering on the radio the welcome wave file: "Welcome to EdgeTx" hangs partway through and the radio is locked for a while before the boot eventually completes. It's really bad.

raphaelcoeffic, 2025-05-15:
> Try opening Mixers on the Model page - takes 3-4 seconds to open.

This is more or less known and yes, due to the complexity of the model. We probably need to come up with a way to make this more direct.

> The problem with freezing audio can take a few hours of flight to turn up, and the sound including timers and special function callouts freezes and no audio is produced for a period of time lasting roughly between 5-60 seconds.

Ok, so while flying, everything is behaving normally, and on a few occasions audio and callouts will just stop working properly for up to one minute. Do I get that right?

When you write "can take a few hours to turn up", that means it will work normally for multiple hours, and then happen, right? Anything special right before?

> upgraded one of my radios to 2.11.0 and the issue is now far worse, to the extent that when powering on the radio the welcome wave file: "Welcome to EdgeTx" hangs partway through and the radio is locked for a while before the boot eventually completes.

Ok, does this happen on every boot with 2.11.0? Once boot completed, do things work normally?

raphaelcoeffic, 2025-05-15:
> I've managed to replicate the "switch hang" issue on my TX16S running ETX 2.10.6, running the basic template. Though in my case it happened a few moments after switch on.

What is the "switch hang" issue?

RC-SOAR, 2025-05-15:
> > I've managed to replicate the "switch hang" issue on my TX16S running ETX 2.10.6, running the basic template. Though in my case it happened a few moments after switch on.
> 
> What is the "switch hang" issue?

It's been a while now, but as I recall shortly after switch on, sounds linked to flight modes were not sounding as I switched between fm's. It was as if the UI had frozen. It must have lasted around 5 seconds, then back to normal. This was without a model.

3djc, 2025-05-15:
> I've upgraded one of my radios to 2.11.0 and the issue is now far worse, to the extent that when powering on the radio the welcome wave file: "Welcome to EdgeTx" hangs partway through and the radio is locked for a while before the boot eventually completes. It's really bad.

Then I would suspect your SD card. Here is your SD content on a tx16s flashed with 2.11 (using buddy)

[IMG_0461.M4V.zip](https://github.com/user-attachments/files/20227689/IMG_0461.M4V.zip)

RC-SOAR, 2025-05-15:
In my case using a freshly installed ETX 2.10.6. Sandisk Ultra card.
Is there any mechanism other than the physical SD card for this kind of UI freeze?

3djc, 2025-05-25:
@RC-SOAR @BasculethiTellur 

We introduced recently a new debugging tool to our arsenal, and I had a look at the model shared. I noticed that the radio never idles, and actually spend most of its time in the menu task, dealing with display. So I created a second empty screen, and when switching to this screen, everything looks much nicer system usage wise. Could you try something similar ? maybe switching to a nearly empty screen when you arm or something. It would be very interesting to know if this fix your issue

BasculethiTellur, 2025-05-25:
Sure, I’ll try that. Do you mean the widget/display screen? I should make a second telemetry screen that is blank and use this whilst flying to reduce CPU load and see if I can generate the bug condition?Sent from my iPhone

3djc, 2025-05-25:
Yes exactly 👍

3djc, 2025-06-04:
Ok, I left yesterday afternoon the radio using your SD content and adding a SF that read time every 10secs. Impossible to replicate.  Almost drove me mad listening to clock announcement.

Then this morning I decided to copy the very same content on a crap SD, and boom, go it right away, and really bad.

So it is SD card related, which is not a good news

3djc, 2025-06-04:
Formating is easy Fat32 (NOT exFat). As for the card itself, it's a real pain as there are so many fakes around, even when purchased for places like Amazon or such. From the feedback we are getting, I would avoid any 'endurance', or heavy usage focused card (we are actually very gentle on write usage) as those include a lot of anti wear techniques that sometimes interfere with our usage

I used an external card reader on my Mac to copy your full sd content, and on the 'bad' card it took several minutes, and on the good one 25s about.

Here are the two ones I used:

[image]

Right came with a radio (don't have any idea which brand or model radio that was), the other was purchased from Amazon. But the difference is obvious using those, and the right one exhibit it nearly instant, when I was unable to replicate it with left one.

We will try to see if we can gain some understand of what is happening with the 'bad' card

BasculethiTellur, 2025-06-04:
OK, thank. I've ordered a couple of different brands, including the one you've used above. Let's see. I will point out that this is an intermittent error, so it may well not be the SD card and it may be that you just didn't run the test for long enough. In the flying field I can sometimes be flying for a couple of hours before the problem shows up.

WimHWimH, 2025-06-12:
> For info, there are 9 custom audio files in E-Soar Plus, all are 16KHz, 16 bits, 256 kb/s, 1 channel, PCM. They vary in size between 13 KB and 46 KB

I'm using a modidfied Esoar Plus F5J template on my Radiomaster Pocket  with an extra audio file (10s countdown) that is rather large (> 500kb) and have not had any such problems so far.

Arthur-de-Partuur, 2025-06-12:
As the Bug is for me not present in the 2.10.6 (I downgraded) but is present in 2.11.1, I assume it is an ETX bug. @BasculethiTellur can you confirm that for you the bug disapears if you downgrade to 2.10.6?

Also, an earlier version of eSoar plus with an F5J model in it (and 6 wing servo's) does not suffer from this in either version.

3djc, 2025-06-22:
What happens is that you hit a 5 seconds timeout delay on low level SD routines provided by STM. This freezes the UI and audio tasks for 5 seconds (mixer is totally unaffected), then recovers from it. It is 100% SD related, is not directly related to the complexity of the model (as in mixer lines), but obviously is scalling with the number of sd accesses (sounds, logs,...)

While we know exactly what happens, it is a very different story to understand why it happens, since it happens very infrequently, and been able to log this type of event is a bit beyond the capabilities of the equipment we have and also the ressources we have.

3djc, 2025-09-21:
Sorry if language is making things more difficult. I was not implying the sd card is at fault here (it can happen too, but that's another thing altogether), just that this issue happens during sd access using low level code we take from stm. We could reduce this timeout, but I'm a bit reluctant to it since it is standard stm code, it would probably be wiser to understand why it happens. I'm also curious to know if it happens with new gen hardware

Arthur-de-Partuur, 2025-09-22:
> Sorry if language is making things more difficult. I was not implying the sd card is at fault here (it can happen too, but that's another thing altogether),

Mmm, indeed I totaly misunderstood that. Thanks for clearing that up!

> just that this issue happens during sd access using low level code we take from stm. We could reduce this timeout, but I'm a bit reluctant to it since it is standard stm code, it would probably be wiser to understand why it happens. 

I Can't judge your time alocation or workload so for me it is unclear if this could be worth investigating. I am no software developer so am strugling a bit with the lingo (STM?), other than some arduino stuff and scripting, I must be considerred a noob. HOWEVER,. if there is something I can do to shed more light on the matter, like testing, running special versions with logs enabled, etc... Let me know. Happy to pitch in where needed.

>I'm also curious to know if it happens with new gen hardware

as in "newer than TX16s MKII,. I would assume....? I am a somewhat oldfashened dude and realy like sticking to the TX16s formfactor,... The newer radio's might have better processors and more/faster RAM, I would assume

Arthur-de-Partuur, 2025-09-26:
@RC-SOAR Still pondering why my older F5J E-Soar (with 6 servo's wing) is not hanging up and my V3 E-Soar (with 4 Servo's wing) is hangin up like crazy,... any thoughts why that could be (i.e. what are the major differences between V2 and V3)?

Arthur-de-Partuur, 2025-09-30:
> Do you have major differences in audio announcements ?

Hi, Sorry for the late reply (work got in the way of pleasure,... again...)
I do not see any large differences in audio announcements. I have been switching models for about half an hour now and I have only encountered the bug once, whereby a couple of days ago I had them basically every time I choose the model from the model menu...

I made a compare with Companion but  I do not realy know what to look for... (Evoa 3 is "hanging" most often and part of Esoar 3, Inside F5j is an older version of Esoar and is performing better)
[F5J vs EVOA3 compare.pdf](https://github.com/user-attachments/files/22625033/F5J.vs.EVOA3.compare.pdf)

Arthur-de-Partuur, 2025-10-16:
> I've moved two of my gliders to my TX15 Max, and this seems to have fixed the problem. The faster processor FTW.

With all due respect, that did not fix the problem... that avoided the problem. Trust me,... it's still here...

BasculethiTellur, 2025-10-19:
> > I've moved two of my gliders to my TX15 Max, and this seems to have fixed the problem. The faster processor FTW.
> 
> With all due respect, that did not fix the problem... that avoided the problem. Trust me,... it's still here...

Given that the problem was caused by the much heavier CPU load of EdgeTx and extra touchscreen code etc, vs the lighter weight OpenTx, getting a faster processor has exactly fixed the problem.

BasculethiTellur, 2025-10-19:
> > > I've moved two of my gliders to my TX15 Max, and this seems to have fixed the problem. The faster processor FTW.
> > 
> > 
> > With all due respect, that did not fix the problem... that avoided the problem. Trust me,... it's still here...
> 
> So just to be clear, the problem is still there with your TX15? Can you describe the symptom?

To be clear I think the problem is fixed as the radio now has more resources in terms of CPU speed and SD card reading speed. I have not seen any example of the sound or screen hang occurring based on a couple of days of flying F5J with my modified timers. I also ran the radio on my desk with a ten second repeated timer callout like I did before, and did not encounter the problem either.

I've even gone further with some additional logical switches and special functions to turn buttons 1-3 into momentary switches to call out some telemetry that's important to me, and to modify the main flight timer so that I get time announced every minute, and then also every 30s as well from 07:30 down to 01:30 remaining, and then every 5 seconds until 30s remaining, and then the normal 30s countdown. That's the sort of thing we normally get our coach to call out when flying a competition round. I've also added a zoom timer that starts from motor off or end of the 30s motor timer.

@Arthur-de-Partuur is the one claiming that moving to TX15 did not fix the problem above. IMO the problem was insufficient resources to run the much more complex EdgeTX OS (more than OpenTx), your template with a lot of mixers and other programming, and some sort of possible SD card reading inefficiency.

philmoz, 2025-10-19:
Could anyone having this issue on a TX16S please test the attached firmware, and report if it is better, worse, or unchanged.

It is recommended to:
- check the SD card for errors and fix if needed
- have a full backup of your SD card just in case

[tx16s_firmware.bin.zip](https://github.com/user-attachments/files/22992920/tx16s_firmware.bin.zip)

philmoz, 2025-10-20:
> I've had three freezes using your firmware.bin, [@philmoz](https://github.com/philmoz). Do you want me to test further?

Thanks for testing. If it's still happening, then no need to test further.

BasculethiTellur, 2025-10-20:
> Thanks for testing. If it's still happening, then no need to test further.

I appreciate your efforts, and those of all the other developers very much. Thank you all!

It's a testament to the excellence of EdgeTx that more and more higher level F5J pilots are using it.

J-Sorenson, 2025-12-09:
Does this happen when you switch models as well, or only on cold power-on?

lphaf, 2025-12-09:
@Arthur-de-Partuur, I am wondering if you see the same behavior I did: If you set the checklist so that it makes you end up in thermal mode, the startup (or switch to the model) is quick.

Arthur-de-Partuur, 2025-12-11:
Maybe @RC-SOAR can think along what could caus this? an earlier version of Isoar that shows much less delay, uses the A-switch directly in the calibration menu, like so:

[image]

in the newer version, (that has the issue on my radio, It is done via three logical switches like so:

[image]
and then:

[image]

No clue why and I lack the time to check, but maybee this is causing this?

Arthur-de-Partuur, 2025-12-11:
Thanx for your answer, @RC-SOAR, but I have not yet understood why an earlier version of this emplate works signnificanly better. This way, I might be able to tweak my plane-profile such that it works better.
The flightmode for thermal are independent of any switch:

[image]

[image]

[image]

That would account for it "Not havving the issue" in thermal mode, right? As such, doesn't need to check all the logical switches?
I might make a copy of my profile and "remove" the assigment to L40, L41 and L42 (basically circumnavigating the I13: S3P option) in the Flightmodes.... see where that brings me...

RC-SOAR, 2025-12-11:
Just loaded the SD card image from @Arthur-de-Partuur. And sure enough, it hung on the Pre-start Checks screen - the tx was unresponsive to touch and key inputs for about 4 minutes. Tried again just now, and the same again. 
It's quite bizaree how it hangs for so long and then recovers... unfortunately it makes it all but unusable.
This is the same hardware (Eachine TX16S) as in my previous post, where it worked fine (albeit with limited testing and using a different ETX version). 
I'll do some further investigation.
Running ETX 2.11.3

RC-SOAR, 2025-12-11:
I think I may have made a very basic mistake during the tests - I thought the Rtn key would escape from the Startup check screen, but I now realise you have to hit Enter.

So I'm going to start the tests again from the beginning.

RC-SOAR, 2025-12-11:
Renamed \SOUNDS folder to \xSOUNDS, and it has failed again, hanging on the last screen of the Pre-Start checks screen. So the presence of sound files does not seem to be a factor.

Arthur-de-Partuur, 2025-12-15:
I did some further test. Not sure if anything is relevant to the issue:
I tried with ELRS on 250Hz and the first instance it froze for about 2 minutes, so assume that it is not solved with that (would be 4ms interval, right?)
Also tried with ELRS on 150Hz (6.66 ms) and at first boot there was no freeze but the second boot it froze up for 2 seconds or so. Subsequent boot-ups show a minor hick-ups but all in all very usable (booted 5 times). I kept the radio off for about 2 minutes and retried and sixth bootup froze again for 1.5 minutes. No idea if the 2 minutes delay in switch off has anything to do with it.

I have now switched off the internal ELRS module in the model settings and have booted several times (6 with varying pauses in between) with no hick-ups/freezes etc.

Is there something in the code that has changed between 2.10 and 2.11 in this regard, as the problem was a lot less present in 2.10?

3djc, 2025-12-15:
It seems we are getting longer mixer times while checklist is displayed, making it into longer than admissible with your settings, and once that happens, there is little way out.

I'm investigating why we have those longer mixer runs

3djc, 2025-12-16:
@Iphaf, I'm currently focusing on @Arthur-de-Partuur settings since I can replicate and we have identified a true issue.

First outcome, I have found an issue in the mixer time displayed, it currently shows only HALF the real value. A PR has been issued for that.

There is another issue when mixer time exceeds the pulse period, but @philmoz had a great idea to mitigate that, still checking it.

Arthur-de-Partuur, 2025-12-16:
> could you try this firmware ?

Done: 

- Flashed firmware,
- almost booted,... read your message better, 
- Yanked out the SD card to make backup, 
- put SD card in but missed the actual SD card slot (minor design flaw in radio),
- opened up the radio (again)
- found SD card (again), and inserted it into slot (again and before fully closing the case),
- Booted fine in 150Hz (still, from previous trials)
- switched to 333Hz and booted 5 or 6 times. All fine so appears to be massive improvement!!!!!!
- I'll keep on trying and report if anything fails but for now this looks awesome.

You and @RC-SOAR showed a debug screen before... Is that something I should be looking for as well (couldn't find it...)

RC-SOAR, 2025-12-17:
Hmmm.... just to add to this, I thought it might be due to the non-existent module. So I fitted my external Ranger module and configured Arthur's setup accordingly. To maximise the period, I set 100 Hz/Full. 

**Model:** evoa-3 configured for external RF
**TMax:** 1.09
**Period:** 10 mS 

Unfortunately it is is still prone to hanging, occasionally.

I won't be able to do any testing till tomorrow afternoon, but will be happy to check out the next iteration. Much appreciate your looking into this.

3djc, 2025-12-17:
That's probably 2 different issues. The issue at start was clearly a cpu ressource starvation. But that should not occur anymore

Arthur-de-Partuur, 2025-12-17:
Mmm, I formated the SD card and put the backup back on. Booted, and flashed the lates from @3djc 
Also added these steps (just to be thourough):
- put SD card in but missed the actual SD card slot (minor design flaw in radio),
- opened up the radio (again)
- found SD card (again), and inserted it into slot (again and before fully closing the case)

After restart , the radio froze on the checklist. I was not able to check the boxes. No audio came out and radio did not recover.
Pulled the battery and restarted.
Startup sound came through and I was able to check the first checkmark after which it fully frooze again...
No luck with this one for me...

Any thing I can do to further test or research the issue (if I can fnd a way to switch the radio on in another plane-profile)?

3djc, 2025-12-17:
It this version, it will reduce rate based on tmixmax, so matter what elrs settings is set at, if it really has to.

Arthur-de-Partuur, 2025-12-17:
Yes, I see. When I set ELRS to 500Hz, the period shows 2ms and sometims goes to 3ms. Presumably because it needs some more time? I coincides with Tmix max jumping up from 1.66 to 2.14 ms

RC-SOAR, 2025-12-18:
I have just resumed testing with the latest self build.

- using evoa-3
- modified to use external Ranger module
- entered UI Setup menu
- Removed screens (in order to assign alternative widgets)
- Pressed the Screen1 (to assign a widget)

At this point ETX becomes unresponsive to touch and key presses.  The only way to recover is to disconnect the battery

The issue is consistently repeatable.

Video here:
https://share.icloud.com/photos/0ff6LuJ_PGX7ed_j4UU-9uEBw

**O/S:** 3.0.0 self build (6b75d518)
**RF:**  ELRS 100Hz 8CH/Full
**Stats:** Tmix max 1.11 mS, Period 4 mS

RC-SOAR, 2025-12-18:
Yes, 10 secs on the power button shuts down the tx successfully.

philmoz, 2025-12-18:
> I have just resumed testing with the latest self build.
> 
> * using evoa-3
> * modified to use external Ranger module
> * entered UI Setup menu
> * Removed screens (in order to assign alternative widgets)
> * Pressed the Screen1 (to assign a widget)
> 
> At this point ETX becomes unresponsive to touch and key presses. The only way to recover is to disconnect the battery
> 
> The issue is consistently repeatable.
> 
> Video here: https://share.icloud.com/photos/0ff6LuJ_PGX7ed_j4UU-9uEBw
> 
> **O/S:** 3.0.0 self build ([6b75d51](https://github.com/EdgeTX/edgetx/commit/6b75d51867e0fb9298ec02273ee183c42e5eedae)) **RF:** ELRS 100Hz 8CH/Full **Stats:** Tmix max 1.11 mS, Period 4 mS

Looks like a corrupted model file - can you upload just that model yaml file.

philmoz, 2025-12-19:
> > Looks like a corrupted model file - can you upload just that model yaml file.
> 
> Yes, deleting screens seems to have put the UI into a bad state.
> 
> Attached zip file with YAML model file, radio file and widgets.
> 
> [UI Setup hang.zip](https://github.com/user-attachments/files/24254689/UI.Setup.hang.zip)

Missing first entry (#0) in screenData section. You could fix it manually by renumbering the screens (1 -> 0, and 2 -> 1).

Question is how did you manage to do this?

RC-SOAR, 2025-12-19:
> Question is how did you manage to do this?

I was deleting the screens one by one, with a view to loading my own choice of widgets. I think - but am not sure - that I was allowed to delete screen 1 (the last screen remaining), and it was when I re-pressed the Screen1 button that the problem started.

3djc, 2025-12-19:
Btw, for user info, the same settings that run at 2.1ms on TX16S class hardware runs bellow 0.7ms on H750. Currently there is not a lot of choice in Etx h750 radios, but that's meant to improve !

RC-SOAR, 2025-12-26:
> I noticed that the startup (or model switch) delay/hang goes away (mostly) if I remove the fade-in and fade-out on FM2 and FM3. This works when starting in flight mode speed, cruise and thermal. I also get a lower (and constant) Tmix max of 0.65 ms.

Thanks for looking in such detail. Yes, I can confirm this, using @Arthur-de-Partuur setup with 2.11.3. It seems to improve symptoms but as you say is not a full cure as it can still hang occasionally on startup. 

The [most recent self build](https://github.com/EdgeTX/edgetx/issues/5976#issuecomment-3664794947) is a big improvement, at least in terms of startup behaviour.

In the meantime, I've suggested some measures for users of E-Soar Plus and ETX 2.10 and 2.11: 
https://rc-soar.com/edgetx/bugs/slow_startup.php

RC-SOAR, 2026-01-05:
@Arthur-de-Partuur I don't want to push people towards a new radio if it's going to be fixed in ETX. In the meantime I've added some info in my [support page](https://rc-soar.com/edgetx/bugs/slow_startup.php).

3djc, 2026-03-06:
It doesn't have a F7, but a H7. Unfortunately, I don't have a cristal ball, but those in this thread having similar issues have seen those disappear when moving either to tx15 or tx16smk3, so there is a good chance it will do the same for you

BasculethiTellur, 2026-03-10:
Seconded on the MK3 - it's amazing. I already sold both my MK2s last year, and I've a second MK3 on the way, as well as my current spare, a TX15 Max.
As for this issue, for me it is resolved/sidestepped. Some combination of the faster processor and fast on-mobo storage means that callouts are working perfectly. I've done around 25 hours of flying this year alone on a combo of TX15 Max and TX16 MK3 and not heard the problem even once.

pfeerick, 2026-03-10:
> you won't be able to use your mk2 anymore

I'm afraid I have to disagree there (for more common non-complex setups)... my F4 gear still gets as much, if not more use, than H7. Just because H7 is the new kid on the block does not automatically equate better or that the older gear is unusable just because it is a little slower. Don't get me wrong, I like my H7 gear also, but the old stuff still works just as well as it always did (it didn't get any slower, after all, rather expectations increased unreasonably).

K121964, 2026-03-25:
Has anyone tried with ELRS V4.0. I have noted a new freeze since updating to 4.0 (including the new LUA script).
If you try and change receiver number in External Module screen, the TX locks up when it reaches certain Rx numbers (eg 22 and 38) very odd !! The other thing that happens is the baud rate status then starts behaving very strangely jumping to 500 then 1000 and 200 (I have external module baud rate set  to 921) and rx set to 100.
I also experienced the ExpressLRS Lua freezing and the tx also when changing bands from 868 to 2.4.

RC-SOAR, 2026-08-24:
Can't say for sure, as not everyone is affected. 
Most of the reports concern one template in particular, E-Soar Plus, and there are mitigation measures that have been shown to work, see

https://rc-soar.com/edgetx/bugs/slow_startup.php

3djc, 2026-08-24:
> Hi All,
> 
> Out of interest , has anyone tried these templates on RM GX12 ? If so is the problem also on the GX12 ?
> I note the GX12 has on board memory rather than SD card and was wondering if that made a difference.

Likely unaffected because ui task are much lighter on BW. Good practice like avoid endless speech at start and use sensible elrs packet rate are in order for super complex models