EdgeTX/edgetx #5980: lvgl can not use OO style api on widgets

offer-shmuely, 2025-03-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

the following code is working in one-time script, but fail in widgets

```lua
    lvgl.clear();

    local box = lvgl.box({x=100, y=10})
    box:rectangle({x=0, y=0, w=100, h=100, color=YELLOW, filled=false, rounded=8, thickness=4})
    box:rectangle({x=10, y=10, w=100, h=100, color=YELLOW, filled=false, rounded=8, thickness=4})

```

[image]

### Expected Behavior

I expect the we can use OO style on widgets as weel
``` 
box:rectangle()
```

### Steps To Reproduce

use this widget
```lua
local app_name="t_lvglCtx4"

local options={}
local function create(zone, options)
    local wgt={zone=zone, options=options}
    return wgt
end

local function update(wgt, options)
    wgt.options=options
    lvgl.clear();
    local box = lvgl.box({x=100, y=100})
    box:rectangle({x=0, y=0, w=100, h=100, color=ORANGE, filled=false, rounded=8, thickness=4})
end

local function background(wgt)
end

local function refresh(wgt)
end

return {name=app_name,create=create,options=options,update=update,refresh=refresh,background=background,useLvgl=true}

```

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

win11

philmoz, 2025-03-10:
Fixed in #5926

offer-shmuely, 2025-08-01:
@pfeerick 
This issue has been fixed
Can you close the issue?

closed 2026-03-14 by offer-shmuely