EdgeTX/edgetx #5981: lvgl children controls are not using the box (parent) coordinates

offer-shmuely, 2025-03-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

the following line of code print the rectangle in 0,0 instead of 200,200
```lua
    lvgl.clear();
    local box2 = lvgl.box({x=200, y=200})
    lvgl.rectangle(box2, {x=0, y=0, w=30, h=30, color=BLUE, filled=false, rounded=8, thickness=4})
```

[image]

### Expected Behavior

I expect it the rectangle coordinate to be relative to the coordinate of box (the parent)

[image]

### Steps To Reproduce

use this widget
```lua
local app_name="t_lvglCtx4"

local options={}
local function create(zone, options)
    local wgt={zone=zone, options=options}
    return wgt
end

local function update(wgt, options)
    wgt.options=options
    lvgl.clear();
    local box2 = lvgl.box({x=200, y=200})
    lvgl.rectangle(box2, {x=0, y=0, w=30, h=30, color=BLUE, filled=false, rounded=8, thickness=4})
end

local function background(wgt)
end

local function refresh(wgt)
end

return {name=app_name,create=create,options=options,update=update,refresh=refresh,background=background,useLvgl=true}

```

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

philmoz, 2025-03-10:
Fixed in #5926

offer-shmuely, 2025-03-27:
work perfectly now (2.11-rc3)

closed 2025-03-27 by offer-shmuely