EdgeTX/edgetx #5983: X7Access radio (ISRM D16 Mode, v2.11.0-rc2) does not connect to receivers

bug400, 2025-03-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Radio does not connect to receivers (ISRM D 16 Mode) on Taranis X7Access with v2.11.0-RC2 firmware from prerelease assets.

### Expected Behavior

Connect to receivers.

### Steps To Reproduce

A 2.10.5 etx file opened with Simulator 2.11. Transferred radio and model settings to radio. Save etx file under a different file name. Flash 2.11.0-rc2 firmware from prerelease assets using boot loader. Power on radio, select model -> no connect to receiver. Tested with two models.

A 2.10.5 etx file opened with Simulator 2.10.5. Transferred radio and model settings to radio. Flash a 2.10.5 firmware using boot loader. Power on radio, select model -> connects. Tested with the same models as above.

### Version

2.11.0-rc

### Transmitter

FrSky Q X7 / Q X7S (ACCESS)

### Operating System (OS)

Linux

### OS Version

Debian bookworm

bug400, 2025-04-22:
Still exists in rc4 with radio firmware from rc4 release assets. Receiver and model data were written with rc4 companion. It seems, that the ISRM module is completely dead with this release. Everything is o.k. if I flash 10.5 again. I  find it astonishing that nobody cares about such a problem.

pfeerick, 2025-04-23:
>  It seems, that the ISRM module is completely dead with this release.

I can only comment for the X9D+2019 (thus also ACCESS/ISRM based) but that is working for me. Is this actually "dead"/non-functional, or does it still show module status/version in `SYS -> VERSION -> Modules / RX Version` (meaning that the module isn't "dead" and is still talking to the handset) and does anything show when you go to MDL -> Internal RF -> Options [Set]`? (mine says Waiting for TX and then No options, but I have had a R9M module give me options there). 

Have you tried rebinding to the RXs at all... i.e. could it be simply that it has lost the bind/rxid?

We only have a handful of FrSky handsets, and no support from FrSky, so will only be able to identify the fault if someone with a handset can help debug what is actually going on, what change broke it.

bug400, 2025-04-23:
SYS->VERSION->MODULES/RX Version says : ---
MDL->Internal RF-> Opion says: Waiting for TX... (indefinitely)
I can gladly help with my radio. Building custom radio firmware is no problem for me. But how can I get edgetx logging output from the radio? If ISRM works with the X9D+2018 it seems that the hardware (power, communication) is misconfigured for the QX7/Access radio.

bug400, 2025-04-25:
Commit identified:  https://github.com/EdgeTX/edgetx/commit/6579905088dc8f3828140978e5a5a79612e4f255 (remove StdPeriph). Previous commit shows transmitter firmware version correctly. Any suggestions how to proceed?

raphaelcoeffic, 2025-04-25:
> Commit identified: [6579905](https://github.com/EdgeTX/edgetx/commit/6579905088dc8f3828140978e5a5a79612e4f255) (remove StdPeriph). Previous commit shows transmitter firmware version correctly. Any suggestions how to proceed?

This only touches hardware layers, and *should* be equivalent as to how it connects to the internal module (simple bidirectional UART). Also, AFAIK, the X9D+2019 is no different from the X7Access in any significant way when it comes to the internal module.

Does it actually work in ACCESS mode?

bug400, 2025-04-25:
It is **not** equivalent! There is an obvious mistake in radio/src/targets/taranis/hal.h on line 2073:

#define INTMODULE_PWR_GPIO               GPIO_PIN(GPIOA, 7) // PC.05

Changing to:
#define INTMODULE_PWR_GPIO               GPIO_PIN(GPIOC, 5) // PC.05

the ISRM module now gets power and works.

raphaelcoeffic, 2025-04-25:
> It is not equivalent! There is an obvious mistake in radio/src/targets/taranis/hal.h on line 2073:

Good catch! That is about the only type of thing I could think of!

bug400, 2025-04-25:
Confirmed. ISRM works with your pr.

PR #6147 (2025-04-25): fix(x7access): internal module power
Fixes #5983