EdgeTX/edgetx #5988: move up input in companion is broken

ProfDrYoMan, 2025-03-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Using move up input (not line) is not working in companion.

### Expected Behavior

The input is moved up.

### Steps To Reproduce

Use the "move up input" function on an input.

### Version

2.10.6

### Transmitter

Other (Please specify below)

pfeerick, 2025-03-15:
Knowing the operating system is quite important in this case, since you are talking about Companion, and there may be something OS specific happening here. I cannot reproduce this with either Companion 2.10.5 or 2.10.6 on Linux - using Lines -> Move Up and Move Down, I can move both single and multi-line inputs up and down without issue - i.e. all the elements of a multi-line input move up and down, and a single-line input also moves up and down correct. So what do you mean when you say "Using move up input (not line) is not working in companion."?

ProfDrYoMan, 2025-03-15:
It's Windows 10/11. Sorry missed that.

It seems the bug is only when the input above (where the move goes to) is empty.

Moving lines up and down (even into empty space) seems to work.
Moving inputs up into empty space is bugged. Moving inputs down into empty space seems to work.

I made a screen recording. You see the error happening at the end.

https://github.com/user-attachments/assets/b0bd2546-853c-4f82-8497-a072601be8e1

elecpower, 2025-03-15:
Thanks for the video that assists greatly to understand the steps to reproduce

elecpower, 2025-03-15:
Confirmed in Ubuntu. Move input down okay. Move input up *broken*.
Move lines up and down okay.

PR #5994 (2025-03-18): fix(cpn): moving input line up to an empty line resulted in input data loss
Fixes #5988

Summary of changes:
- new sort inputs function (based on sort mixes)
- replace complex shuffles with simple reference swap and sort (much lower risk)
- move down uses same method
- minor housekeeping
closed 2025-03-18 by pfeerick