EdgeTX/edgetx #5989: Using drag and drop on inputs is unclear

ProfDrYoMan, 2025-03-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

On drag and drop it is unclear if the input is moved or the input line is moved.
Also the modifiers are unclear.
It seems shift should move and not copy (but input or line).

### Expected Behavior

Not clear what the expected behavior is.

### Steps To Reproduce

Move by drag and drop. Try modifiers (Shift, Ctrl).
There should be a consistent way to move inputs and or input lines by drag and drop.

### Version

2.10.6

### Transmitter

Other (Please specify below)

pfeerick, 2025-03-15:
When are you pressing the keys? i.e. start your drag operation, and *then* either use Shift to make it a move, or nothing (or Ctrl, if it helps you) for it to be a copy.

ProfDrYoMan, 2025-03-15:
Missed the OS. It's WIN 10/11.

That is what I expected and tried. I am pressing the modifiers as soon as I started drag.
If I do not press any modifier a small (+) icon appears at the cursor. Same for pressing CTRL.
If I drag and then press SHIFT icon at the cursor is still (+), but it seems a move is done. 
^^^ Sorry, my screen resolution is so high, I know realized that the (+) changed to a (->). :)

And on top, move/copy is clear now as I also realize that I cannot place anything in between, that means I need to an "input->insert" if I want to get a complete input moved (marking all lines) and not added as lines to an existing input.

If this is the intended functionality this bug can be closed and the only problem for me is #5988.

It might be nice to get a bigger (+) or (->) icon at the cursor, since this is very small comparing to all other situation where this drag and drop flow is used in other programs.