EdgeTX/edgetx #5995: Latest FrSky Archer Plus SR8 firmware not compatible with SxR LUA scripts

M-i-c-h-a-e-l1, 2025-03-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

Latest FrSky Archer Plus SR8 (with gyro) firmware is not compatible with SxR LUA scripts:

The[ FrSky Archer Plus SR8 receiver (with gyro) firmware has been updated on 2024-07-31 to version 1.0.11 and on 2024-12-20 to version 3.0.1](https://www.frsky-rc.com/archer-plus-sr8/). 
LUA scripts are necessary to calibrate and configure the gyro function of the receiver. 
As stated by FrSky, the LUA scripts version 3.0 or higher are needed to calibrate and configure the SR8 receiver as from firmware version 1.0.11 and 3.0.1.

**Issues:**

- The SxR LUA scripts v 2.x supplied with EdgeTX do not work with the SR8 gyro receiver firmware 1.0.11 and above any more.
- The LUA script v 3.0.1 for ETHOS does not work on the Radiomaster TX16S MkII.

### Expected Behavior

Use the LUA scripts to calibrate and configure the SR8 gyro receiver.

**Temporary work around:**

- Downgrade the SR8 gyro firmware to 	v1.0.10 of 2023-12-27 and the LUA scripts v 2.x work to calibrate and configure the SR8 gyro receiver.

**Request:**

- Could the LUA scripts be updated so that they can calibrate and configure the gyro function of the SR8 receiver?
- New features of the receiver also include auto calibration and device reset via the LUA script.

Kind regards
M

### Steps To Reproduce

**Set-up:**

Transmitter: 
- Radiomaster TX16s MkII
- EdgeTX: 2.10
- MPM firmware 1.3.40

Receiver:

- FrSky Archer Plus SR8 with gyro
- Frirmware: Not sure as I do not have the means to check the firmware. As the receiver was bought in March 2025, it will probably be the v3.0.1 of 2024-12-20.

1. Bind the receiver to the transmitter.
2. If bind is successful check if the telemetry data incluses gyro telemetry data such as acceleration and angle measurements in degrees.
3. Run the LUA script SxR on the transmitter to calibrate and configure the gyro settings of the receiver.
4. While the values for the gyro settings appear on the transmitter screen it does not seem to save the settings on the receiver. When relaunching the LUA script the default values for the gyro configuration reappear.

### Version

2.10.0

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

W10

Maden82, 2025-03-23:
Is there a way to downgrade the sr8 with the tx16s so i can use the work around?

M-i-c-h-a-e-l1, 2025-03-24:
Hello,

I used the FrSky STK USB to flash the receiver. The following method will probebly also work. Pay good attention to the polarity of the cable: [](https://www.youtube.com/watch?v=Jwp2Pd_x6I0)

Kind regards
M

M-i-c-h-a-e-l1, 2025-03-24:
When looking at the FrSky page it is clear that all other receivers with a gyro are also affected by this:

- ARCHER PLUS SR6 Mini
- ARCHER PLUS SR12+
- ARCHER PLUS SR10+

Kind regards
M

blopa1961, 2025-03-26:
> When looking at the FrSky page it is clear that all other receivers with a gyro are also affected by this:
> 
>     * ARCHER PLUS SR6 Mini
> 
>     * ARCHER PLUS SR12+
> 
>     * ARCHER PLUS SR10+
> 
> 
> Kind regards M

This is not a bug or problem with EdgeTX, it's FrSky's decision to stop supporting OpenTX (now more than 2 years old) and push their own ecosystem under ETHOS.
As you can see there's no OpenTX LUA 3.x for these receivers; I suggest you complain to FrSky (and close this issue).

duxa08, 2025-04-16:
I use Radiomaster tx16s opentx. Receiver sr6 mini Firmware 1.0.10 lua 2x. When starting lua, the Enter button is displayed, when pressed, nothing happens, the script freezes. How to solve this problem?

blopa1961, 2025-04-16:
> I use Radiomaster tx16s opentx. Receiver sr6 mini Firmware 1.0.10 lua 2x. When starting lua, the Enter button is displayed, when pressed, nothing happens, the script freezes. How to solve this problem?

Are you actually running opentx? If so either post your question in the opentx github or install EdgeTX 2.10.6 and try again.
I assume your receiver is bound to your radio in ACCST mode and you are getting telemetry, right?

M-i-c-h-a-e-l1, 2025-04-20:
Hello,

There is no specific LUA script for EdgeTX from FrSky. I use the [OPTX LUA script v 2.0.4 ](https://www.frsky-rc.com/wp-content/uploads/Downloads/TOOL/R9S-SxR%28lua%29/AP_STAB_V2.04/FrSky_AP_STAB_RX_v2.04.zip)wich works on my Radiomaster TX16 Mk2 with EdgeTX 2.10.

If this does not work try the standard pre-installed LUA script on your transmitter with EdgeTX.

Kind regards
Michael

blopa1961, 2025-04-21:
> Hello,
>
> There is no specific LUA script for EdgeTX from FrSky. I use the [OPTX LUA script v 2.0.4 [...]

There is no working LUA for EdgeTX/OpenTX for receiver firmware 3.x, [OPTX LUA script v 2.0.4 ] does not work with that firmware.

M-i-c-h-a-e-l1, 2025-04-21:
Yes indeed, this is what I have written in my initial post on 16 March 2025.

blopa1961, 2025-04-21:
> Yes indeed, this is what I have written in my initial post on 16 March 2025.

Then close this issue, it's not an EdgeTX problem, it's an FrSky problem.

M-i-c-h-a-e-l1, 2025-04-21:
Hello,

Since the LUA scripts exist for Ethos, the idea could be that someone with a good knowledge of LUA could adapt/rewrite the Ethos LUA to EdgeTX LUA.

I can provide a receiver with the 3.x FrSky firmware to make the necessary tests when rewriting the LUA script.

Kind regards
Michael

frankiearzu, 2025-04-25:
Just to let you know that I have a prototype working with Archer SR10+ with firmware 3.0.1.
I need to do some cosmetic changes, but it runs well on TX16..will have to make more changes to make it work on smaller BW screens.

Let me know if you are willing to help test it.  The EdgeTX code was based on the Ethos code.

M-i-c-h-a-e-l1, 2025-04-25:
Yes, I will definitely test the LUA scripts for EdgeTX. I can test them on a Radiomaster TX16 Mk2 with colour screen and on a Radiomaster Zorro with b/w screen with a FrSky Archer Plus SR8.

Many thanks for your effort, much appreciated.

Kind regards
M

frankiearzu, 2025-04-25:
Copy the content into your /SCRIPTS/TOOLS.
It will have a single file on /SCRIPTS/TOOLS/FrSky_SR_Suite.lua,   and then everything else inside FrSky_SR folder..  Easier to remove/update that way. Runs on 2.10.x and 2.11.x

[FrSky_SR_Suite-v1.0-RC1.zip](https://github.com/user-attachments/files/19917510/FrSky_SR_Suite-v1.0-RC1.zip)

Maarten112, 2025-04-27:
> ZIP file with the LUAs [FrSky_SR_Suite-v1.0-RC1.zip](https://github.com/user-attachments/files/19917510/FrSky_SR_Suite-v1.0-RC1.zip)
> 
> Copy the content into your /SCRIPTS/TOOLS.
> 
> You will end up with: /SCRIPTS/TOOLS FrSky_SR_Suite.lua: main program FrSky_SR folder.. with everything else
> 
> Easier to remove/update that way. Runs on 2.10.x and 2.11.x. Only work on color radios right now. No touch screen, so use the keys ENTER/RTN/PAGE to navigate.. Still working on some scripts to get back to the main menu when exiting.
> 
> As always, make sure that everything works as intended after setup and before first flight!!!
> 
> It has functionality for the SxR (S6R/S8R) as well as Archer Plus SR8+/SR10+/SR12+ with firmware v1.x or v3.x.
That's awesome @frankiearzu !

Will try it out tomorrow! :)

M-i-c-h-a-e-l1, 2025-04-27:
Hello,

I did some basic **bench testing** as follows:

**Transmitter: Radiomaster TX16S Mk2**
- OS: EdgeTX v2.10.0
- Internal multi module: V1.3.4.0AETR

**Receiver settings for receiver A and B:** 
- Mode: Multi / FrSky X2
- RF protocol: D16

**Receiver A:** 
- Brand / model: FrSky / Archer Plus SR8
- Firmware: v3.0.1
- LUA SCRIPT v3.0.6

**Receiver B:**
- Brand / model: FrSky / Archer Plus SR8
- Firmware:  v1.0.10
- LUA SCRIPT: v2.04.10

**LUA sctipt installation:**
- Worked as described
- I particularly like that it is a suite and it works both with the 1.x firmware and the 3.x firmware and the screen #4 with a reminder for the switches.

* **Receiver A** Firmware 3.0.1

Stabilisation setup:
_Page 1/4_
- Group/Gyro1
- Only simple mode tested (glider)
- Group/Gyro2 not tested (is this only for the SR10 and SR12 receivers?)

_Page 2/4 Gyro/Group1_
-  Why can hover and knife edge values be changed if simple mode is selected?
- I am not sure what stick priority and the indicated % mean (my lack of knowledge ...)

_Page 3/4 Gyro/Group 2_
- Could not make ant changes to the values (only for SR10 and SR12?)

_Page 4/4 Switch reminder_
- Very nice and short reminder on the channels and switches to be used.

_PreCalibration_
Level/Stick Range Cali Gyro 1
- Level calibration: ok
- Stick center calibration: ok
- Stick range calibration: ok

Gyro2 not able to test

_6-Axis Calibration:_
Manual calibration process ok.

I then used the script to select the FW v1.x and this also seems to work with the FW v3.x firmware? There I can then enable/disable Gyro 2?

* **Receiver B** Firmware 1.0.10
Stabilisation setup:

_Page 1/4_
- Group/Gyro1
- Only simple mode tested (glider)
- Group/Gyro2 not tested (is this only for the SR10 and SR12 receivers?)

_Page 2/4 Gyro/Group1_
-  Why can hover and knife edge values be changed if simple mode is selected?

_Page 3/4 Gyro/Group 2_
- Could not make ant changes to the values (only for SR10 and SR12?)

_Page 4/4 Switch reminder_
- Very nice and short reminder on the channels and switches to be used.

_6-Axis Calibration:_
Manual calibration process ok.

Kind regards
Michael

frankiearzu, 2025-04-27:
The group/gyro 2 only works when using it with ACCESS.. with ACCST only group 1.  Will see if there is a way in the lua to detect the protocol used.

If the rx firmware is v3.x, but using v1 LUA, the requests for data for v1 to the RX always returns 0, that looks like most defaults values (would be better if it should just not answer).. but if you change them, they have no effect.

Currently there is no way to read what version does the RX has in FW v1.x..
Could be just is NOT v3.x

Will see if I can add a way to detect the RX

frankiearzu, 2025-05-01:
Just FYI,
Finally merged the code with shmuely's repository, so for updates, look there:
https://github.com/offer-shmuely/edgetx-x10-scripts

@pfeerick, this can be closed.

M-i-c-h-a-e-l1, 2025-05-02:
Hello,

Many thanks for the grat work done!

Would it be possible to have this updated [FrSky_Gyro_Suite](https://github.com/offer-shmuely/edgetx-x10-scripts/tree/main/SCRIPTS/TOOLS/FrSky_Gyro_Suite) as standard included on the EdgeTX SD card. This would replace the current existing FrSky receiver configuration script which is outdated and the users would't need to seach for the updated FrSky_Gyro_Suite or may not even knnow that this even exists.

Kind regards
Michae

pfeerick, 2025-05-02:
You can download the entire repo as a zip file via https://github.com/offer-shmuely/edgetx-x10-scripts/archive/refs/heads/main.zip and then pick out what you want. 

There is a download option under the green "Code" button up the top right-ish side of the main page. 

[image]

> Would it be possible to have this updated [FrSky_Gyro_Suite](https://github.com/offer-shmuely/edgetx-x10-scripts/tree/main/SCRIPTS/TOOLS/FrSky_Gyro_Suite) as standard included on the EdgeTX SD card.

This is entirely up to Offer... if he wants to do a PR to add it, it is welcome. Regardless, now that this is a thing, it will need to be added to https://github.com/EdgeTX/lua-scripts. 

Much appreciated to you both for working on this @frankiearzu @offer-shmuely 💪 🥳

closed 2025-05-02 by pfeerick