EdgeTX/edgetx #5997: Automatic Model Menu Close

inventor7777, 2025-03-17:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Hey all! 

My father recently got an MT12 to use on his cars and crawlers. One improvement that he suggested which I really liked was that the “select model” menu automatically close once a different model is selected. (Which I think would make most sense as a setting that you can change)

### Describe the solution you'd like

Have a setting that when enabled, automatically returns to the home page after choosing a new model.

### Additional context

This request is mainly for B&W surface radios but it could be useful for color and air radios as well.

pfeerick, 2025-03-18:
Color LCD already does this, however I don't see why it couldn't be an option for both, as sometimes it may make sense to the user that the model menu does or doesn't automatically close.

inventor7777, 2025-03-19:
@pfeerick i haven’t used my TX16 in so long that I forgot it does have that! 🤣 

My plane is in the shop right now because I have to unlock my RX to support the new servo system I’ve added but I can’t get Spektrum Programmer to work 😡

inventor7777, 2025-07-23:
@pfeerick any update on this?

inventor7777, 2025-10-18:
@pfeerick do you think this could be added for 3.0?