EdgeTX/edgetx #5998: Model conversion from OpenTx to EdgeTx gives inverted result for pre-flight checks on potentionmeters

webeule, 2025-03-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I just switched from OpenTX 2.3.15 to EdgeTX 2.10.6 on my Taranis X9D+.
Before I did the switch I only did a binary backup of the models and settings (.bin file) and have not read the models and settings from the radio and then wrote as otx file. When I recognized that I need this otx file to transfer the models to the radio after the upgrade to EdgeTX, I reloaded the backup into the OpenTX Companion and then saved the otx file.
When I load this file into the EdgeTX Companion, I do not get any message saying it needs to convert the models and I am able to transfer it to the radio. Anyhow, in one of my models, I have a pre-flight check for the LS potentiometer. After the conversion the potentionmeter checks have been inverted, so they are checking for S1, S2 and RS and not for LS.

### Expected Behavior

The pre-flight checks for the potentiometers are converted correctly.

### Steps To Reproduce

See current behaviour.

### Version

2.10.6

### Transmitter

FrSky X9D+

### Operating System (OS)

Windows

### OS Version

Windows 2010 22H2

elecpower, 2025-05-02:
Warnings conversion was a casualty of consolidation of the B&W and colour radio settings handling as they had diverged to the point conversion rules had gotten out of hand. So unfortunately some settings need to be manually reset by the user as part of the normal checking procedure when upgrading.

pfeerick, 2025-05-03:
@elecpower So should we consider it to be a "known issue" requiring manual review after ugrade and leave it at that?

gagarinlg, 2025-05-03:
Yes, not many people left who move from OpenTX to EdgeTX, most either migrated already or do not want to migrate

3djc, 2025-05-03:
Yes, time would be much more efficiently spent other places

closed 2025-05-03 by pfeerick