EdgeTX/edgetx #6002: Trim switch max/min values incorrect for backlight control

psfales, 2025-03-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using T6 as a global function to control backlight, the backlight goes off at the very top of the range and goes to full brightness at the very bottom (but behaves as expected ever the rest of the range) 

### Expected Behavior

Something strange with the T6 trim switch on my TX16S.  Up until now, I've been using S1 to control the backlight and it works as expected.    As an experiment, I tried changing the global function to control the display brightness with T6.   It **mostly** works as expected:  T6+ makes the display brighter and T6- makes  the display dimmer, EXCEPT at the very end of the range.      As I keeping hitting T6- the display goes dimmer and dimmer and eventually is completely black until I hear "minimum trim reached" and the display comes on at full brightness.   Similarly if I keeping hitting T6+ it gets brighter and brighter until "maximum trim reached" and the display goes black.

This feels like a bug to me - like something is unexpectedly wrapping around.   I can somewhat work around it by connecting a spare input to T6, and using that input to control the backlight.   But that requires adding that input input and special function to every model.   Using the global function and T6 should (in theory) work everywhere with no model-specific changes.

### Steps To Reproduce

1.   Set up a global function with "Backlight" connected to T6
2.  Starting hitting T6+ and watch the display get brighter 
3.  When you hear "maximium trim reached" the display goes black
4.  Hit T6-  and the display comes on at full brightness
5.  Starting hitting T6- and watch the display get dimmer
6. When you hear "minimum trim reached" the display suddenly comes on af full brightness
7. Hit T6+ one time, and the display goes black

### Version

2.10.3

### Transmitter

RadioMaster TX16S / TX16SMK2

psfales, 2025-03-18:
A friend was unable to reproduce this on his TX16S running 2.10.6, so I tried updating my radio to 2.10.6 and I'm still seeing the problem.   There seems to be something different between his radio and mine -- I don't know if it's different hardware or something in the configuration

philmoz, 2025-03-18:
Turn off 'Extended Trims'.

psfales, 2025-03-19:
[https://photos.app.goo.gl/s28EfySHV54twHCU9]   I added an GF and LS to play the value of T6 whenever it changes

psfales, 2025-03-19:
I don't use extended trims.   (In other words "Extended Trims" is already off)

pfeerick, 2025-03-19:
For 2.10.x this is not related to Extended Trims, but is instead a bug I think we fixed along the way in 2.11.x. In 2.11, this will only occur if you enable Extended Trims.

inventor7777, 2025-03-19:
Funnily enough, I used to use this bug to quickly switch between on and off backlight on my MT12. (In addition to moving trims up and down) When I updated to a newer version I couldn’t do it anymore LOL 

I probably should have reported it but I got so used to it working that way that I didn’t even think to make a bug report…

psfales, 2025-03-19:
Thanks for the info.    It was more a question of wanting to solve the puzzle than a critical problem.   I can live with the current behavior until 2.11 comes out.

3djc, 2025-03-19:
Will introduce a PR to limit() the values

psfales, 2025-03-20:
Someone on Facebook found some very useful additional information.   They wrote:

"I think I figured it out! At least I can duplicate it on my radio. Go to System, click the gear and then the Backlight button. Go to the ON brightness slider and using the roller wheel, lower it one click down from maximum brightness. Next, go to the OFF Brightness and make that one click up from minimum. That should solve your problem. Let me know if that works for you. "

I made that change on my radio, and it solved (or at least provided a work-around) for my problem.

pfeerick, 2025-03-20:
I'd have to check the code again around there to be 100% certain, but that should "just" be making it so the absolute min and max can't be reached, hence why it's a valid workaround. #6009 should cure it completely though, by limiting the input into the backlight special function to the range it expects.

psfales, 2025-03-21:
Agree.   It's probably just hiding the problem, rather than a real fix.  But it seems to do a pretty good job of hiding it for now!  :-)

3djc, 2025-03-21:
Not really, as you don't always have a way to know the value range. You can try to guess it for radio inputs, but you could use lua values, telemetry ones, with absolutely no ways to know range.

PR #6009 (2025-03-21): fix(radio): constrain Backlight SF input value to valid range
Limit the backlight SF input value to admissible range

This fixes #6002