EdgeTX/edgetx #6011: Scroll Wheel Reversed

jhoppenrath, 2025-03-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a Radio Master Pocket Flite Test Edition.  I just updated to Centurion 2.10.6 and now my scroll wheel is working in reverse.  So when the scroll wheel is moved to the right it decrease the selection.

### Expected Behavior

When the scroll wheel moves to the right it increases the selection.

### Steps To Reproduce

Enter a menu and move scroll wheel to the right and left.

### Version

2.10.6

### Transmitter

RadioMaster Pocket

### Operating System (OS)

Windows

pfeerick, 2025-03-23:
Check what SYS -> Radio Setup -> RotEnc Mode is set to.

jhoppenrath, 2025-03-23:
Thank you.  I looked all over for that and missed it every time!!!!!  I am back to normal thanks to your help.

Jason

closed 2025-03-23 by jhoppenrath