EdgeTX/edgetx #6018: elevons don't elevate but do aileron. channel 6 also not sent

arkarkark, 2025-03-24:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

This is going to be hard to believe but I took a [video to prove it](https://youtu.be/PI4EGG5jt9w).

I have a delta wing plane with motor and two elevons and a buzzer on channel 6. It unexpectedly crashed and when I found it I noticed my matek buzzer on channel 6 wasn't working. then I noticed that even though both servos moved when I moved my aileron input left and right, they did nothing when I moved the elevator input.

I looked at the radio and the display registered the stick movement correctly (the little o in the boxes moved as expected) and then I looked at the channel output and showed ailerons correctly moving channel 1 & 2 but no movement on those channels for the elevator input. The servos did not move for elevator input.

I power cycled both the radio and the plane and everything went back to working correctly. I wish I had power cycled just one and then the other and then both but I'm afraid I didn't. It's hard to believe it was on the plane's side since the mix is done in the radio.

### Expected Behavior

moving the elevator input should move the elevons up and down.

### Steps To Reproduce

1. push elevator up
2. watch elevons not move up

this is inconsistent and has only happened once.

### Version

Other (Please specify below)

### Transmitter

RadioMaster GX12

### Anything else?

I'm running the firmware that came with my replacement gx12 motherboard. It's listed as `2.11.0-RM-Build`
git# `2e1becd1`

I'm waiting to install 2.11 RC3 which I see has been tagged but not on the releases page yet, I hear it has some other GX12 fixes in it.

arkarkark, 2025-03-24:
here is the model file. if you need it

[pumpkin.yml.txt](https://github.com/user-attachments/files/19438727/pumpkin.yml.txt)

3djc, 2025-03-26:
Your setup is really dangerous. If both L11 and L12 are disabled, you loose elevator, which happens is SB↓

Always use a catch all line...

3djc, 2025-03-26:
There might be other conditions that trigger it. Bottom line is always end every input and mix line with a line without switch condition

RustyWraith, 2025-03-26:
I had a recurring issue with 2.11 RC1-2.
On NV14 in random moments all channels except AETR stop responding on switches.
AETR were included in INPUTS tab. Other channels are present only in MIXER tab.
I'm not having this issue with 2.10 for now.

[model4.yml.txt](https://github.com/user-attachments/files/19476164/model4.yml.txt)
Records in INPUTS were deleted in this model file.

arkarkark, 2025-03-26:
@RustyWraith that sounds similar to a [different problem](https://github.com/EdgeTX/edgetx/issues/5841) I was having where model files would get truncated. Check the size of your model yml files, are they always 1024 bytes after this thing happens. the fix for me was a new GX12 motherboard.

RustyWraith, 2025-03-26:
No, I mean I had reorganized my model before I came up with idea to report a bug.
Reboot helps to restore the response in my case.