EdgeTX/edgetx #6037: CMake 4 Compatibility (Companion)

lodriguez, 2025-03-30:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

Building Companion I run in following Errors:

```
CMake Error (cmake_policy):
  Policy CMP0026 may not be set to OLD behavior because this version of CMake
  no longer supports it.  The policy was introduced in CMake version 3.0.0,
  and use of NEW behavior is now required.
```
and

```
CMake Error (cmake_minimum_required):
Compatibility with CMake < 3.5 has been removed from CMake.

Update the VERSION argument value. Or, use the ... syntax
to tell CMake that the project requires at least but has been updated
to work with policies introduced by or earlier.

Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.
```

### Expected Behavior

build companion without errors

### Steps To Reproduce

1. install CMAKE 4
2. built companion

### Version

2.11.0-rc

### Transmitter

Other (Please specify below)

### Operating System (OS)

Linux

### OS Version

arch 6.14

### Anything else?

I've only tried companion (I maintain the build script for the AUR)

To made it build, I had to set:

-DCMAKE_POLICY_VERSION_MINIMUM=3.5

and patch:

 sed -i 's/cmake_policy(SET CMP0026 OLD)/cmake_policy(SET CMP0026 NEW)/' edgetx/companion/src/CMakeLists.txt
  sed -i 's/cmake_policy(SET CMP0023 OLD)/cmake_policy(SET CMP0023 NEW)/' edgetx/cmake/GenericDefinitions.cmake

elecpower, 2025-03-30:
I have raised an issue report with project https://github.com/richgel999/miniz/issues/337

pfeerick, 2025-03-31:
Also keep in mind that 4.0 has only just been released (March 28, 2025) and we are currently not using or support CMake 4.0. The next revision of the build environment will probably suck it down though, so we will probably need to update soon.

pagrey, 2025-04-22:
#6070 resolved this for me.

pfeerick, 2025-04-23:
Thanks... I should know better than to not link the PR to the issue 😆

closed 2025-04-23 by pfeerick