EdgeTX/edgetx #6040: FrSky X10s blue LED is switching off after startup in 2.11.0-rc3

elekSENDER, 2025-03-31:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

By switching on FrSky X10s the blue LED is on until the startup procedure is finished. After startup the LED switches off.

### Expected Behavior

Blue LED remains on.

### Steps To Reproduce

Switch on a X10s with 2.11.0-rc3

### Version

2.11.0-rc

### Transmitter

FrSky X10 / X10S (ACCST)

### Anything else?

Radio is a X10s (not express) with ISRM-upgrade module.

In 2.11.0-rc2 the blue LED remains on.

[image]

[image]

pfeerick, 2025-04-01:
So looking at a few videos, and it looks like the X10 should have RGB power led? i.e. you should have seen Red, Green and Blue from it at some point in time?

What if you start up the radio with a switch or stick warning active... does it come one green then once the radio has booted?

MRC3742, 2025-04-01:
The X10 Series of radios use the RGB Green LED as a charge indicator.
It is on while charging and off when fully charged.

I don't believe that the Green LED on these radios is controlled by an MCU pin ?

elekSENDER, 2025-04-01:
Yes, I have seen red, green and blue on this led.

Normal startup procedure:
While startupscreen with the EdegTX logo the led is blue.
If there is any warning condition the red led is on.
Normal operation is Blue led is on.

I have only seen the green light while charging the radio.
If charging in operation there are blue and green on, like a kind of turquoise.

elekSENDER, 2025-04-01:
> The X10 Series of radios use the RGB Green LED as a charge indicator. It is on while charging and off when fully charged.

exactly same behavior on mine radio.

MRC3742, 2025-04-02:
I also have an X10S EXPRESS (ACCESS) that shows the same behavior after updating to v2.11.0-RC3
BLUE on start - RED on Warning - LED is off after startup.

Reverting to versions 2.11.0-RC2 or lower results in the normal behavior of
BLUE on start - RED on Warning - BLUE on after startup.

pfeerick, 2025-04-02:
Ok, perfect... that suggests my suspicion that the green LED is incorrectly defined as GPIO controlled is correct. 

As one final check, can you try the attached firmware (2.11 head, standard build options) that cycles the power led colours on startup, and if it doesn't go R G B, but R _ B, either the GPIO assignment is wrong, or we can't control the G led at all. Note: the booloader may still initially set the blue on start, but it will then be turned off as the main firmware starts, and then R G B  in sequence, with a 0.5 second between each... then turned off again and then normal startup will continue. It would be great if you can both try it, since that would test both the x10-accst and x10-access variants. 

[x10express-rgbtest-70b5a74a44.zip](https://github.com/user-attachments/files/19560157/x10express-rgbtest-70b5a74a44.zip)
[x10-rgbtest-70b5a74a44.zip](https://github.com/user-attachments/files/19560158/x10-rgbtest-70b5a74a44.zip)

MRC3742, 2025-04-02:
Tested on X10 EXPRESS
I get a quick Blue as bootloader starts then R _ B and off again after startup.
NO Green at all.

I am unable to test my X10 (ACCST) as it has the touch screen mod which is preventing startup on current firmware.
@elekSENDER please test on your X10 transmitter

elekSENDER, 2025-04-02:
> Tested on X10 EXPRESS I get a quick Blue as bootloader starts then R _ B and off again after startup. NO Green at all.

tested on X10S, exact the same sequence.

PR #6048 (2025-04-04): chore(x10): LED_GREEN_GPIO should not be configured
Fixes #6040

Summary of changes:
- LED_GREEN_GPIO should not be configured on X10/X10 Express as it is not MCU controlled
closed 2025-04-04 by pfeerick