EdgeTX/edgetx #6046: Radio X10 w/ Touch Screen Mod - WRONG PCB DETECTED since 2.11.0-RC1

MRC3742, 2025-04-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Since loading firmware v2.11.0-RC, all 3 versions, the FrSky X10 Series (non ACCESS)  with touch screen mod fails to start with WRONG PCB DETECTED. The only way to power down TX is remove the back cover and unplug the battery.

Returning firmware to v2.10.6 and TX will now start and operate normally.
I think I've found the problem in the updated code and working on a PR to fix.

Rich

### Expected Behavior

TX to start and operate.

### Steps To Reproduce

Load any version of 2.11.0-RC firmware on the X10 with Touch Screen Mod.
Start Radio - WRONG PCB DETECTED screen

### Version

2.11.0-rc

### Transmitter

FrSky X10 / X10S (ACCST)

pfeerick, 2025-04-02:
Can you see if you can repo #6040 once you get it working again? ;)

MRC3742, 2025-04-08:
@pfeerick 

I am having a little difficulty in understanding the current code in order to rectify this PCBREV failure.

In this file the values are set to determine the Standard X10 and X10 EXPRESS.
https://github.com/EdgeTX/edgetx/blob/50b32e6cfc37fe70a829e0c4b64f017993526f08/radio/src/targets/horus/board.h#L78-L80

It also has the code to check for which board is detected.
https://github.com/EdgeTX/edgetx/blob/50b32e6cfc37fe70a829e0c4b64f017993526f08/radio/src/targets/horus/board.h#L89-L96

These are the pin configurations for the different models. 
X10 STD:  GPIO pins PH.07 and PH.08 are pulled down to ground, PA.06 is floating
X10 STD w/ TS Mod: PH.07 and PH.08 are SCL and SDA, PA.06 is pulled down to ground
X10 EXPRESS: pin PH.07 is pulled up to 3.3v and PH.08 and PA.06 are floating
X10 EXPRESS w/ TS Mod: PH.07 and PH.08 are SCL and SDA, PA.06 is floating

The previous code used the line below to set the PCBREV_VALUE which returned these ?
X10 STD = ( 0 + 0 ) * 1                            //value 0
X10 STD w/ TS Mod = ( 1 + 2 ) * 0           //value 0
X10 EXPRESS = ( 1 + 2 ) * 1                     //value 3
X10 EXPRESS w/ TS Mod = ( 1 + 2 ) * 1    //value 3
https://github.com/EdgeTX/edgetx/blob/a77fd73e78e667e6f05797b7c0eb08d1ac3a0e19/radio/src/targets/horus/hal.h#L422

The current code has changed to this
https://github.com/EdgeTX/edgetx/blob/50b32e6cfc37fe70a829e0c4b64f017993526f08/radio/src/targets/horus/hal.h#L617-L621
Although the multiplier I added to allow PCBREV for the TS Mod PA.06 pin is missing,
I don't understand how this current piece of code can return "0" or "3".
What am I missing?

Thanks - Rich

pfeerick, 2025-04-14:
I've not forgotten about this, will look into it tomorrow or tonight, time permitting.

MRC3742, 2025-04-17:
Thanks for your input but that will not work with the current code.
I have no debugging equipment so the only way for me to find the values is by trial and error.
The believe the values on the X10 EXPRESS are as follows
`PCBREV_GPIO_1     GPIO_PIN(GPIOH, 7) // PH.07`  has a value of "128" (this pin has a 1K pullup resistor to 3.3v on board)
`PCBREV_GPIO_2     GPIO_PIN(GPIOH, 8) // PH.08`  has a value of "256" (this pin is floating but pulled up internal in "pwr_driver.cpp"

`#define PCBREV_VALUE()     ((gpio_read(PCBREV_GPIO_1) | gpio_read(PCBREV_GPIO_2)) >> 7) `
`#define PCBREV_VALUE()`     (0000 0000 1000 0000) "128"  |  (0000 0001 0000 0000) "256" ) >> 7)
16 bit binary result = ((0000 0001 1000 0000) >>7 ) = "3" = PCBREV (EXPRESS)

`PCBREV_TOUCH_GPIO     GPIO_PIN(GPIOA, 6) // PA.06`  (which was used to determine if the touch screen was added on the X10) always returns a value of "0"  (this pin is floating but does not change value by pulling up or down in "radio/src/targets/common/arm/stm32/pwr_driver.cpp"

The current code works for non touch radios but I cannot use the GPIOA Pin 6 as a factor because it always returns "0" ?

Thanks in advance for any help on me understanding my error(s) - Rich

pfeerick, 2025-04-18:
Will not, or does not? 🤪 

Try this then... if that don't work, will have to bug JC or see if Raphael has come out of hiding 🤭 

```cpp
#define PCBREV_VALUE() ((gpio_read(PCBREV_GPIO_1) ? 1 : 0) + ((gpio_read(PCBREV_GPIO_2) ? 1 : 0) << 1)) * (gpio_read(PCBREV_TOUCH_GPIO) ? 1 : 0)
```

Hm... this looks somewhat fishy also... I don't think GPIO2 should be being pulled up when testing if PCBREV_TOUCH_GPIO_PULL_UP is defined 🤪 Perhaps that was a copy and past error, and supposed to be pulled up like with PCBREV_GPIO_1, and PCBREV_TOUCH_GPIO pulled down if defined, unless specifically pulled up? 

[image]

i.e. 
```cpp
  gpio_init(PCBREV_GPIO_1, GPIO_IN_PU, GPIO_PIN_SPEED_LOW);
  gpio_init(PCBREV_GPIO_2, GPIO_IN_PU, GPIO_PIN_SPEED_LOW);
  #if defined(PCBREV_TOUCH_GPIO)
    #if defined(PCBREV_TOUCH_GPIO_PULL_UP)
      gpio_init(PCBREV_TOUCH_GPIO, GPIO_IN_PU, GPIO_PIN_SPEED_LOW);
    #else
      gpio_init(PCBREV_TOUCH_GPIO, GPIO_IN_PD, GPIO_PIN_SPEED_LOW);
    #endif
```

MRC3742, 2025-04-19:
"Will not work with current code format" is probably a better description but "does not" is simpler! :stuck_out_tongue:

Your updated code for "horus/hal.h" is now working on all radios tested.
Who would have thought to use the bitwise ternary or conditional operator! .....(not me that's for sure)

[image]
`https://www.geeksforgeeks.org/bitwise-operators-in-c-cpp/`

I had already made the changes you had suggested for "pwr_driver.cpp" but still could only get a "0" value from PA.06 ?
I also simplified it a bit because the only time that pin should be pulled down is adding a resistor to ground in the TS Mod.

Please review the PR and make any changes you feel are needed.

Thanks for your help and knowledge on working through this problem - Rich

MRC3742, 2025-04-19:
Poking around I think I found a similar but different problem in "taranis/hal.h" with PCBREV for the X7
https://github.com/EdgeTX/edgetx/blob/8f6f4cc07ca41bee031f16aa2d6f978c971305b9/radio/src/targets/taranis/hal.h#L2367-L2372
Line 2369 should be "#define PCBREV_GPIO" not "#define PCBREV_GPIO_PIN" ?
I don't have any of these radios to test but it appears to be incorrect.

Could make a separate issue if needed? - Rich

pfeerick, 2025-04-19:
Yeah, that *looks* wrong also, and stems from the same PR, so throw into your PR also. I'll get JC to review also to make sure everything makes sense still. You've basically been able to test all the X10 combinations, haven't you?

MRC3742, 2025-04-20:
commit added to PR 6108 to correct X7 PCBREV GPIO pin define.

Yes the X10 has tested OK in all combinations.

MRC3742, 2025-04-23:
Closing this issue as the update fix was merged in 6108

PR #6108 (2025-04-23): fix(radio): WRONG PCB DETECTED - X10 with Touch Screen Mod
Fixes #6046

WRONG PCB DETECTED error on X10 with Touch Screen Mod.
Since updates to new GPIO driver ported from RIOT-OS in PR 5023 was merged.

Small corrections to detect proper PCBREV on these radios with the added touch screen.

Thanks to pfeerick for his help on solving this problem. :smiley:
@pfeerick please review and squash before merge.

Rich
closed 2025-04-23 by MRC3742