EdgeTX/edgetx #6049: Channel name not displayed in Overr SFs

inventor7777, 2025-04-02:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I just noticed that with newer Edge versions, regardless of input, mix, or channel name, that the Overr special function only shows channel numbers CH1-CH32, and not the channel names.

### Expected Behavior

In previous versions of EdgeTX, you could see the channel name which was to be overridden. 

### Steps To Reproduce

Name an output and override it with an SF. The SF will only show the channel number, but no name.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

macOS

### Anything else?

[image]
[image]

I'm running the 3/31 nightly.

philmoz, 2025-04-02:
PR #5892

inventor7777, 2025-04-03:
Great. Any idea when it will be released into EdgeTX?

pfeerick, 2025-04-03:
I think you may misunderstand. That behaviour was a bug introduced in 2.10,
and fixed in 2.11 and a yet to be released 2.10.7. Prior to 2.10.x, the
channel number would be shown in relation to the Override SF, not the name,
but this was due to `putsChn()` function not behaving as it should be
generally. In this case, perhaps as with other instances, if an output is
named, the name should be shown. Some TLC generally might be nice, as on
colorlcd you have to set the channel number, but then only the name (if
set) is shown in the overview... would probably be nice to have the ...
regular ... source pick list there to show both name and number, and thus
the name can be consistently used in colorlcd, and then on B&W also.

inventor7777, 2025-04-03:
Oh, I get it now. I didn’t read the pull request very thoroughly as i was quite tired…..

In that case i think it should be at _least_ a setting so that the channel name is shown in SFs, etc as i really liked it and thought it was a new feature when i was using versions before the fix.