EdgeTX/edgetx #6055: EdgeTX Frizing during Bluetoot connection

Enzo720710, 2025-04-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The problem only appeared today when I connected my new headtracker via bluetooth, it had never given me this problem before.
After 12 minutes of flight I started hearing a beep from the radio, the screen was frozen but the connection to the model was still active.
I landed and turned off and on the radio and it is alive.

### Expected Behavior

That it would continue to function as it always has

### Steps To Reproduce

fly for 12 min with bluetooth connected as a trainer

### Version

2.11.0-rc

### Transmitter

FrSky X10 / X10S (ACCST)

frankiearzu, 2025-04-12:
I was testing a 2.11.0-RC3 build with Bluetooth on a TX16S.
Trainer configured as "Master/Bluetooth".
When I click "Discover".. it says "Scanning" and then I get "Emergency Mode".
I think is around the time when the BT module starts to send the remote address back.. if i don't have any other BT around.. it stays in "Scanning".  
Works well with 2.10.6.

stefan73, 2025-05-14:
I had the same effect last weekend with a Jumper T16 (mods: BLE Module on Aux2 and Flysky Hall gimbals) on 2.11
I tried to use the telemetry viewer app. Did not work well disconnected and reconnected all few times per minute. As I was in flight I let the cellphone doing it. After approx 15 min into the flight I started to get inactivity alarm audio messages. As I was clearly moving sticks, I pressed a button. That silenced the alarm. After landing I recognized that the radio UI was completely frozen. Had to pull the battery to shut it down. Quite happy that the sticks/switches remained working during this crash.