EdgeTX/edgetx #6058: Subtrim Values displayed incorrectly on Outputs main page

inventor7777, 2025-04-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When the subtrim value is set to something on the output settings, it shows as a much larger value on the outputs page. 

### Expected Behavior

The number should be the same on the outputs page as on the output setting page.

### Steps To Reproduce

Set a subtrim on an output and click back to the main outputs page. Limit numbers are fine but the subtrim value is way off.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

Windows, macOS

### Anything else?

I believe I’m running the 3/31 nightly. Behavior is also present on my dad’s MT12 which has a nightly about 3 weeks or so older.

Note that I have my radio set to always show ųs, or whatever the setting is called.

philmoz, 2025-04-07:
When PPM Unit = μs the list display is μs; but the edit is still %.

inventor7777, 2025-04-07:
That’s kind of what i figured, thanks for fixing it!

inventor7777, 2025-04-07:
@philmoz Why is it that subtrim is displayed in ųs but not the output min/max? e.g 988, 2012

(Or am i just misunderstanding the screenshots you sent)

philmoz, 2025-04-07:
> [@philmoz](https://github.com/philmoz) Why is it that subtrim is displayed in ųs but not the output min/max? e.g 988, 2012
> 
> (Or am i just misunderstanding the screenshots you sent)

Not sure what you mean. Both screenshots in the PR are with PPM Unit at the default (not μs).

inventor7777, 2025-04-08:
Oh, sorry. I didn’t notice that 😂 

Thank you!

PR #6064 (2025-04-10): fix(radio): inconsistent display for sub trim in outputs, bold indicators for bw128
Fixes #6058, fixes #6052

Changes (when PPM Unit = μs):
- B&W 128x64 - show subtrim edit value as μs
- Color - show subtrim edit value as μs
- Color - show subtrim value as μs on the list page

Wide screen B&W radios do not need any changes.
closed 2025-04-10 by pfeerick