EdgeTX/edgetx #6074: Text too long in  Modelmenu

bernd57, 2025-04-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In the model menu of BW version 2.10.6
the term range is cut off (German)

[image]

### Expected Behavior

I have not tried if it is ok in the English language in the menu

### Steps To Reproduce

Error is always there does not need to be created

### Version

2.10.6

### Transmitter

FrSky X9D+2019

pfeerick, 2025-04-20:
This is because the translation for German is too long... 

```cpp
#define TR_MODULE_RANGE        		   BUTTON(TR("Rng", "Reichweite")) 
```

Can you suggest an alternative for "Reichweite" that is either 5 or 6 characters (or both, so I can use the longest that fits, as the characters are not fixed width). The English firmware uses "Range" so does fit.

bernd57, 2025-04-20:
Hello pfeerick
It is difficult to replace the word “range” with a German term. I would therefore use the term “range”.
The German language now has many words that come from
English and are indispensable in the technical field.
That is my personal opinion

Greetings bernd

pfeerick, 2025-04-20:
@ParkerEde That make sense to you also for DE? i.e. 

```cpp
#define TR_MODULE_RANGE        		   BUTTON(TR("Rng", "Range")) 
```

bernd57, 2025-04-20:
The term “range” makes sense to me because I have a technical background. Whether this also makes sense for other German users
I can only guess. But many Germans now have to
with devices that have an English menu.

Bernd

ParkerEde, 2025-04-20:
@pfeerick 
I think on color radios the translation “Reichweite” fits perfectly on the button. So it seems to be only a B/W problem. What if the “Rng” part of the translation is used on the X9D?

[image]

ParkerEde, 2025-04-20:
Otherwise, we can also use
#define TR_MODULE_RANGE        		   BUTTON(TR("Rng", "Reichw"))

pfeerick, 2025-04-20:
Just double checked, and we still have the triple translation macro (bw128, bw212, colorlcd), so can make it all three :grin:

```
#define TR_MODULE_RANGE        		   BUTTON(TR3("Rng", "Reichw", "Reichweite")) 
```
or 
```
#define TR_MODULE_RANGE        		   BUTTON(TR3("Rng", "Range", "Reichweite")) 
```

Pick one! 😆 

Hm... looks like `TR_RECEIVER_NUM` is a bit _wonky_ for DE also 🤔

ParkerEde, 2025-04-21:
Great.
#define TR_MODULE_RANGE        		   BUTTON(TR3("Rng", "Reichw", "Reichweite"))

is okay

ParkerEde, 2025-04-21:
In 2.11.0-rc4 TR_RECEIVER_NUM is OK. The screenshot is from 2.10. 6. The input field was too wide.

PR #6123 (2025-04-23): fix(bw212): german text for "Range" too long
Also some minor file formatting. Fixes #6074

Is now deliberately shortened to `Reichw` for bw212, unchanged for bw128 and colorlcd. 
[image]
closed 2025-04-23 by pfeerick