EdgeTX/edgetx #6075: GX12 RC Command traces not smooth, irregular steps.

AlinVictor, 2025-04-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

RC command traces show irregular, inconsistent steps. I’ve noticed this issue over the past month on every recent nightly or release candidate firmware I’ve tested.

 - ADC filter is OFF in both global and model settings.

 - The behavior is present at any RC refresh rate, whether telemetry is enabled or disabled.

     - At 250Hz, irregularities appear roughly every 20ms

     - At 500Hz, irregularities appear roughly every 10ms

When reverting the radio to the firmware version it was shipped with, the RC traces appear normal and consistent, as expected.

[image]

### Expected Behavior

Depending on the set refresh rate, uniform and regular steps are expected.

### Steps To Reproduce

1. Analyze the log without smoothing enabled and observe the RC Command traces.
2. The issue is visible at any RC refresh rate, including 250Hz, 500Hz, etc.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster GX12

### Operating System (OS)

Windows

### Anything else?

[image]

3djc, 2025-04-13:
I think I recall something like this and the fix was a gimbal firmware upgrade from Radiomaster

AlinVictor, 2025-04-13:
Indeed, but RadioMaster gimbal firmware will fix/give option to the user to apply or not deadband on sticks. Because now there is a deadband on throttle stick. 
Will be nice to fix this problem also.

3djc, 2025-04-14:
So help me understand what you expect from us. You know the issue you mentioned is fixed by Radiometer in a gimbal firmware update, and cannot be fixed by EdgeTX, and now you mention another issue, that also cannot be fixed by EdgeTX

AlinVictor, 2025-04-14:
During discussions with RM technical support, concluded the following:
- deadband on throttle - will be fixed by future gimbal firmware released by RM
- Iregular steps - "expect EdgeTX developers to resolve the issues"

At this point, I appreciate your clear clarification. Not EdgeTx issue. 
Thanks!

3djc, 2025-04-14:
As far as I know, we were able to clearly reproduce the issue before the gimbal fix, but not after the update (and RM acknowledged it was an issue on their side). Are you using the updated version ?

3djc, 2025-04-14:
Spoke with RM. They have the fix ready, but are concerned that there might be very unlikely circumstances where gimbal could be bricked during an update. I have raised a PR addressing that concerns, that should allow them to release that firmware

3djc, 2025-04-18:
RM got what they needed to ensure safe updates of gimbal firmware should anything bad happen during flashing, so I'm confident they will provide the gimbal update you need now that it is safe. Nothing more we can do on Edgetx front I think

AlinVictor, 2025-04-18:
Great to hear that!
Waiting for RadioMaster to release new gimbal firmware now. 
Thank you!

3djc, 2025-04-21:
They are waiting for 2.11 release, so that they can instruct users to use 2.11 or later without having to go to nightlies

closed 2025-05-07 by AlinVictor

3djc, 2026-03-23:
> Almost an year later, i am v2.12 nightly, was on 2.11 before commenting. But either way, the issue hasn't been resolved and there is no public gimbal firmware from RM for GX12.

You should open a ticket with Radiomaster. Several have already updated their gimbals.