EdgeTX/edgetx #6078: BlueTooth Discovery/Scanning  go into Emergency Mode

frankiearzu, 2025-04-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hardware:
Radio:TX16S
BTModule: ESP32 BT module
EdgeTX: 2.11.0-RC3 build (April 11, 2025) with Bluetooth ON.

### Expected Behavior

Should show a menu of the Bluetooth devices to connect.

### Steps To Reproduce

1. Trainer configured as "Master/Bluetooth".
2. When I click "Discover".. it says "Scanning" and then I get "Emergency Mode".
I think is around the time when the BT module starts to send the remote address back to the GUI.. if i don't have any other BT around, it stays in "Scanning".
So, probably when creating the address list, or the GUI to display it.
Works well with 2.10.6.

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

frankiearzu, 2025-04-12:
After 2h of try/error by commenting/restoring parts of the code, this is what I found:

All the Bluetooth plumbing is correct.

The problem is creating the BTDiscoverMenu() without lines/items.  If I create a fake line in the beginning, I don't get the "Emergency Mode".
Probably crash when trying to display an empty Menu..

[image]

@philmoz  This one is for you...

philmoz, 2025-04-12:
I don't have any bluetooth enabled radios to test with; but I think I know where the problem is.

Before I create  PR, can you try adding this line at the start of the onDrawBegin() function in radio/src/gui/colorlcd/libui/menu.cpp:
```C
    if (lines.size() == 0) return;
```

frankiearzu, 2025-04-12:
That fixed it!!!  

[image]

PR #6079 (2025-04-12): fix(color): bluetooth setup menu crashes when discovering devices
Fixes #6078
closed 2025-04-12 by pfeerick