EdgeTX/edgetx #6081: 2.11.0-RC3 FUNC PLAYSOUND does not work in SPECIAL FUNCTION  call [GX12]

Gennnar0, 2025-04-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

**Device :** RadiomasterRC GX12
**Firmware Version :** 2.11.0-RC3 or 2.11.0 RM Build
**Current behavior :**
when i set a trigger to call the playsound function via a Special Function, there is no action - no sound is played. However, using the same trigger with playtrack function works correctly ("Ply Trk"  Function), and plays the expected sound.
I try different Warning, alarm, Beep, Robot... same thing.
thanks

### Expected Behavior

I would like that play sound function works properly :)
thanks

### Steps To Reproduce

1. create a trigger, in SPECIAL FUNCTION tab
2. wait for that trigger goes up
3. listen sound

### Version

2.11.0-rc

### Transmitter

RadioMaster GX12

### Operating System (OS)

Windows

### OS Version

Windows 11 64

pfeerick, 2025-04-12:
Not reproducible here with GX12 and 2.11.0-RC3, I can play any of the beep, warning, alarm, robt, SciFi, etc sounds. Have you ensured that the enable checkbox on the far right is ticked? The "Beep volume" setting in in Radio Settings/SYS -> Radio Setup -> Sound does have some influence over the volume of Play Sound, since these are generated sounds, but even at it's lowest, if you can hear other play track sounds, you should be able to hear those ones.

Gennnar0, 2025-04-13:
Hi pfeerick,

I 've checked all settings and i saw something about page 3, RADIO SETUP.
Volume wave is ok and volume beep is ok.
But sound mode was set to alarm, not to 'nokey' or 'all'.
Now it's ok
Thank you

closed 2025-04-13 by Gennnar0