EdgeTX/edgetx #6083: FrSky X10S full negative spikes on channel output

elekSENDER, 2025-04-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The following description occurs on my FrSky X10S with v2.11.0-RC1/2/3.
While flying I noticed sometimes strange twitches on a heli at the nick-channel. That means in my TAER configuration CH3.
Because this flight behavior was definitely not controlled by stick input I made some tests and look also in "min., max. and range" in Analogs to see what happend.

I saw there values which cannot be produced by stickmovement.

Benchtest with all sticks in the middle:
[image]

Min/max controlled by stick:
[image]

These twitches occure every couple of seconds to few minutes without touching the radio:
[image]

[image]

At first I thought that it was an issue with my radio, maybe the hall sensor defective which cause the glitches.
Then I go back to v2.10.6 and for in summary about 4 hours bench test it never ever happened a spike again. And never recognized while some hours of flying with v2.10.6.
Reloaded to v2.11.0-rc3 the described twitches are immediately back.

It seems to me there is a difference in the channel signal processing between 2.10 an 2.11 which causes this spikes on a (my?) X10s.
For sure this could be an isolated case, but maybe someone can find something similar or a solution to my problem.

### Expected Behavior

No spikes on channel output.

### Steps To Reproduce

- Load any version of 2.11.0-RC firmware on the X10s
- Go to "min., max. and range" in Analogs, don´t touch radio and wait some time.
- Watch for values which cannot be produced by stick movement.

### Version

2.11.0-rc

### Transmitter

FrSky X10 / X10S (ACCST)

### Anything else?

Log from a heliflight with very recognisable nick-twitches:

[image]

Log from bench test:

[image]

robertw2a, 2025-05-10:
I am having the same issue with my X10S Express when using 2.11 release.
The elevator channel channel output momentarily pulses to -100 and then returns to normal.
This behavior keep repeating. It is visible on the channel output, monitor screens and the actual servo.
It only appears to be the elevator channel that is showing this issue 

Switched back to 2.10.6 and it works as it should.

elekSENDER, 2025-05-12:
> I am having the same issue with my X10S Express when using 2.11 release.

Good to hear, because so it seams the hardware of my radio is ok.

The last days I swaped the gimbals on my radio and the same occured: negative spikes on 03 in Analogs (and of course in servomonitor and servos itself), so the gimbal was eliminated as a source of error.

As well a change of "Default channel order" from TAER to anything or "Mode" from 1 to 4 causes no change, it is always and only 03 with the twitches.

If anybody detects this on his X10 I really recommend not to fly with any version of 2.11.0 (rc1-rc4, released), it´s pretty not much fun to land a heli with full negative nick twitches. ;-) 

Rolling back to 2.10.6 this issue never ever occcured.

hpolig, 2025-05-18:
Same problem here:

Last year I tried EdgeTx 2.10.0 and had my existing models converted, but didn't do anything else
because the telemetry screens were not included and it was simply too much work for me to recreate them all.

This time I was determined to switch to EdgeTx 2.11.0 and equip all models with a standardized telemetry screen. for all models.

During the test with my Zorca E (I also have a pure glider version) I noticed an irregular twitching of the tail servos.
The twitching also appeared in the channel overview and even in the calibration screen I could see short irregular deflections of the right stick from center to bottom of the right stick from center to bottom (i.e. down elevator). A check of the Hall sensor connections revealed no error.
Calibrating the system did not change anything either.

So I flashed back to OpenTX 2.3.15 for testing and the error is gone.
I went back to EdgeTx and the error is there again, so it is clearly due to EdgeTx.
Incidentally, the error also occurs with other models (Silence Peak, Polaris, Zorca).

My Transmitter is a Frsky X10S from 2017 with later added ISRM modul.

Merga189, 2025-05-19:
I can not reproduce that issue on my FrSky X10S Express with 2.11.1

[image]

TAER and channel 3.

Is that now an issue only for the pre Express X10 radios or what?? The posts here are confusing ....

hpolig, 2025-05-19:
I have a non-express X10S with that problem

raphaelcoeffic, 2025-05-19:
> did you change a PWM driver or something like that?

I don't think the code has been really changed btw. 2.10 and 2.11, but that might be some side effect of another change.

> I can not reproduce that issue on my FrSky X10S Express with 2.11.1

The PWM code has been tightened to improve checks btw. 2.11.0 and 2.11.1 (see https://github.com/EdgeTX/edgetx/issues/6161), which may or may not have fixed something.

elekSENDER, 2025-05-19:
I tried 2.11.1 yesterday with my X10S with upgrade ISRM-module.
It has gotten significantly worse in comparision to 2.11.0, now you can see the twitches sometimes twice a second, counted 20-25 times per minute.

raphaelcoeffic, 2025-05-19:
Ok, as far as I can see, there are **no changes at all** between 2.10.6 and 2.11.0 in the PWM gimbal driver. So I'm having hard times to understand where this is coming from.

Merga189, 2025-05-19:
I am a bit confused. 

robertw2a wrote

> I am having the same issue with my X10S Express when using 2.11 release.
> I am using AETR channel order.
> The elevator channel channel output momentarily pulses to -100 and then returns to normal.
> This behavior keep repeating. It is visible on the channel output, monitor screens and the actual servo.
> It only appears to be the elevator channel that is showing this issue

I have the Express with 2.11.1 TAER in mode 3 and don't see that issue. Is the Express affected at all? Is that a mode thing?

Changed from mode 3 to mode 2 for a short test. No spikes on channel 3 ....

elekSENDER, 2025-05-19:
@raphaelcoeffic  > is that actually always the same gimbal where that happens? (elevator; right side vertical)

> The last days I swaped the gimbals on my radio and the same occured: negative spikes on 03 in Analogs (and of course in servomonitor and servos itself), so the gimbal was eliminated as a source of [...]

No, it was not always the same gimbal as I described above.
With both gimbals the same issue on the same "channel" 03: elevator, right side vertical, 
Twitches only from middle to direction zero respectivly minus-values in servomonitor.

hpolig, 2025-05-19:
The test firmware is even worse, 1 to 2 ticks per second with sometimes a pause of max 2 seconds.

elekSENDER, 2025-05-19:
same on mine X10S, 10-15 spikes per minute.

raphaelcoeffic, 2025-05-19:
Ok, here I shuffled the location in RAM (code is the exact same):
[x10-test2-issue-6083.zip](https://github.com/user-attachments/files/20297489/x10-test2-issue-6083.zip)

elekSENDER, 2025-05-19:
Better, but not gone, 2-5 times per minute.
A little bit like 2.11.0.  ;-)

raphaelcoeffic, 2025-05-19:
Ok, let's try something different: this time I added a better validation of the pulses detected (not tested at all on any hardware):
[x10-issue-6083-validate-pulses.zip](https://github.com/user-attachments/files/20298149/x10-issue-6083-validate-pulses.zip)

elekSENDER, 2025-05-19:
Looks very good, I made a 20 minute benchtest and no spikes at all, just like 2.10.6.

I also had the idea of anything inducing, so I tested it with the gimbals out of the housing, the gimbal cable straight conected to the board, completely free of anything else. No difference, same issue as everytime with both gimbals.
As well speak against inducing that there are no issues with 2.10.6.
But if anyone could test this better than my "external gimbal test", would be a good crosscheck.

Is this firmware safe for a real flighttest?

hpolig, 2025-05-19:
Agree, regarding the pulses it seems to work fine.

robertw2a, 2025-05-20:
My radio is X10S Express. Can you please provide the test firmware with this fix for to verify if this fixes the issue.

Thanks

raphaelcoeffic, 2025-05-20:
> Can you please provide the test firmware with this fix for me to verify if this fixes the issue.

Please use the firmware built in #6283 

> Is this firmware safe for a real flighttest?

The patch is directly on top of current 2.11 branch, so it *should* be ok. Please test before flying ;-)

elekSENDER, 2025-05-20:
I just flew 2 heli-flights with 9 minutes duration and there was not any issue.
For my X10S this _should_ be the solution. ;-)

Great job, thanks a lot!

robertw2a, 2025-05-21:
It fixed the issue on my X10S Express.
Thanks for the quick fix.

closed 2025-05-23 by pfeerick

ABLomas, 2025-06-23:
Need clarification - i see "fix" is merged, so i should see results in nightlies?
If so, then this does not fix issue for my radio. I have X10S carbon and after upgrade noticed twitches on pitch axis. Unplugged entire gimbal - they "migrated" to another gimbal.
Tried firmware from https://github.com/EdgeTX/edgetx/issues/6083#issuecomment-2891167057 - analog inputs on gimbals do not work at all.
Tried 06.22 nightly - all channels twitch (including switches).
Rolled back to 2.10.6 - no issues, works OK

So - maybe my hardware is a bit different from others (how many X10S revisions are here) - or fix is not merged and nightly builds built without fix?

hpolig, 2025-06-23:
Same here:

Thank of the adapted version 2.11.1 for the X10S (fault pulses on the right stick) I installed them and yesterday
spent half the day on holiday with the adaptation of telemetry, logical switches etc. of the brought models.

The first annoying error is that when selecting the telemetry sensors for the logical switches, always the same completely inappropriate Source selection appears and the correct sources are not selectable. I avoided that with Companion.

However, it was then lost to me when the calibration of the sticks was lost twice and could not be renewed.

Video on youtube: https://youtu.be/Hu8iHNfxI6U

ABLomas, 2025-06-24:
@raphaelcoeffic can you please clarify - which binary should be flashed for X10S to test it?
I also tested fresh nightly, no difference from earlier - all channels twitch.

ABLomas, 2025-06-29:
2.11.2 was released but, like i posted above - issue not resolved.
On 2.11.2 and X10S all inputs are twitching (on analog debug menu), radio becomes unusable.
With 2.10 all fine, so it's not HW fault.

Will open new issue as i can't re-open this

ABLomas, 2025-06-30:
> [@ABLomas](https://github.com/ABLomas) do you have a X10S (express or not, doesn't matter)?

X10S, not express (carbon version if this does matter)

Yups1337, 2025-11-02:
As mentioned above, 2.11.2 works fine. 
Please install this version and do a stick calibration.

PR #6283 (2025-05-23): fix(radio): verify PWM gimbal pulse length
Fixes #6083
PR #6284 (2025-05-23): fix(radio): verify PWM gimbal pulse length
Fixes #6083