EdgeTX/edgetx #6088: Radiomaster GX12  -Extended limits not working

nen77, 2025-04-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The signal should follow the channel percentage even beyond ±100% (up to ±150%) when Extended Limits are enabled, allowing proper calibration with flight controllers like MSH Brain.

### Expected Behavior

Expected Behavior:
The signal should follow the channel percentage even beyond ±100% (up to ±150%) when Extended Limits are enabled, allowing proper calibration with flight controllers like MSH Brain.

Actual Behavior:
The signal appears capped at ±100%, even when set beyond that in the mixer.

### Steps To Reproduce

Power on the Radiomaster GX12 running EdgeTX.

Create a new model or open an existing one.

Go to the Model Settings > Limits page.

Enable the Extended Limits option.

Go to the Outputs or Mixer page.

Set one of the channels (e.g., CH1) to a value greater than ±100%, such as +125% or -110%.

Observe the actual channel output using:

Telemetry feedback (e.g., flight controller input)

Channel monitor on the radio

The output will still cap at ±100%, despite the configured percentage being higher

### Version

2.11.0-rc

### Transmitter

RadioMaster GX12

### Operating System (OS)

Windows

3djc, 2025-04-14:
Isn't that more an ELRS limit then ?

nen77, 2025-04-14:
> Isn't that more an ELRS limit then ?

Absolutely not. In the version it came with, the transmitter was working, but it would randomly freeze. _**It also works properly on the Radiomaster Zorro with ELRS, and on the TX16S with ELRS**_

3djc, 2025-04-14:
I correctly get -150 to 150 on the channel monitor, are you not getting that ?

[image]

[image]

nen77, 2025-04-14:
> I correctly get -150 to 150 on the channel monitor, are you not getting that ?
> 
> <img alt="Image" width="559" src="https://private-user-images.githubusercontent.com/5167938/433252478-8611a590-df44-44da-ab76-b0503d99d573.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQ2MTQwNjEsIm5iZiI6MTc0NDYxMzc2MSwicGF0aCI6Ii81MTY3OTM4LzQzMzI1MjQ3OC04NjExYTU5MC1kZjQ0LTQ0ZGEtYWI3Ni1iMDUwM2Q5OWQ1NzMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDQxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA0MTRUMDY1NjAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjQxYWFjODQwZmZjYWYzOTFiMmIyY2E2ZGNhYmIyOWIzNjA5OWZmMDU1MjJjNDZmMWFlYWUyYmRiMzk0MmFlMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ghZQ1dtQtkUfXcOWKzSCasZHt3VvHZhuotS0RKDDJNc"> <img alt="Image" width="556" src="https://private-user-images.githubusercontent.com/5167938/433252569-a3ccb6a4-f260-48f8-8b6f-8335df007a68.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQ2MTQwNjEsIm5iZiI6MTc0NDYxMzc2MSwicGF0aCI6Ii81MTY3OTM4LzQzMzI1MjU2OS1hM2NjYjZhNC1mMjYwLTQ4ZjgtOGI2Zi04MzM1ZGYwMDdhNjgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDQxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA0MTRUMDY1NjAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Nzk5NzVlODViZmZiMWVkMGMxNWYxYTUxNzg0NmI1MDcwM2VjYmMzZmRlMDg2YzFhZTk0NzIyYTRhYTE2Y2E4NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.WHeMeHaUdKy_xJDly_xbBaY9qYeDruwCOIRzaqQIAUQ">

Yes, the signal is modified on the monitor, but in the receiver, the signal is not modified – it remains at the original value. However, the modification works when the value is below 100

pfeerick, 2025-04-14:
Since you've got 150% showing, you've obviously also increased the min/max values for the output for those channels (and not just enabled extened limits, and increased weight on the channel), else they would be clipped on handset at the output stage. Are you using either 100Hz or 333Hz full resolution packet modes for ELRS? As those are the only ones that are capable of going under/over 100%, to a maximum of -120% / 120% (885-2115us). If you use any other packet rate, you will be limited to -100% / 100% (988 - 2012us). 

https://www.expresslrs.org/software/switch-config/#full-resolution-switch-configuration-modes

nen77, 2025-04-14:
> Since you've got 150% showing, you've obviously also increased the min/max values for the output for those channels (and not just enabled extened limits, and increased weight on the channel), else [...]

Thank you! I will test later today and come back with feedback. It might be that you are right.

nen77, 2025-04-15:
> Since you've got 150% showing, you've obviously also increased the min/max values for the output for those channels (and not just enabled extened limits, and increased weight on the channel), else [...]

thank you a lot, this was the solution. Best regards!

closed 2025-04-15 by nen77