EdgeTX/edgetx #6092: Use flightmodes as switches

wimalopaan, 2025-04-15:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I would like to use the active filghtmode as a switch.

### Describe the solution you'd like

Add a switch for every flightmode. These switches are on if the corresponding flightmode is active. These switches would appear as normal switches like physical switches or logical switches.

philmoz, 2025-04-18:
This already exists. (For flight modes > 0 only those with a defined switch are listed.)

[image]

wimalopaan, 2025-04-18:
Oh yes, my bad. I was pretty sure that I had used that before but couldn't find the way ...

closed 2025-04-18 by wimalopaan