EdgeTX/edgetx #6101: ETX 2.10.6 Inconsistency with lcd.sizeText between telemetry and widget scripts.

RC-SOAR, 2025-04-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

The context is that I am writing a script for dual use - it can run either as a telemetry script (mono screens) or a widget (colour). 

In the init() callback I have a call to lcd.sizeText()  in order to cache the font height.  Using the Companion simulator (TX16S), this works fine when runing as a widget. However in telemetry mode (TX12) it fails because lcd.sizeText is nil. 

It would be useful to have lcd.sizeText() available in the init() function in telemetry scripts - this would allow screen layouts to be planned up front as well as being consistant with behaviour in widgets.

### Expected Behavior

See above

### Steps To Reproduce

1. Create a simple telemetry script
2. In the Init() function, attempt to call lcd.sizeText()

### Version

2.10.6

### Transmitter

RadioMaster TX12 / TX12MK2

### Operating System (OS)

Windows

### OS Version

10

pagrey, 2025-04-17:
lcd.sizeText() isn't supported at all on B&W screens regardless of where it is called. There are a limited set of functions available on B&W screens, I think you'll always have to check what you're running on and draw accordingly.

RC-SOAR, 2025-04-17:
Aha, that explains it - thanks!

pfeerick, 2025-04-23:
So where do we stand with this issue? `lcd.sizeText()` is (and has always been) a colorlcd only function, so this isn't a bug, but incorrect use of it on a non-colorlcd platform.

RC-SOAR, 2025-04-23:
The documentation should be more explicit, there's no mention of restrictions. https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/lcd-functions-less-than-greater-than-luadoc-begin-lcd/sizetext

pfeerick, 2025-04-23:
Ok, for now it will have to be the documentation.