EdgeTX/edgetx #6102: Jumper T15 -6 Button LEDS No longer show charging status. OK with 2.10.5,  Broke in 2.11.0 RC3

SteveintheIoW, 2025-04-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The 6 button LEDs on Jumper T15 are used to show charging status changing from Red to White as charging progresses.

during testing of 2.11.0 RC3 i have noticed these LEDs no longer illuminate (all off) when the charging USB is connected.    I tried reverting back to 2.10.5 and LED functionality returned OK.  Changing back again to to RC3 there is No LED status.

This seems like a serious bug  - how do you visually monitor status during charging ! 

### Expected Behavior

The 6 LEDs usually show charging  progress incrementally changing from Red to White as changing progresses

### Steps To Reproduce

With Tx switched off, apply a USB C power source to Tx.  
The 6 Button LEDS should illuminate and show charge status

### Version

2.11.0-rc

### Transmitter

Jumper T15

pfeerick, 2025-04-18:
This is not a bug, but a deliberate change via #5512. The voltage sense for reading the battery voltage is too inaccurate, so it is better to use the built-in charge led indicator. If you need to see state of charge, unplug the radio, and power it on.

closed 2025-04-18 by pfeerick

Mick51, 2025-05-15:
We want to see this feature come back please.

druckgott, 2025-05-18:
same for me since 2.11.* the leds are not any more working, but it was quite well to se a rought status please wring it bake.