EdgeTX/edgetx #6112: Flysky el18 Switch Funktion lost

Boxxxer64, 2025-04-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Flysky EL18 EdgeTX 2.11RC3. The switches stop working at random intervals. Sometimes for 20 minutes, sometimes for an hour. It's very dangerous if the copter won't let you disarm.

### Expected Behavior

Flysky EL18 EdgeTX 2.11RC3. The switches stop working at random intervals. Sometimes for 20 minutes, sometimes for an hour. It's very dangerous if the copter won't let you disarm.

### Steps To Reproduce

Dont know 

### Version

2.11.0-rc

### Transmitter

Flysky EL18

### Operating System (OS)

macOS

### OS Version

Latest

Boxxxer64, 2025-04-20:
Video

Boxxxer64, 2025-04-20:
https://youtube.com/shorts/sNU78yLwD00?si=tDT54h76nSo4V6-q

richardclli, 2025-04-21:
The symptom can be related to hardware cabling connectivity problem, did you check this?
Can downgrading to 2.10 solves the problem?

Boxxxer64, 2025-04-21:
I don't think it's a hardware issue. All switches are affected, and the rear ones are connected to a different board. I'll have to test whether a downgrade fixes the problem. Apparently, there was a problem with 2.11 that was supposed to be fixed with RC3. I'll try upgrading to 2.10 in the next few days.

Boxxxer64, 2025-04-23:
With 2.10.6 The hole day no Problem. No Switch Problem.

richardclli, 2025-04-28:
Interesting though, let me test this when I have spare time.

richardclli, 2025-05-12:
Is the symptom the same as #6200?

richardclli, 2025-05-13:
@RustyWraith Follow up your problem here

richardclli, 2025-05-13:
Please test the release version 2.11.0 to see if the problem persists.  Or wait for the 2.11.1 releases, should be out soon.

RustyWraith, 2025-05-13:
I can confirm the issue with 2.11.0 and NV14

richardclli, 2025-05-14:
> I can confirm the issue with 2.11.0 and NV14.

Can the problem be reproduced by switching on the radio and wait for 1-2 hours?

richardclli, 2025-05-15:
Check if 2.11.1 fixes the problem.  There are quite a few critical bugs and fixed in 2.11.1, so not sure if the problem is gone or not.

richardclli, 2025-05-15:
> > > I can confirm the issue with 2.11.0 and NV14.
> > 
> > Can the problem be reproduced by switching on the radio and wait for 1-2 hours doing nothing? Or requires one to move.the gimbals/switches to reproduce?
> 
> I've kept the radio on for 3 hours without touching anything. Switches were working.

So one must operate the radio to repeat the problem, right?

RustyWraith, 2025-05-15:
> So one must operate the radio to repeat the problem, right?

I've tested the radio two times for about 5-6 hours total. It was sitting on the desk on batteries without any input.
Worked fine. It could be just luck, though. The pattern to reproduce is unknown for me.

RustyWraith, 2025-05-15:
> Check if 2.11.1 fixes the problem. There are quite a few critical bugs and fixed in 2.11.1, so not sure if the problem is gone or not.

I've installed todays nightly via ETX buddy. I'll report if issue persist or not.

RustyWraith, 2025-05-27:
Can confirm the bug persists in 2.11.1.

gagarinlg, 2025-06-01:
I have an EL18 running since 16 hours now, with an external ELRS module connected and the error did not happen.
Could someone please share their radio config, model and widgets with me, so that I can test that specific setup?

richardclli, 2025-07-03:
Just remember one thing that worth to check.  So for those who are able to reproduce this.  Please reproduce the problem and check the hardware debug screen and see if the switches all malfunctioned?  Or the moving sa will show sb moving, etc.