EdgeTX/edgetx #6121: Allow theme creators to customize channel widget background and text colors

siyeongjang, 2025-04-22:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

### Problem

Even when a custom theme is applied, the channel widget (Channel Monitor), commonly used on the main screen, always displays a blue background and black text.  
These colors do not match the rest of the theme and often look visually out of place.

### Describe the solution you'd like

### Proposed Solution

I would like to suggest allowing the channel widget (Channel Monitor) colors to be defined in the `theme.yaml` file.

For example, the following keys could be supported:

- `CHANNEL_BG`: background color of the channel bars  
- `CHANNEL_FILL`: fill color of the active value  
- `CHANNEL_TEXT`: color of the channel label text  

If these keys are not defined in the theme file, the system could fall back to the current default colors (e.g., blue and black).  
This would greatly improve the flexibility and visual consistency for theme designers.

If you could point me to the relevant source file(s) and a suggested testing method for this feature, I would be happy to try implementing it myself and submit a PR.

### Describe alternatives you've considered

There are no effective alternatives, as these colors are currently hardcoded in the firmware and cannot be overridden through theming or user settings.

philmoz, 2025-04-22:
The background color, text color and bar color in the 'Outputs' widget are set in the widget settings.
By default these will be theme colors (SECONDARY3, PRIMARY1 and SECONDARY1 respectively).
If you change theme the outputs widgets colors will automatically change to the new theme color.

If you have manually changed any of the three color values in the widget settings to non-theme colors then your chosen colors will remain when changing theme.

pfeerick, 2025-04-23:
As stated by @philmoz, by default the "Outputs" widget defaults to the SYS palette colours (which the colours for the active theme are shown on the left of the image below). It doesn't "hard code" the colour , but rather set which of the theme specific placeholders will be used...

[image]

...so by default, the widget *will* follow theme colours as set. And if you set them explicitly in the widget options, the widget will honour that setting instead. 

So, the premise of this issue is incorrect. So, is the current level of customisation sufficient, or do you think more is needed (still)? If so, `/radio/src/gui/colorlcd/themes/theme_manager.cpp` is probably a good place to start ;)

siyeongjang, 2025-04-23:
Oh, I didn’t realize that feature was already there — thanks for pointing it out!  
Looks like my suggestion wasn’t needed after all. Appreciate the quick reply!

pfeerick, 2025-04-23:
No problem. If you do think there are areas that could be improved with theming, please don't hesitate to make other suggestions. If you are on Discord, can also discuss it there.