EdgeTX/edgetx #6139: Failed to build EdgeTX on Ubuntu 20.04

snowuyl, 2025-04-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Build system

### Current Behavior

git clone --recursive -b main https://github.com/EdgeTX/edgetx.git edgetx
cd edgetx
./tools/setup_buildenv_ubuntu20.04.sh
mkdir build
cd build
cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
make configure
=== Step 1: Checking if i386 requirement is satisfied ===
i386 requirement satisfied!
=== Step 2: Updating Ubuntu package lists. Please provide sudo credentials, when asked ===
[sudo] password for snowuyl: 
Err:1 http://tw.archive.ubuntu.com/ubuntu focal InRelease
  Temporary failure resolving 'tw.archive.ubuntu.com'
Err:2 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu focal InRelease      
  Temporary failure resolving 'ppa.launchpad.net'
Err:3 http://security.ubuntu.com/ubuntu focal-security InRelease                   
  Temporary failure resolving 'security.ubuntu.com'
Err:4 https://www.scootersoftware.com/debian bcompare5 InRelease                   
  Temporary failure resolving 'www.scootersoftware.com'
Err:5 https://dl.google.com/linux/chrome/deb stable InRelease                      
  Temporary failure resolving 'dl.google.com'
Err:6 https://packages.microsoft.com/repos/code stable InRelease       
  Temporary failure resolving 'packages.microsoft.com'
Err:7 http://tw.archive.ubuntu.com/ubuntu focal-updates InRelease
  Temporary failure resolving 'tw.archive.ubuntu.com'
Err:8 http://tw.archive.ubuntu.com/ubuntu focal-backports InRelease
  Temporary failure resolving 'tw.archive.ubuntu.com'
Reading package lists... Done
W: Failed to fetch http://tw.archive.ubuntu.com/ubuntu/dists/focal/InRelease  Temporary failure resolving 'tw.archive.ubuntu.com'
W: Failed to fetch http://tw.archive.ubuntu.com/ubuntu/dists/focal-updates/InRelease  Temporary failure resolving 'tw.archive.ubuntu.com'
W: Failed to fetch http://tw.archive.ubuntu.com/ubuntu/dists/focal-backports/InRelease  Temporary failure resolving 'tw.archive.ubuntu.com'
W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/focal-security/InRelease  Temporary failure resolving 'security.ubuntu.com'
W: Failed to fetch https://dl.google.com/linux/chrome/deb/dists/stable/InRelease  Temporary failure resolving 'dl.google.com'
W: Failed to fetch https://www.scootersoftware.com/debian/dists/bcompare5/InRelease  Temporary failure resolving 'www.scootersoftware.com'
W: Failed to fetch http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu/dists/focal/InRelease  Temporary failure resolving 'ppa.launchpad.net'
W: Failed to fetch https://packages.microsoft.com/repos/code/dists/stable/InRelease  Temporary failure resolving 'packages.microsoft.com'
W: Some index files failed to download. They have been ignored, or old ones used instead.
=== Step 3: Installing packages ===
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gcc is already the newest version (4:9.3.0-1ubuntu2).
libusb-1.0-0-dev is already the newest version (2:1.0.23-2build1).
pv is already the newest version (1.6.6-1).
python-is-python3 is already the newest version (3.8.2-4).
zip is already the newest version (3.0-11build1).
clang-7 is already the newest version (1:7.0.1-12).
libclang-6.0-dev is already the newest version (1:6.0.1-14).
libfox-1.6-dev is already the newest version (1.6.57-1build1).
libgtest-dev is already the newest version (1.10.0-2).
libqt5serialport5-dev is already the newest version (5.12.8-0ubuntu1).
libqt5svg5-dev is already the newest version (5.12.8-0ubuntu1).
libsdl2-dev is already the newest version (2.0.10+dfsg1-3).
npm is already the newest version (6.14.4+ds-1ubuntu2).
openocd is already the newest version (0.10.0-6build1).
python-clang-7 is already the newest version (1:7.0.1-12).
python-pil is already the newest version (6.2.1-3).
qtcreator is already the newest version (4.11.0-2build2).
qtmultimedia5-dev is already the newest version (5.12.8-0ubuntu1).
qttools5-dev is already the newest version (5.12.8-0ubuntu1).
qttools5-dev-tools is already the newest version (5.12.8-0ubuntu1).
stlink-tools is already the newest version (1.6.0+ds-1).
build-essential is already the newest version (12.8ubuntu1.1).
cmake is already the newest version (3.16.3-1ubuntu1.20.04.1).
git is already the newest version (1:2.25.1-1ubuntu3.14).
lib32ncurses-dev is already the newest version (6.2-0ubuntu2.1).
lib32z1 is already the newest version (1:1.2.11.dfsg-2ubuntu1.5).
python3-setuptools is already the newest version (45.2.0-1ubuntu0.2).
python3-tk is already the newest version (3.8.10-0ubuntu1~20.04).
software-properties-common is already the newest version (0.99.9.12).
wget is already the newest version (1.20.3-1ubuntu2.1).
python-pip-whl is already the newest version (20.0.2-5ubuntu1.11).
python3-pip is already the newest version (20.0.2-5ubuntu1.11).
qt5-default is already the newest version (5.12.8+dfsg-0ubuntu2.1).
libncurses5:i386 is already the newest version (6.2-0ubuntu2.1).
libpython2.7:i386 is already the newest version (2.7.18-1~20.04.7).
The following packages were automatically installed and are no longer required:
  binutils-aarch64-linux-gnu cpp-9-aarch64-linux-gnu cpp-aarch64-linux-gnu
  gcc-10-cross-base gcc-9-aarch64-linux-gnu-base gcc-9-cross-base
  libasan5-arm64-cross libatomic1-arm64-cross libc6-arm64-cross
  libc6-dev-arm64-cross libgcc-9-dev-arm64-cross libgcc-s1-arm64-cross
  libgomp1-arm64-cross libitm1-arm64-cross liblsan0-arm64-cross
  libstdc++6-arm64-cross libtsan0-arm64-cross libubsan1-arm64-cross
  linux-libc-dev-arm64-cross usb-modeswitch usb-modeswitch-data
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 103 not upgraded.
=== Step 4: Installing Python packages ===
Requirement already satisfied: filelock in /usr/local/lib/python3.8/dist-packages (3.16.1)
Requirement already satisfied: asciitree in /usr/local/lib/python3.8/dist-packages (0.3.3)
Requirement already satisfied: jinja2 in /usr/local/lib/python3.8/dist-packages (3.1.6)
Requirement already satisfied: pillow==7.2.0 in /usr/local/lib/python3.8/dist-packages (7.2.0)
Requirement already satisfied: clang==6.0.0 in /usr/local/lib/python3.8/dist-packages (6.0.0)
Requirement already satisfied: future in /usr/lib/python3/dist-packages (0.18.2)
Requirement already satisfied: lxml in /usr/local/lib/python3.8/dist-packages (5.3.2)
Requirement already satisfied: lz4 in /usr/local/lib/python3.8/dist-packages (4.3.3)
Requirement already satisfied: MarkupSafe>=2.0 in /usr/local/lib/python3.8/dist-packages (from jinja2) (2.1.5)
=== Step 5: Fetching GNU Arm Embedded Toolchains ===
mkdir: cannot create directory ‘build’: File exists
-- Python found, version: 3.8.10
-- EdgeTX 3.0.0-selfbuild @ bf3c2209
-- Internal GPS enabled
-- Adding support for USB serial
-- Adding support for PXX1 as internal module
-- Adding support for PXX2 as internal module
-- Adding support for MULTI as internal module
-- Adding support for CRSF as internal module
CPP Compiler: /usr/bin/c++, version 9.4.0
CMake Error at radio/src/CMakeLists.txt:511 (message):
  Only Arm GNU Toolchain version 14.2.rel1 is supported!!

-- Configuring incomplete, errors occurred!
See also "/home/snowuyl/samba/workspace_EdgeTx/edgetx/CMakeFiles/CMakeOutput.log".
See also "/home/snowuyl/samba/workspace_EdgeTx/edgetx/CMakeFiles/CMakeError.log".
make: *** No rule to make target 'configure'.  Stop.
~/samba/workspace_EdgeTx$ gcc --version
gcc (GCC) 14.2.0
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

### Expected Behavior

Can build EdgeTX on Ubuntu 20.04 successfully.

### Steps To Reproduce

git clone --recursive -b main https://github.com/EdgeTX/edgetx.git edgetx
cd edgetx
./tools/setup_buildenv_ubuntu20.04.sh
mkdir build
cd build
cmake -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
make configure

### Version

2.11.0-rc

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Linux

### OS Version

Ubuntu 20.04

pfeerick, 2025-04-25:
It looks like you are trying to build 2.11 using the 2.10 toolchain, which is not supported?

snowuyl, 2025-04-25:
Thanks for your reply! Could you kindly to let me know how to install Arm GNU Toolchain version 14.2.rel1 on Ubuntu 20.04?

pfeerick, 2025-04-25:
Have a look at #4314 as I mentioned above. You will need to also install a later version of QT5, which is done there also IIRC.

snowuyl, 2025-04-25:
I have upgraded Qt5 to v5.12.8 and using EdgeTx 2.11 branch to build again. 
cd edgetx
git branch
qmake --version
./tools/setup_buildenv_ubuntu20.04.sh
mkdir build
cd build
cmake -DARM_TOOLCHAIN_DIR=/opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
make configure
 2.11
  main
QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
=== Step 1: Checking if i386 requirement is satisfied ===
i386 requirement satisfied!
=== Step 2: Updating Ubuntu package lists. Please provide sudo credentials, when asked ===
[sudo] password for anddyueng: 
Hit:1 https://dl.google.com/linux/chrome/deb stable InRelease                      
Hit:2 https://packages.microsoft.com/repos/code stable InRelease                   
Hit:3 https://download.docker.com/linux/ubuntu focal InRelease                     
Hit:4 http://security.ubuntu.com/ubuntu focal-security InRelease                  
Hit:5 http://tw.archive.ubuntu.com/ubuntu focal InRelease                   
Hit:6 http://tw.archive.ubuntu.com/ubuntu focal-updates InRelease           
Hit:7 http://tw.archive.ubuntu.com/ubuntu focal-backports InRelease         
Hit:8 http://ppa.launchpad.net/ubuntu-toolchain-r/test/ubuntu focal InRelease      
Ign:9 https://www.scootersoftware.com/debian bcompare5 InRelease
Hit:10 https://www.scootersoftware.com/debian bcompare5 Release
Reading package lists... Done
=== Step 3: Installing packages ===
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gcc is already the newest version (4:9.3.0-1ubuntu2).
libusb-1.0-0-dev is already the newest version (2:1.0.23-2build1).
pv is already the newest version (1.6.6-1).
python-is-python3 is already the newest version (3.8.2-4).
zip is already the newest version (3.0-11build1).
clang-7 is already the newest version (1:7.0.1-12).
libclang-6.0-dev is already the newest version (1:6.0.1-14).
libfox-1.6-dev is already the newest version (1.6.57-1build1).
libgtest-dev is already the newest version (1.10.0-2).
libqt5serialport5-dev is already the newest version (5.12.8-0ubuntu1).
libqt5svg5-dev is already the newest version (5.12.8-0ubuntu1).
libsdl2-dev is already the newest version (2.0.10+dfsg1-3).
npm is already the newest version (6.14.4+ds-1ubuntu2).
openocd is already the newest version (0.10.0-6build1).
python-clang-7 is already the newest version (1:7.0.1-12).
python-pil is already the newest version (6.2.1-3).
qtcreator is already the newest version (4.11.0-2build2).
qtmultimedia5-dev is already the newest version (5.12.8-0ubuntu1).
qttools5-dev is already the newest version (5.12.8-0ubuntu1).
qttools5-dev-tools is already the newest version (5.12.8-0ubuntu1).
stlink-tools is already the newest version (1.6.0+ds-1).
build-essential is already the newest version (12.8ubuntu1.1).
cmake is already the newest version (3.16.3-1ubuntu1.20.04.1).
git is already the newest version (1:2.25.1-1ubuntu3.14).
lib32ncurses-dev is already the newest version (6.2-0ubuntu2.1).
lib32z1 is already the newest version (1:1.2.11.dfsg-2ubuntu1.5).
python3-setuptools is already the newest version (45.2.0-1ubuntu0.2).
python3-tk is already the newest version (3.8.10-0ubuntu1~20.04).
software-properties-common is already the newest version (0.99.9.12).
wget is already the newest version (1.20.3-1ubuntu2.1).
python-pip-whl is already the newest version (20.0.2-5ubuntu1.11).
python3-pip is already the newest version (20.0.2-5ubuntu1.11).
qt5-default is already the newest version (5.12.8+dfsg-0ubuntu2.1).
libncurses5:i386 is already the newest version (6.2-0ubuntu2.1).
libpython2.7:i386 is already the newest version (2.7.18-1~20.04.7).
The following packages were automatically installed and are no longer required:
  binutils-aarch64-linux-gnu cpp-9-aarch64-linux-gnu cpp-aarch64-linux-gnu
  gcc-10-cross-base gcc-9-aarch64-linux-gnu-base gcc-9-cross-base
[... 117 lines cut ...]
    PPM_UNIT

-- Build files have been written to: /home/anddyueng/samba/workspace_EdgeTx/edgetx/build/native
[ 50%] Built target native-configure
Scanning dependencies of target arm-none-eabi-configure
[ 60%] Creating directories for 'arm-none-eabi'
[ 70%] No download step for 'arm-none-eabi'
[ 80%] No patch step for 'arm-none-eabi'
[ 90%] No update step for 'arm-none-eabi'
[100%] Performing configure step for 'arm-none-eabi'
loading initial cache file /home/anddyueng/samba/workspace_EdgeTx/edgetx/build/arm-none-eabi-prefix/tmp/arm-none-eabi-cache-Debug.cmake
CMake Error at cmake/toolchain/arm-none-eabi.cmake:14 (cmake_path):
  Unknown CMake command "cmake_path".
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:3 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
make[3]: *** [CMakeFiles/arm-none-eabi-configure.dir/build.make:63: arm-none-eabi-prefix/src/arm-none-eabi-stamp/arm-none-eabi-configure] Error 1
make[2]: *** [CMakeFiles/Makefile2:224: CMakeFiles/arm-none-eabi-configure.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:259: CMakeFiles/configure.dir/rule] Error 2
make: *** [Makefile:196: configure] Error 2

snowuyl, 2025-04-25:
I have upgrade Qt5 to and using EdgeTx 2.11 branch to build again.
cd edgetx
#git checkout 2.11
git branch
qmake --version
#./tools/setup_buildenv_ubuntu20.04.sh
mkdir build
cd build
cmake -DARM_TOOLCHAIN_DIR=/opt/gcc-arm-none-eabi/bin -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
make configure
* 2.11
  main
QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_ARGS: -DARM_TOOLCHAIN_DIR=/opt/gcc-arm-none-eabi/bin;-DDEFAULT_MODE=2;-DGVARS=YES;-DLUA_MIXER=YES;-DPCB=X10;-DPCBREV=TX16S;-DPPM_UNIT=US
-- CMAKE_BUILD_TYPE: Debug
-- Configuring done
-- Generating done
-- Build files have been written to: /home/anddyueng/samba/workspace_EdgeTx/edgetx/build
Scanning dependencies of target arm-none-eabi-configure
[ 10%] Creating directories for 'arm-none-eabi'
[ 20%] No download step for 'arm-none-eabi'
[ 30%] No patch step for 'arm-none-eabi'
[ 40%] No update step for 'arm-none-eabi'
[ 50%] Performing configure step for 'arm-none-eabi'
loading initial cache file /home/anddyueng/samba/workspace_EdgeTx/edgetx/build/arm-none-eabi-prefix/tmp/arm-none-eabi-cache-Debug.cmake
CMake Error at cmake/toolchain/arm-none-eabi.cmake:14 (cmake_path):
  Unknown CMake command "cmake_path".
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeDetermineSystem.cmake:93 (include)
  CMakeLists.txt:3 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
make[3]: *** [CMakeFiles/arm-none-eabi-configure.dir/build.make:63: arm-none-eabi-prefix/src/arm-none-eabi-stamp/arm-none-eabi-configure] Error 1
make[2]: *** [CMakeFiles/Makefile2:224: CMakeFiles/arm-none-eabi-configure.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:259: CMakeFiles/configure.dir/rule] Error 2
make: *** [Makefile:196: configure] Error 2

snowuyl, 2025-04-25:
The above Unknown CMake command "cmake_path" issue has been solved after upgraded cmake version to 4.0.1. But I can't find any executable file under edgetx/build/ directory.
ls edgetx/build/
arm-none-eabi         CMakeCache.txt  cmake_install.cmake  native
arm-none-eabi-prefix  CMakeFiles      Makefile             native-prefix

snowuyl, 2025-04-25:
Thanks for your reply! Since I would like to run EdgeTX on stm32f429 board, I am not sure which folder under build directory are needed.

gagarinlg, 2025-04-25:
Are you aware that we will not accept any new radio with STM32F4 MCU?
With the exception of radios from existing partners that we already know about.

pfeerick, 2025-04-25:
>  I would like to run EdgeTX on STM32F429 board. 

Is this for a radio to be supported by EdgeTX or a personal project? As if it is the former, as mentioned above, STM32F4 designs are no longer accepted. 

> I try to run make command under edgetx/build/arm-none-eabi folder. But "Built target AccessDenied" is shown.
> $ make
> Built target AccessDenied
> 

`AccessDenied` is simply one of the third party libraries used in the build.

Are you following the steps on to wiki page here? 
 https://github.com/EdgeTX/edgetx/wiki/Build-Instructions-under-Ubuntu-20.04#building-edgetx-firmware-for-the-radio

As you don't just run `make` with no parameters, you have to tell it what target to build ... i.e. 

```
make firmware
```

or 

```
make -j`nproc` firmware
```

to use multiple threads to speed up the build.

snowuyl, 2025-04-25:
Thanks for your reply! The following is log messages after adding "make -j`nproc` firmware" command.
cd edgetx
#git checkout 2.11
git branch
qmake --version
#./tools/setup_buildenv_ubuntu20.04.sh
mkdir build
cd build
cmake -DARM_TOOLCHAIN_DIR=/opt/gcc-arm-none-eabi/bin -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
make configure
make -j`nproc` firmware
 2.11
  main
QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
mkdir: cannot create directory ‘build’: File exists
-- CMAKE_ARGS: -DARM_TOOLCHAIN_DIR=/opt/gcc-arm-none-eabi/bin;-DDEFAULT_MODE=2;-DGVARS=YES;-DLUA_MIXER=YES;-DPCB=X10;-DPCBREV=TX16S;-DPPM_UNIT=US
-- CMAKE_BUILD_TYPE: Debug
-- Configuring done (0.1s)
-- Generating done (0.0s)
-- Build files have been written to: /home/anddyueng/samba/workspace_EdgeTx/edgetx/build
[ 50%] Built target arm-none-eabi-configure
[100%] Built target native-configure
[100%] Built target configure
[100%] Built target arm-none-eabi-configure
[  0%] Generating tx16s.json
[  0%] Building C object radio/src/CMakeFiles/freertos.dir/thirdparty/FreeRTOS/list.c.obj
[  0%] Building C object radio/src/CMakeFiles/stm32cube_ll.dir/thirdparty/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_gpio.c.obj
[  0%] Building C object radio/src/CMakeFiles/stm32_stdperiph.dir/targets/common/arm/stm32/f4/system_stm32f4xx.c.obj
[  0%] Built target AccessDenied
[  1%] Building CXX object radio/src/CMakeFiles/stm32_drivers_w_dbg_fw.dir/targets/common/arm/stm32/sdcard_spi.cpp.obj
[  1%] Building CXX object radio/src/CMakeFiles/stm32_drivers.dir/targets/common/arm/stm32/stm32_hal.cpp.obj
[  1%] Building C object radio/src/CMakeFiles/stm32usb_device_fw.dir/targets/common/arm/stm32/usbd_conf.c.obj
[  1%] Generating bmp_plug_usb.lbm
[  1%] Building C object radio/src/CMakeFiles/stm32usb.dir/thirdparty/STM32_USB_Device_Library/Core/Src/usbd_core.c.obj
[  1%] Generating splash_logo.lbm
[  1%] Building CXX object radio/src/CMakeFiles/board.dir/targets/horus/board.cpp.obj
[  1%] Building C object radio/src/CMakeFiles/freertos.dir/thirdparty/FreeRTOS/queue.c.obj
Traceback (most recent call last):
Traceback (most recent call last):
  File "/home/anddyueng/samba/workspace_EdgeTx/edgetx/radio/util/encode-bitmap.py", line 4, in <module>
  File "/home/anddyueng/samba/workspace_EdgeTx/edgetx/radio/util/encode-bitmap.py", line 4, in <module>
    from PIL import Image
    from PIL import Image
ModuleNotFoundError: No module named 'PIL'
ModuleNotFoundError: No module named 'PIL'
make[7]: *** [radio/src/bitmaps/480x272/CMakeFiles/bm480_bootloader_bitmaps.dir/build.make:75: radio/src/bitmaps/480x272/bmp_plug_usb.lbm] Error 1
make[7]: *** [radio/src/bitmaps/480x272/CMakeFiles/bm480_splash_logo.dir/build.make:74: radio/src/bitmaps/480x272/splash_logo.lbm] Error 1
make[6]: *** [CMakeFiles/Makefile2:1283: radio/src/bitmaps/480x272/CMakeFiles/bm480_bootloader_bitmaps.dir/all] Error 2
make[6]: *** Waiting for unfinished jobs....
make[6]: *** [CMakeFiles/Makefile2:1091: radio/src/bitmaps/480x272/CMakeFiles/bm480_splash_logo.dir/all] Error 2
[  1%] Building C object radio/src/CMakeFiles/freertos.dir/thirdparty/FreeRTOS/tasks.c.obj
[  1%] Building C object radio/src/CMakeFiles/stm32cube_ll.dir/thirdparty/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_fmc.c.obj
Traceback (most recent call last):
  File "/home/anddyueng/samba/workspace_EdgeTx/edgetx/radio/util/hw_defs/generate_hw_def.py", line 7, in <module>
    from generator import generate_from_template
  File "/home/anddyueng/samba/workspace_EdgeTx/edgetx/radio/util/hw_defs/generator.py", line 3, in <module>
    import jinja2
ModuleNotFoundError: No module named 'jinja2'
make[7]: *** [radio/src/CMakeFiles/minimal_board_lib.dir/build.make:75: radio/src/tx16s.json] Error 1
make[7]: *** Deleting file 'radio/src/tx16s.json'
make[6]: *** [CMakeFiles/Makefile2:629: radio/src/CMakeFiles/minimal_board_lib.dir/all] Error 2
[  1%] Building C object radio/src/CMakeFiles/stm32usb.dir/thirdparty/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c.obj
[  2%] Building CXX object radio/src/CMakeFiles/board.dir/targets/horus/led_driver.cpp.obj
[  2%] Building C object radio/src/CMakeFiles/stm32usb_device_fw.dir/targets/common/arm/stm32/usbd_desc.c.obj
[  2%] Building CXX object radio/src/CMakeFiles/stm32_drivers.dir/targets/common/arm/stm32/timers_driver.cpp.obj
[  3%] Building C object radio/src/CMakeFiles/stm32cube_ll.dir/thirdparty/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_tim.c.obj
[  3%] Building CXX object radio/src/CMakeFiles/stm32_drivers.dir/targets/common/arm/stm32/diskio_spi.cpp.obj
[  3%] Building C object radio/src/CMakeFiles/stm32usb.dir/thirdparty/STM32_USB_Device_Library/Core/Src/usbd_ioreq.c.obj
[  3%] Building ASM object radio/src/CMakeFiles/stm32_stdperiph.dir/boards/generic_stm32/startup/startup_stm32f42_43xxx.s.obj
[... 67 lines cut ...]
[ 10%] Building CXX object radio/src/CMakeFiles/board.dir/targets/common/arm/stm32/watchdog_driver.cpp.obj
[ 10%] Building CXX object radio/src/CMakeFiles/board.dir/drivers/frftl.cpp.obj
[ 10%] Building CXX object radio/src/CMakeFiles/board.dir/targets/horus/lcd_driver.cpp.obj
[ 11%] Building CXX object radio/src/CMakeFiles/board.dir/targets/common/arm/stm32/heartbeat_driver.cpp.obj
[ 11%] Building CXX object radio/src/CMakeFiles/board.dir/targets/common/arm/stm32/lsm6ds_driver.cpp.obj
[ 11%] Built target stm32cube_ll
[... cut ...]
[ 12%] Building CXX object radio/src/CMakeFiles/board.dir/targets/horus/usb_charger_driver.cpp.obj
[ 12%] Built target board
make[5]: *** [CMakeFiles/Makefile2:905: radio/src/CMakeFiles/firmware.dir/rule] Error 2
make[4]: *** [Makefile:384: firmware] Error 2
make[3]: *** [CMakeFiles/firmware.dir/build.make:70: CMakeFiles/firmware] Error 2
make[2]: *** [CMakeFiles/Makefile2:486: CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:493: CMakeFiles/firmware.dir/rule] Error 2
make: *** [Makefile:280: firmware] Error 2

snowuyl, 2025-04-25:
Thanks for your great support! Firmware and bootloader can be built successfully on Ubuntu 20.04 with the following commands.
pip3 install clang
pip3 install jinja2
pip3 install lz4
pip3 install pillow

#git clone --recursive -b main https://github.com/EdgeTX/edgetx.git edgetx
cd edgetx
#git checkout 2.11
git branch
qmake --version
#./tools/setup_buildenv_ubuntu20.04.sh
mkdir build
cd build
cmake -DARM_TOOLCHAIN_DIR=/opt/gcc-arm-none-eabi/bin -DPCB=X10 -DPCBREV=TX16S -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DLUA_MIXER=YES -DCMAKE_BUILD_TYPE=Debug ../
make configure
make -j`nproc` firmware
find . -name *.elf
./edgetx/build/arm-none-eabi/firmware.elf
./edgetx/build/arm-none-eabi/radio/src/targets/common/arm/stm32/bootloader/bootloader.elf

closed 2025-04-25 by snowuyl