EdgeTX/edgetx #6148: Playback from a different switch's position

MrBlackmidi, 2025-04-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

For example the motor hold switch, when you flick it off there isn't a way to know what the for example the flight mode or bank switch is set to

### Describe the solution you'd like

Allow the for example the motor hold switch to play the track from a different switch's (flick on hold, it would say "hold", flick off hold it would say "bank one"  for example)

### Describe alternatives you've considered

You could like with Mikado Vbar and set the hold switch to say "motor off" and "motor on" but the issue of the bank or flight mode switch being temporarily unknown to the user

3djc, 2025-04-27:
You can choose today what sound you play for what switch position (or combination of switches position using logical switches) none of that is predefined, so what you want is really unclear.

MrBlackmidi, 2025-04-27:
what i mean is like on Spektrum Radios, when you flick off hold while in normal mode or stunt one, it would say one or the other than silence i.e. on a TX16s, hold would be on the H switch and the flight mode or bank switch would be on the E switch and when you flick off hold, it would say the track based on the position of the E switch https://youtu.be/mLR3lgkJb-U?si=7oz364rAfhPn36Yk

pfeerick, 2025-04-27:
> For example the motor hold switch, when you flick it off there isn't a way to know what the for example the flight mode or bank switch is set to

So what about using the Play Track SF, to play plays Hold when SH is pressed. And then something like three logical switchs to check the state of E when H is up, and trigger the correct Play Track SF for each switch position? 

Unlike Spektrum, **YOU** can program the transmitter to play things on whatever conditions and inputs you can think up ;)

pfeerick, 2025-04-27:
i.e. something like this actually works...

[image]

[image]