EdgeTX/edgetx #6153: 2.11.0 RC4 "Write Models and Settings to Radio" disabled in Companion

RC-SOAR, 2025-04-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Even after verifying a valid USB connection, "Write Models and Settings to Radio" is shown disabled (grayed out) in the Companion menu. Applies to the GX12 profile and some others.

### Expected Behavior

The option should be enabled.

### Steps To Reproduce

Open Companion 2.11.0 RC4. Create a model and attempt to write to radio.

### Version

2.11.0-rc

### Transmitter

RadioMaster GX12

### Operating System (OS)

Windows

### OS Version

10

pfeerick, 2025-04-26:
This may just be new unexpected behaviour. Is the model status indicator indicating there is a issue with your model setup? 

ie. here I have MODEL55 in error (there is one model with errors), and if I click the greyed out button in the toolbar after selecting the model with the *name* in red, I'll get told what the issue with the model is. 

[image]

[image]

RC-SOAR, 2025-04-26:
You're right of course ... the status indicator was red!

I didn't notice it, neither did I relate it to the disabled state of the 'Write Models' menu item. Partly because a red indicator for rx # didn't prevent uploading in 2.10. And partly because I simply unaware of the new behaviour. I think the new behaviour needs to be more obvious, also it's not obvious how to display a list of errors. 

A couple of suggestions:

1. IMO the correct location for the 'display errors' button is next to the Models status indicator. See red, click the button next to it, voila!

2. How about keeping the 'Write Models' menu item enabled, but display a message 'Errors in setup must be fixed before writing'? I think it would help to avoid confusion, as it's a fairly major change in behaviour. (A user of one of my templates was also confused by this.)

(BTW the 'display errors' button doesn't actually work at the moment - I suspect this is already due to be fixed?)

pfeerick, 2025-04-26:
> BTW the 'display errors' button doesn't actually work at the moment - I
suspect this is already due to be fixed

No, it does work... It only displays errors for the selected model, when
relevant. i.e. you need to click on the model name with red name for it to
work.

RC-SOAR, 2025-04-27:
My mistake - it's working as you describe.

pfeerick, 2026-03-12:
Yeah, I think something will be needed ... effectively if there is an error on the document, pressing any of the currently disabled options could just behave the same as the show errors button instead of doing their normal task. I.e short circuit their operation with explanation.

FZLAE, 2026-05-17:
Same here. Edge Tx user since 3 years, have 25 model files. Updated to "Queen Ann's Revenge" and ALL model files are broken both in Companion and on the TX16. It had no source on the input channels, logical switches incl. sound files 
assigned to them disappeard. 

Fixed this on TX16 first, it was a pain in the a.. Was test-flying cheap old foamy models to verify all is working. Flights went uneventful, no issues at all.

There seems to be an issue with the sound files too. E.g. mine tells me I am in FM-1, disarmend, flaps up etc. When turning on the TX16, there is some jitter "fl-iiii-ght m-oooo-de 1" when selecting another model (not fixable) and it had set calling to "1s" instead of "1x" as was. This resulted in the same message repeated in an endless loop. Was fixable by setting it back to "1x".  

But when I read SD card content in Companion, it once again marks all models in red saying input source is missing. And it only shows me the flight modes (which I have on the 6 center buttons) but everything in logical switches is gone. 

This is clearly an error in Companion and should be fixed ASAP.

I am running Companion on Ubuntu 24.

Cheers
Christian