EdgeTX/edgetx #6159: GX12 controls plane without full mixes switches stop working

arkarkark, 2025-04-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Twice now with two different models (Pumpkin and Diana Jet) my GX12 has got into a state where buttons/switches no longer work and it appears to be in high rates mode. I have these models set up so that when I press momentary shoulder switch SG I get full throws on my ailerons. Usually they only go to 40%. This allows me to do sick quick rolls, but control my plane normally when not being fancy.
Last week after doing some super fast rolls using SG the plane remained in high rates mode and the switches stopped working since I put it in RTH and it carried on flying under my control, not RTH. Luckily I still had control, but no switches worked. The same thing happened today with my pumpkin, I've attached the model file since it's simpler than the jet which has a flight controller.

I managed to bring this one down somewhat safely when I worked out what was going wrong and took a video of the radio sending full throws and the switches not working. It says "excellent" when I press SG, but it no longer did it in this weird stuck state. Then, leaving the plane powered up I power cycled my radio and it went back to normal mode and after connecting I only had 40% throws again until I press SG.

https://youtu.be/asbKazfmVcQ

[pumpkin.yml.txt](https://github.com/user-attachments/files/19932137/pumpkin.yml.txt)

This might be somewhat related to https://github.com/EdgeTX/edgetx/issues/5810 but it's not quite as serious and I've had a new motherboard and I'm running latest firmware since that happened.

I'm running the latest 2.11 RC 4

### Expected Behavior

releasing SG should put my plane back into 40% throws mode and the switches and buttons should still work.

### Steps To Reproduce

1. fly around and press SG and do super cool quick rolls.
2. at some point (not every time) it might get stuck in high rates mode (SG down) and it will no longer play track that says "Excellent"

### Version

2.11.0-rc

### Transmitter

RadioMaster GX12

pfeerick, 2025-04-28:
Ouch! I've not in a position to watch the video atm, but the crux of this is SG stopped responding completely? Can you list what other switches haven't worked in this state also, to see if there is anything common to those. However, I suspect if one stops working, they all stop, ideally (i.e. fixable) due to a firmware/timing bug.

arkarkark, 2025-04-28:
I'm pretty sure they all stopped working. but for sure the momentary SG and the 3 position SB (since I use that for RTH)
next time I can confirm that they all stop working. Thanks for the quick reply.

[image]

arkarkark, 2025-04-28:
I also use SH to blip 100% throttle so next time I fly I'll be playing these buttons like an accordion to try and trigger this bug and collect more data.

arkarkark, 2025-04-28:
not sure if it's worth mentioning, but I ram the same setup on my jumper t-pro (and t-pro v2) and never had this problem. I'm not stoked with how squishy these gx12 buttons are and wonder if it's debouncing logic causing it? just a totally uninformed guess, but you mentioning timing brought it to mind.

philmoz, 2025-04-28:
You said you replaced the motherboard - might be worth double checking all the cables and connections are properly seated.

arkarkark, 2025-05-04:
This happened again twice today. Luckily I was able to land both times since I was flying my dumb wing and the only behavior was full throws. I took another video if you want to see it I can share it. I also have the sdcard log file too if that might help.

I tested more and indeed no switches on the top or the radio work (SB -> SG) they don't do anything and the indicator on the screen showing their state doesn't change too. and SG the momentary press button that behaves like it is locked down remains in the pressed state on the screen even though it is not being pressed.

The 6 color led buttons just below the power button also do not work. I have some of them call a specail function and they do not do that.

The trim buttons DO work. as do the MDL,  jogwheel/enter and RTN, Page> page< (I didn't check tele or sys)
The power button also worked to turn the radio off and when I turned it back on everything worked as usual and expected.

It is not just this model, I have another model the same thing has happened on. that one crashed because my return to home switch change didn't get picked up.
I've never had this problem with the same models on my jumper t-pro (v1 and v2). I'm close to switching back to that as I'm losing faith in this GX12.

arkarkark, 2025-05-07:
I managed to reproduce this at home while not flying today. I verified that the pots and sliders (even the ones on the button/switch breakout board) still work. I also switched models and then back it still had the same problem. I also went into sys->hardware and changed some of the buttons from toggle to 2pos and then back that did not fix the problem either.
I'll try and work out a quick way to get into this state so I can help verify a fix or collect more data for you. I could even run a special build which would collect data from some sys or model menu item if that'd help.

Krystm, 2025-05-13:
I am having MAYBE a similar issue. Although may be slightly different. None of my switches are identifying themselves in the radio. Like They are on the back end, and SEEM to work on the outputs screen when they want to work. But the weird thing is that when setting a switch at special feature / global or even mixes. It will see the postion like Up or Down or even a Pot. BUT it doesnt put the lettering on it so you dont really know. So like SE is my Arm. All I can see is the Arrow go up or down when its flicked. I am also using a GX12. I was working fine on 2.11 RC4 but went to the final today and its doing all that. I even completely wiped the "internal drive since no SD" and did a fresh install and nothing. Even on a blank fresh install it wouldn't identify the buttons properly.

arkarkark, 2025-05-26:
This is still happening to me on a regular basis. Not every single flight but at least one time each day that I fly. It has caused quite a few crashes and I'm worried I'll end up losing a plane because of it.
Is there anything I can do to help track this down?

I'm sure I'm triggering this bug because I press SG a lot, but that doesn't mean other users won't come across it and also lose control of their models. The big issue is, once this happens the switches stop working completely and so you can not put your model into RTH or some other failsafe, whatever you wanted to happen momentarily with the switch is now happening all the time and there is no way to stop it or go back to normal unless you have the presence of mind and time to turn of your transmitter and turn it back on and hope you reconnect with your model.

raphaelcoeffic, 2025-05-27:
@arkarkark since you did not mention it: did you update to 2.11.1 or are you still using 2.11.0-rc0?

arkarkark, 2025-05-28:
I thought I was, but I was only running 2.11-rc4
I just flashed 2.11.1

arkarkark, 2025-05-31:
I tried the new nightly today (I confirmed it was the one with #6325 in it)  and I thought this was fixed, but towards the end of the morning the buttons all locked up again. this time it was the SE switch (3 pos, not momentary) which caused the lock. I never had the buttons lock up by using SG and I hammered on SG all morning long. flipping SE was an accident, but no other buttons would work after it locked up.

gagarinlg, 2025-06-02:
The sticks are using a different interface to the processore than the switches, so it is etremely unlikely to affect the sticks

hw999, 2025-06-02:
@3djc yes - I can try this binary. The GX12 does not have a removable SD-card and I have problems opening the built-in SD-card under Linux mint. The SD-card opens read only all the time. Companion can not find the SD-card. Never encountered this on a couple of TXs with removable SD-cards. Do you have a hint how to solve the problem ? When connecting to a Windows laptop I don't have this problem.

hw999, 2025-06-02:
@3djc Checked with the firmware you sent. The momentary switches SA SD SG and SH are sometimes counting more than two - one for pushing and one for releasing. In the region of 1 out of 5 triggers they count 3 or 4. I did not test for a long time but the other switches seem not to be affected.

3djc, 2025-06-02:
2 is normal, that's one for each position changes, more means some bouncing does happens, but likely not critically if you get 'only' 3 or 4. I'd rather not have those tho

arkarkark, 2025-06-04:
I'm also seeing 3 & 4 count increments with SA SD SG SH 
My SD is configured as a 2POS and it still sometimes counts 3

Should I go back to the nightly or is this firmware just as good for flying as the nightly?

pfeerick, 2025-06-04:
> The SD-card opens read only all the time.

This could indicate file system corruption or that the handset was disconnected without unmounting the file system first. When the file system has the "dirty" bit set, Linux will only mount read-only unless you fix or force it otherwise (not recommended). 

Run `sudo dosfsck -a /dev/##devicename##` to check and fix file system, which will also clear the dirty bit if everything is ok. You can `lsblk` to figure out the device name, or check the last few lines of `dmesg` if you've just plugged the handset in.

3djc, 2025-06-04:
> I'm also seeing 3 & 4 count increments with SA SD SG SH 
> My SD is configured as a 2POS and it still sometimes counts 3
> 
> Should I go back to the nightly or is this firmware just as good for flying as the nightly?

You can fly with it

hw999, 2025-06-04:
@pfeerick  Direct hit ! Problem solved. Thanks a lot.

hw999, 2025-06-05:
@arkarkark  yes, post was about sd-card. 
Regarding the switch lock - I have to wait and see. It occured only once in 10-15 hours of use running 2.11.0. Now running 2.11.1. Around two hous of use running fine.

3djc, 2025-06-07:
Working on an upcoming radio, we identified a rare occurrence of an issue that may affect the port expander used also in GX12.  @hw999 and @arkarkark could you test this ?

[firmware.bin.zip](https://github.com/user-attachments/files/20638182/firmware.bin.zip)

arkarkark, 2025-06-07:
excellent! will do later today. Thanks.

hw999, 2025-06-07:
@3djc I test if you tell what I should be looking for. The same same a in the previous test-firmware ? Toggle count ?

3djc, 2025-06-07:
Nop, toggle count was more for me to try to understand, just check switch always work

arkarkark, 2025-06-07:
Sadly, this new firmware didn’t fix it for me. SG still locks on. 

> Working on an upcoming radio, we identified a rare occurrence of an issue that may affect the port expander used also in GX12. [@hw999](https://github.com/hw999) and [...]

arkarkark, 2025-06-21:
> [@arkarkark](https://github.com/arkarkark), [@hw999](https://github.com/hw999) could you try this ? I look like it might nail it finally
> 
> [firmware.bin.zip](https://github.com/user-attachments/files/20644757/firmware.bin.zip)

I’m still seeing the problem with this firmware. Would the nightly be a better option? I’m at Flite Fest right now and will be flying a lot tomorrow too.

pfeerick, 2025-06-21:
You'll need to wait for @3djc to comment as to the specifics of that particular bin as I have no idea what he was testing there... but if it is the same fix as #6358 then a nightly is probably not going to be any different, but at the same time *does* contain all the fixes and extra checks to date for that hardware... so is probably your best bet moving forward. If there are still issues moving forward, we will have play another round of wack-a-mole... 😇

3djc, 2025-06-22:
We have now introduced a complete set of measures:
- security so that interrupt cannot be missed (#5949)
- security so that if chip doesn't send an interrupt, then we query it not matter what at least every 100ms (#6325)
- security that if the chip doesn't respond to a query generated by 1 or 2, we reset the chip (#6358)

Those measures are now In main have been proven 100% effective on a different radio using the same chip with several hundreds of man hours of testing on a large sample of radios. Not once did it fail with this in place

arkarkark, 2025-06-28:
I’m still seeing this problem with #333c7fdd nightly. I took a video if you need it.

Krystm, 2025-06-28:
I am also still seeing this, funny enough it also happens when using simulator though usb. I feel like i kinda had it figured out it seems like multiple switch presses at the same time can trigger it although I haven’t found the exact combination. Definately scary when you land and cant disarm.  At least the sticks havent faulted yet.

arkarkark, 2025-07-01:
@pfeerick can you reopen this please? I'm still [seeing](https://youtu.be/rW_lZgfPzls) this issue

hawku, 2025-07-03:
I have same problem on my GX12.

Usually the problem occurs when trying to use turtle mode and suddenly I can't disarm the drone until I shutdown the radio.
I have used the radio over 6 months and the current mainboard for about 3 months without this issue but now I have to shutdown the radio almost daily to disarm the drone.

I think that the problem started after updating to 2.11.1, but I'm not sure.

Updated to 2.11.2 yesterday and today I had to shutdown the radio while playing simulator.
Radio was connected to PC through ELRS dongle.

arkarkark, 2025-07-04:
@3djc can you reopen this please? still seeing this problem on my GX12

pfeerick, 2025-07-05:
Sorry for the delay, just got back home.

3djc, 2025-07-21:
I cannot think about what else we could be doing given all we have already added to recover from this issue if it happens, nor can anyone replicate it, and this issue seems to affect only an extremely limited amount of users, so unsure

gagarinlg, 2025-07-21:
I'll send you a message via Discord, my username there is Malte

arkarkark, 2025-07-29:
my new GX12 motherboard has arrived and been installed. I'll comment back here if I still experience this problem. I'll close this issue, hopefully for good. Thanks for all the time you spent on this and dealing with me <3

arkarkark, 2025-08-03:
Alas. New motherboard and I still see this issue. 
I’m mailing my old motherboard to  @gagarinlg  next week.

arkarkark, 2025-08-17:
Whelp, still see this issue with a new motherboard. Running jolly 2.11.2
I took a video if you need it.

arkarkark, 2025-08-17:
Worth noting that this isn’t just sg switch that causes this, although that’s the one I press very frequently when flying (it gives me full aileron throws). This last time it happened with my arm switch (sc)

J-Sorenson, 2025-08-21:
A shot in the dark, but what global variables and functions do you run?  We won't see those from your model's yml file.

arkarkark, 2025-08-21:
> A shot in the dark, but what global variables and functions do you run? We won't see those from your model's yml file.

[image]

I run this [modnam.lua](https://gist.github.com/arkarkark/a255488c2cbdeccfdc094b1ee26b6238#file-modnam-lua) global function once, it plays a wav file of my model name.

I just added a lua script that monitors my battery that adjusts GV9 but this happened way before I wrote that script.

I think the big thing is that I have a momentary button that gives me full aileron throws and I push it a lot to do quick fun rolls. so I'm probably pressing the button an awful lot and most people aren't using their switches quite as much.

I have an MT12 that I also press buttons on a lot (to do throttle ramp up and also to start a race) but I don't press them nearly as much as my 'super aileron button'. this super aileron button moved over from my Jumper t-pro and t-pro v2 and I never had this problem with those.

Next time I fly (sadly in two weeks) I'm going to power cycle my radio after every battery and see if I get the problem again. 
I've already limited the damage this does to my flying by making my super aileron button only stay on for 1.5 seconds at most, that way if the buttons do lock on, I'm not left in super aileron mode which is what usually causes my crashes.

[image]

also pressing the momentary button plays a wav file (Bill and Ted's guitar riff)

[image]

[image]

but this isn't limited to just SG, I recently had it lock up the buttons when I flicked my arm switch (SC). It might be a red herring, but that button also plays a wav file (the ready to fly ding dong real airplanes make when the fasten seatbelt light turns on)

J-Sorenson, 2025-08-25:
Yet another example of perfectly good software getting challenged by real-life hardware idiosyncracies. If we actually lived in a perfect simulation, then the developers wouldn't need to filter analog interfaces.  This is too low-level for me, but it sounds like the devs will have to find a way to recover from this, rather than prevent it from occurring.  The question is where is the potential infinite loop and/or hung subprocess that is blocking the usual control flow?  There needs to be a way to escape out of that.

dividebysandwich, 2025-09-03:
I had the exact same issue as OP on v2.11.2 today, right after powerup. The first button I pressed once was SG and it did not work. And no other switch or button after that worked, except for menu navigation and the sticks. Power cycling restored button functionality.

dividebysandwich, 2025-09-07:
Okay so... V2.11.3 and here is a video. I'm afraid I will have to go back to the RM Pocket for the the time being.

https://youtu.be/56gzOpJrb1g

Better screen visibility:

https://youtu.be/K7gzfnB-OMA

3djc, 2025-09-13:
Why would you need a custom build for lua mixer scrips, they are in by default ?

At this point in time, we have not been able to replicate at all, and the build above does have a lot of mechanism to recover, which makes it even more strange

dividebysandwich, 2025-09-13:
> At this point in time, we have not been able to replicate at all, and the build above does have a lot of mechanism to recover, which makes it even more strange

You should be able to just by smashing SG for a few minutes. Sometimes it happens right away, sometimes after some time. If you really cannot replicate it then it sounds like some hardware problem of a limited number of units. It's probably not the mainboard though, since we know someone had theirs replaced.

3djc, 2025-09-13:
I smashed the keys and switches for 40mins this morning. All I got was my hands hurting, but switches kept working

dividebysandwich, 2025-09-13:
> I smashed the keys and switches for 40mins this morning. All I got was my hands hurting, but switches kept working

I too had trouble recreate the issue while trying to film the whole process. The thing worked for over 10 minutes of button mashing. So I thought I'd just shut down the radio and start it up again, and try again. Lo and behold, in under 3 minutes I got it to lock up.

Here's a video from when I started up the radio again until the buttons hang:

https://youtu.be/LNtgxqVp8pc

And here's my model file:

[model00.yml.txt](https://github.com/user-attachments/files/22314478/model00.yml.txt)

RadekRadekP, 2025-09-13:
I have just done it. No lua mixer scrips and no custom switches there. Done by Buddy.  So these options are not default.

pfeerick, 2025-09-13:
> I have just done it. No lua mixer scrips and no custom switches there. Done by Buddy. So these options are not default.

Depending on exactly what you mean there, no, sorry, but you are wrong. Lua mixer scripst are available for 2.11.3 for GX12 without the need for a custom build - you just need to go to either model or radio setup (depending on if you want it visible for one or all/most models), scroll down to enabled features, and turn it on! And of course there won't be any custom(isable?) switches enabled... you would need to configure them! They are still there though, under model setup, ready to be configured. 

> I have read that radios with small bw displays have lua scripts not included. 

Which was true until 2.11, as due to optimisations made during 2.11 development enough flash space was recovered to turn this on for targets with smaller flash storage space (it wasn't strictly anything to do with screen size/type).

RadekRadekP, 2025-09-13:
You are right,  I was mistaken.  Thank you for your patience.  I have found the customs scripts and  they work. Sorry for bothering you. Best regards,  Radek

3djc, 2025-09-14:
@dividebysandwich , please try with this firmware: 

[firmware.bin.zip](https://github.com/user-attachments/files/22318554/firmware.bin.zip)

It does allow logging errors handling to USB serial terminal:
- in sys/hardware, set Serial Port/USB-VCP to Debug
- plug USB cable and choose "USB serial (VCP)"

You need to use a serial terminal, like putty on windows, to display the radio logs. You might also need USB VCP drivers (can find here: https://blog.quadmeup.com/2016/12/03/stm32-virtualcomport-driver-for-windows/)

3djc, 2025-09-14:
ouch, it doesnt seem to be able to restore I2C bus, not great. Will need to think about it

3djc, 2025-09-14:
@dantob thanks for your test, really usefull info. I made another version with more info on I2C reset, could you test with this one? (my gx12 won't reproduce, I tried so much yesterday that I can barely move my left wrist today :s))

[firmware.bin.zip](https://github.com/user-attachments/files/22319190/firmware.bin.zip)

3djc, 2025-09-14:
Ok, here is an attempt to improve I2C reset

[firmware.bin.zip](https://github.com/user-attachments/files/22319864/firmware.bin.zip)

dividebysandwich, 2025-09-14:
> Ok, here is an attempt to improve I2C reset
> 
> [firmware.bin.zip](https://github.com/user-attachments/files/22319864/firmware.bin.zip)
> 
> If it works, you should see (on crash): ERROR: resetting PCA95XX ERROR: resetting PCA95XX I2C bus
> 
> but NOT followed by: ERROR: Unrecoverable error on PCA95XX

The behavior is slightly different now, everything (including sticks) hangs for a few seconds, then I get some stick movement again, then it hangs again. The buttons hang all the time.

I can't get any serial output to show. I am doing ```sudo cat /dev/ttyACM0 > output.log``` but I get nothing. I also tried setting various baud rates, for example ```sudo stty -F /dev/ttyACM0 115200 raw -echo```

dividebysandwich, 2025-09-14:
Ok now once I trigger the bug, the sticks hang everytime I press a button or toggle a switch, they hang for a few seconds and then start working again.
However the switches and buttons don't recover, as before.

```
46870ms: I18N_PLAY_FUNCTION v 127
50960ms: I18N_PLAY_FUNCTION v 127
54710ms: I18N_PLAY_FUNCTION v 127
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
56910ms: ERROR: resetting PCA95XX
56910ms: ERROR: resetting PCA95XX I2C bus
56910ms: ERROR: Unrecoverable error on PCA95XX
57510ms: I18N_PLAY_FUNCTION v 127
63950ms: [XF] ModelID 1
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
64010ms: ERROR: resetting PCA95XX
64010ms: ERROR: resetting PCA95XX I2C bus
64010ms: ERROR: Unrecoverable error on PCA95XX
65550ms: I18N_PLAY_FUNCTION v 127
68870ms: [XF] ModelID 1
69580ms: ERROR: resetting PCA95XX
69580ms: ERROR: resetting PCA95XX I2C bus
69580ms: ERROR: Unrecoverable error on PCA95XX
69580ms: ERROR: resetting PCA95XX
69580ms: ERROR: resetting PCA95XX I2C bus
69580ms: ERROR: Unrecoverable error on PCA95XX
69580ms: ERROR: resetting PCA95XX
69580ms: ERROR: resetting PCA95XX I2C bus
69580ms: ERROR: Unrecoverable error on PCA95XX
69580ms: ERROR: resetting PCA95XX
69580ms: ERROR: resetting PCA95XX I2C bus
69580ms: ERROR: Unrecoverable error on PCA95XX
69580ms: ERROR: resetting PCA95XX
69580ms: ERROR: resetting PCA95XX I2C bus
```

3djc, 2025-09-14:
ok, thanks. I have tried a very very raw reset of I2C bus, yet it is still not enough. I wonder if we should EM. We can try.

[46 more replies]

PR #6358 (2025-06-17): fix(gx12): use PCA95xx reset pin in case of read issues
This resets the PCA95xx expanders if they are found non responding to I2C query,

Fixes #6159

For 2.11 also please.
closed 2026-02-22 by pfeerick