EdgeTX/edgetx #6175: Not all channels shown in Play Val SF

inventor7777, 2025-05-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On my RadioMaster MT12, I have Channels 1,2, and 3 set up as ST, TH, EBRK, and Channels 13-16 set up for gyro controls. 

I need the Play Val function to play the value of the gyro gain channel, but it only shows the first three channels. 

### Expected Behavior

All channels should appear as an option when setting up PV SF.

### Steps To Reproduce

Try to make a PV SF which plays a channel above 3. 

Interestingly, my dad’s radio allows the first four channels, and it’s set up the same way minus the e-brake channel. Maybe it’s something like it will only go through three named channels?

### Version

Other (Please specify below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

macOS

### Anything else?

I’m using my self built no RTC firmware 4-15-25

Maybe it isn’t present in later versions, but I checked the issues and I saw nothing related.

I can send a video if needed but it’s a huge pain uploading to GitHub.

philmoz, 2025-05-01:
Can you upload the model file.

inventor7777, 2025-05-01:
@philmoz 

[model03.yml.zip](https://github.com/user-attachments/files/19999901/model03.yml.zip)

inventor7777, 2025-05-01:
I think I know why. It’s only showing channels which have mixes, but I’m controlling gyro channels with just overrides, so they don’t show up. Maybe a setting could be added like “Show All Channels” or another long press entry next to “Channels” like “All Channels”. Idk though

gagarinlg, 2025-05-01:
The gyro axis are input, just like stocks and not channels

inventor7777, 2025-05-01:
@gagarinlg 

Sorry if I didn’t make it clear. This is for the *car* receiver gyro, which I’m controlling with overrides. It’s not the MT12 gyro expansion module.

inventor7777, 2025-05-01:
I guess that’s what I’ll be forced to do

philmoz, 2025-05-01:
> I guess that’s what I’ll be forced to do

Yes. The 'Channels' section of the source picker is actually selecting mixes - not outputs.

pfeerick, 2025-05-01:
And for all intents and purposes, if it doesn't exist in the mixer, it doesn't exist... this is I think one of the reasons the SF Override function was a bad idea, and why there was an effort in OpenTX to get rid of it - as it breaks continuity. You should be able to use a mixer line in pretty much all, if all situations you could use a SF override. Just add a mixer line that sets the channel to a given position, which is only active on a switch... and it's done.

closed 2025-05-01 by pfeerick

J-Sorenson, 2025-05-02:
I have to agree that it's always been better to assign a mixer that was activated with a logical or physical switch.  It keeps everything in one window.  I also avoid the CH Override, as it can caused some mysterious issues in the past (I think it bypassed output limits).
Unfortunately, removing it now would cause much outcry with existing systems.  I vote to support what it currently does, but not to extend it.  3djc has the best advice: use a proper mixer when you need all of the features of a mixer.

3djc, 2025-05-02:
Mixer are really very very versatile and powerful.

You can do in a single line with a curve, several lines using switches, but we would need more details on your control settings

J-Sorenson, 2025-05-04:
Lots of people can help at https://www.rcgroups.com/forums/showthread.php?3916381-Official-EdgeTX-Discussion-Thread.  Anyhow, 2 strategies:  

1. Create a "Curve" with 3 points at -100, -70, and 10.  Create a Mixer with the switch as a "source" and apply the curve.
2. In Mixers, you could create 3 mixes for the channel with "MAX" as the input and set your desired amount as the "Offset."  Then enable each offset with the switch position.  This is essentially your original "CH Override" special function, which is why I still feel like the "CH Override" feature was redundant and added unnecessary complexity.

inventor7777, 2025-05-05:
Okay, interesting! As I further improve and simplify programming on my radios I will look into these (hopefully more elegant) solutions. Thanks 🙏